lumapu/ahoy #15: Loosing WiFi connection after X minutes

andreas5232, 2022-04-28:
Most important: You guys rock! :-)

I'm trying to run ahoy on a NodeMCU board with NRF attached.

Everything is working fine at first. Data is read from HM-600 and displayed correctly on the web interface. But after roundabout 10 minutes the system seems to loose its wifi connection. The web interface is no longer available and the device can't be pinged. The LED on the board keeps blinking in the same interval as before the break down. After a reboot/reset of the board everything is fine again for a few minutes. 

I'm not sure if I'll be able to debug this on my own and if it does affect other setups too, so I opened this issue.

Keep up the great work!

Cheers
Andreas

lumapu, 2022-04-28:
Hey Andreas,
I faced the same behaviour. Once it happend while my serial console was connected and I were able to check the exception.
I read somewhere that there should be no `delay(xxx)` in ESP code (while WiFi connection is established)

For me I commented two lines in app.cpp
https://github.com/grindylow/ahoy/blob/1e7a8ea8696b4ab503208b20ccfc18f42d34a992/tools/esp8266/app.cpp#L167
and
https://github.com/grindylow/ahoy/blob/1e7a8ea8696b4ab503208b20ccfc18f42d34a992/tools/esp8266/app.cpp#L187

Since then I have an uptime of more than 2 hours . I will commit this change if it does not interrupt any more in the next hours.

andreas5232, 2022-04-28:
Hey lumapu,

I've just flashed my NodeMCU with your potential fix.
Guess what? It's running since 30min without any interruptions or necessary reboots. So I'd consider it as a working fix. 

Thanks for your support and all your efforts with this project!

stefan123t, 2022-04-29:
@lumapu, yes this is correct the WiFi is running on the same core in the ESP8266. The ESP32 has two cores, hence the WiFi stack is running on one CPU core while most of the user program runs on the other. You have this concept also on the nRF52832/52840 where the BLE stack runs on one core and the user program on the other.

Hence you have to feed the watchdog every now and then. You can purposefully feed it using the yield() call method. Though I think you already committed that [somewhere](https://github.com/lumapu/ahoy/commit/b3bdf2811ec36edd0fbba435d313d3b12b968a40).

I have also seen that you implemented three tickers , which should be useful to order things in main/app that occur after specific periods, e.g. every 10 seconds or every 5 minutes. I tend to use old school C:
```
if (millis() < timer) 
   {
       timer = millis()+10000; 
       do the needful
}; 
```
as a kind of pseudo scheduler to prevent running some code more often than necessary.

lumapu, 2022-04-29:
@stefan123t, thank you for your comment - very interesting topic.
Does 'specific period' have a lower limit of time? For now two tickers which have an interval of 1s (uptime and send command), second has an interval of 10s (mqtt send).
I understand your old school C example. I' ll check the docs about tickers again.

For the uptime I think the ticker is more precisely because of a fired interrupt (I don't know it exactly but this is my idea how it works)

stefan123t, 2022-05-01:
@lumapu naturally the sky is the limit. Here we have to take the runtime of our individual co-routines into account. hence if they take more than several seconds to complete their tasks that will be the lower limit "sometimes". While if these long running sub routines are not scheduled at the moment, then the loop can complete quite fast, i.e. several times per second. The timers are based on milliseconds as a default so this is the numeric minimum =^D.

Regarding the interrupts I am not sure how well they behave as we also have a hardware interrupt for reading the nRF24L01+ and the WiFi is also kicking in via an interrupt. As mentioned this is probably more of an issue on ESP8266 which is why ESP32 having two cores is usually considered more stable.

You may also add Debug Level HTTP_SERVER or similar CORE+...+MDNS and use Debug Port Serial to enable the information from the ESP8266WebServer module. I have seen some request being queued on the network / router which seem to be worked by the WebServer after it comes up. Should we also reconsider when to enable and load the individual "webpages" in the WebServer ? Currently it starts the WebServer even before the WiFi connection is up and running.

stefan123t, 2022-05-01:
Early knowledge gained from issues on the esp8266 project by igrr (Ivan Grokhotkov) and others making the same mistakes before us. I have just re-read that and it looks like we have to call yield() or delay() every now and then to feed the watch dog to prevent it from starvation. **:heart: & :cut_of_meat: for :dog2:**

https://github.com/esp8266/Arduino/issues/34
https://randomnerdtutorials.com/interrupts-timers-esp8266-arduino-ide-nodemcu/

Oh BTW I have been diving deep into the rabbit hole some months ago with a thread called [Embed with Elliot](https://hackaday.com/tag/embed-with-elliot/) on Hackaday. Not sure I understood everything but I assume he got us covered with interrupts the good, the bad and the ugly: 

https://hackaday.com/2015/09/18/embed-with-elliot-interrupts-the-good/
https://hackaday.com/2015/09/25/embed-with-elliot-interrupts-the-bad/
https://hackaday.com/2015/10/02/embed-with-elliot-interrupts-the-ugly/

OT-Bonus:
https://hackaday.com/2015/09/04/embed-with-elliot-practical-state-machines/
https://majenko.co.uk/blog/our-blog-1/the-evils-of-arduino-strings-32

lumapu, 2022-05-01:
I just published a version which uses the 'old-school-c' timer as you described before. Testing it over the day had the same result as the ticker variant - we will see if the uptime can grow to more than a day with that implementation.

stefan123t, 2022-05-02:
Mist noch während ich den Kommentar im Forum abgesetzt habe, hat sich der ESP wohl unbemerkt von mir verabschiedet.
Werde jetzt doch mal die 0.3.2/3 kompilieren und testen...

Version war 0.2.11:

```
18:20:27.473 -> http-server loop: conn=1 avail=0 status=wait-read
18:20:27.473 -> check_poison_block is called for free block 0x3fff2208
18:20:27.473 -> Fatal exception 3(LoadStoreErrorCause):
18:20:27.473 -> epc1=0x401007ed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x400322d9, depc=0x00000000
18:20:27.473 -> 
18:20:27.473 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
18:20:27.506 -> 
18:20:27.506 -> Exception (3):
18:20:27.506 -> epc1=0x401007ed epc2=0x00000000 epc3=0x00000000 excvaddr=0x400322d9 depc=0x00000000
18:20:27.506 -> 
18:20:27.506 -> >>>stack>>>
18:20:27.506 -> 
18:20:27.506 -> ctx: sys
18:20:27.506 -> sp: 3fffec10 end: 3fffffb0 offset: 0190
18:20:27.506 -> 3fffeda0:  3ffeff58 00000004 3fff2208 40100a45  
18:20:27.506 -> 3fffedb0:  00000456 00000456 3ffe85dc 40100b51  
18:20:27.506 -> 3fffedc0:  00000456 00000456 3ffe85dc 3fff2214  
18:20:27.539 -> 3fffedd0:  00000000 00000020 3fff220c 40100e76  
18:20:27.539 -> 3fffede0:  00000000 00000000 00000000 40100e94  
18:20:27.539 -> 3fffedf0:  00000111 3ffe85dc 3fff3d90 4010057c  
18:20:27.539 -> 3fffee00:  00000111 00000000 3fff3a70 40224d90  
18:20:27.539 -> 3fffee10:  00000763 00000763 3ffe85dc 4021ccd1  
18:20:27.539 -> 3fffee20:  00000763 00000763 3ffe85dc 4021d7d8  
18:20:27.539 -> 3fffee30:  3fff2214 00000020 3fff3a74 4021de5f  
18:20:27.539 -> 3fffee40:  3fff2214 00000000 00000000 40100e94  
18:20:27.572 -> 3fffee50:  000063aa f5e2ab9f 00000000 4010057c  
18:20:27.572 -> 3fffee60:  3fff2214 3ffeff08 00000006 3ffeff04  
18:20:27.572 -> 3fffee70:  3fff2214 3ffeff08 00000005 4021dec8  
18:20:27.572 -> 3fffee80:  3fff2214 3ffeff08 00000005 402268df  
18:20:27.572 -> 3fffee90:  00000000 00000000 4bc6a7f0 00000000  
18:20:27.572 -> 3fffeea0:  00000019 00000000 3fff3338 401009b0  
18:20:27.572 -> 3fffeeb0:  00000000 00000000 3fff32f8 3ffefab8  
18:20:27.606 -> 3fffeec0:  3ffefd38 3fff3a9a 3fff3a7c 40224081  
18:20:27.606 -> 3fffeed0:  00000014 3ffefab8 3ffe85dc 3fff1e0c  
18:20:27.606 -> 3fffeee0:  00000000 00000020 3fff32fc 40100e76  
18:20:27.606 -> 3fffeef0:  3fffdc80 3fff03a4 3fff3304 3fff1e0c  
18:20:27.606 -> 3fffef00:  3ffefab8 00000008 3fff3a7c 4021cba1  
18:20:27.606 -> 3fffef10:  3fffdc80 3fff03a4 3fff3304 4021c9a0  
18:20:27.606 -> 3fffef20:  40244f29 3fff03a4 3fff3304 40244f3b  
18:20:27.639 -> 3fffef30:  3fff3a8c 3fff3a7c 00000000 3ffe85d8  
18:20:27.639 -> 3fffef40:  4024198b 00000000 3fff3304 40246213  
18:20:27.639 -> 3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49  
18:20:27.639 -> 3fffef60:  40000e19 00058a85 00000000 00000005  
18:20:27.639 -> 3fffef70:  3fffefc0 aa55aa55 000000ed 40105881  
18:20:27.639 -> 3fffef80:  40105887 00000000 00000005 40100e50  
18:20:27.639 -> 3fffef90:  4010000d 646f623c 00058a85 401000ac  
18:20:27.672 -> 3fffefa0:  40236910 3fffef3c 402368c9 3ffffe48  
18:20:27.672 -> 3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
18:20:27.672 -> 3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.672 -> 3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.672 -> 3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.672 -> 3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.672 -> 3ffff000:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.672 -> 3ffff010:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.705 -> 3ffff020:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.705 -> 3ffff030:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.705 -> 3ffff040:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.705 -> 3ffff050:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.705 -> 3ffff060:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.705 -> 3ffff070:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.705 -> 3ffff080:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.738 -> 3ffff090:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.738 -> 3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
18:20:27.738 -> 3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
[... 316 lines cut ...]
0x4021a540: _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 602
0x40215ee4: _scanf_float at /workdir/repo/newlib/newlib/libc/stdio/nano-vfscanf_float.c line 231
0x40212050: uart_init(int, int, int, int, int, size_t, bool) at /home/sthieme/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/uart.cpp line 661
0x40212050: uart_init(int, int, int, int, int, size_t, bool) at /home/sthieme/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/uart.cpp line 661
0x4020d68c: dhcps_stop() at /home/sthieme/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/LwipDhcpServer-NonOS.cpp line 61
0x40212841: experimental::_SPICommand(uint32_t, uint32_t, uint32_t, uint32_t, uint32_t*, uint32_t, uint32_t, uint32_t) at /home/sthieme/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_spi_utils.cpp line 77
0x40217bdd: memmem_P at /workdir/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c line 208
0x40210c2f: dtostrf(double, signed char, unsigned char, char*) at /home/sthieme/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_noniso.cpp line 88
0x402083fe: Main::setupAp(char const*, char const*) at /tmp/arduino_build_778324/sketch/main.cpp line 105
[... cut ...]
0x402084aa: Main::setupAp(char const*, char const*) at /tmp/arduino_build_778324/sketch/main.cpp line 110
0x40203e3f: app::loop() at /tmp/arduino_build_778324/sketch/hmInverter.h line 111
0x40204b00: S2Stream::peekConsume(unsigned int) at /home/sthieme/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/StreamString.h line 183
0x40210cbc: dtostrf(double, signed char, unsigned char, char*) at /home/sthieme/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_noniso.cpp line 107
```

lumapu, 2022-05-02:
Version 0.3.3 läuft bei mir seit 23:05h ohne Absturz, das Upgrade lohnt sich bestimmt. Die Uhr geht auch nur ein dreiviertel Minute nach - das passt von der Genauigkeit, man kann ja alle 24h einen NTP Update einbauen.

andreas5232, 2022-05-03:
Ich habe auch erfolgreich die 0.3.3 über Nacht laufen lassen und jetzt eine Uptime von über 24h. Aus meiner Sicht ist es damit stabil. Das NTP Update einmal am Tag wäre natürlich noch eine feine Sache.

rosch99, 2022-05-03:
Ich habe ebenfalls die 0.3.3 auf einem Wemos D1 mini der momentan direkt neben dem AccessPoint liegt und bekomme nur sehr selten Verbindung. Ich habe das Gefühl die WIFI-Verbindung war in den Versionen vor 0.3 schon mal stabiler. 
Nach einem Power OFF/ON funktioniert die Verbindung inkl. MQTT für 15-30min, anschliessend nur mehr eine (1) MQTT Verbindung ziemlich genau jede Stunde. Die Webseite ist nicht mehr erreichbar bis zum nächsten Power OFF/ON.
Ich werde es aber zur Sicherheit in den nächsten Tagen noch mal mit neuer Hardware probieren.

lumapu, 2022-05-03:
Den MQTT Fehler sehe ich bei mir auch, das zwickt sich mit der Systemstabilität da es mit dem delay auf MQTT Seite besser funktioniert hat. Interessant ist hier dieser Issue von der MQTT Lib:
https://github.com/knolleary/pubsubclient/issues/182

stefan123t, 2022-05-03:
Ich habe ihn auch mutmaßlich längere Zeit am Laufen. Nur hat er wegen Empfangsproblemen die Verbindung zum WLAN eingestellt, so daß ich nicht weiß ob er durchgelaufen ist oder einen Reset hatte ... Im Serial Monitor sehe ich nur die Nachrichten alle paar Sekunden.
Also zusätzlich zum NTP Update min. ein/mehrmals am Tag sollte das Programm m.E. auch versuchen die konfigurierte WLAN Verbindung alle 10 Minuten wieder zu aktivieren.

rosch99, 2022-05-05:
[image]
Anbei ein screenshot (Grafana) mit einem Vergleich HM1200 und einer ShellyPM am selben AC-Anschluss.
Angezeigt werden die letzten beidenTage, gestern mit der Firmware 0.3.5 und heute mit 0.3.6.
MIr ist aufgefallen, dass beim restart des WIFI manchmal die MQTT Verbindung nicht neu connected wird,
das scheint hier dann ebenfalls als "offline" auf.

stefan123t, 2022-05-06:
Hallo Lukas, ich habe die 0.3.6 gestern morgen kompiliert und deployed. Er startet jetzt mit dem AP obwohl er offenbar auch eine WLAN Verbindung bekommt. Also er scanned die Netzwerke und findet auch den WLAN Access Point des Routers, vermutlich klappt es nicht beim ersten Versuch, denn er macht dann gleich den eigenen AP auf und startet den 3x60=180 Sekunden Countdown. Danach passiert meist das Selbe, d.h. es dauert mehrere 180 Sekunden Intervalle bevor es mal beim ersten Versuch gelingt eine Verbindung mit dem Router aufzubauen.

Ich hätte erwartet, daß er 30 Sekunden lang versucht eine Routerverbindung aufzubauen und erst dann für 180 Sekunden einen lokalen AP aufmacht und offenhält. 
  
Interessant fand ich die `join` und `leave` Meldungen im AP Modus. Damit könnte man auch die Clients die sich am AP angemeldet haben zählen/bemerken und ggf. den Ticker steuern, der aktuell alle 3 Minuten (3x60s) den AP beendet. Wenn der AP also in den 180 Sekunden keinen `join` sieht, dann kann er den auch wieder zumachen. Andernfalls ist man ja mutmaßlich am Konfigurieren und er sollte den offenlassen. Wenn er im AP Modus ein `leave` sieht (oder ein zugehöriger Zähler verkleinert wurde) dann kann er ja nochmal 3 Minuten Zeit geben bevor er den AP wieder zu macht. Evtl. kann man auch Aufrufe des WebServers im AP Modus zum Triggern dieser 3 Minuten Grace Period verwenden ?

Ggf. alle 15 Minuten dann nochmal die Verbindung mit dem Router für 30 Sekunden versuchen, damit wir auch Daten hochladen können ...

stefan123t, 2022-05-06:
> Den MQTT Fehler sehe ich bei mir auch, das zwickt sich mit der Systemstabilität da es mit dem delay auf MQTT Seite besser funktioniert hat. Interessant ist hier dieser Issue von der MQTT Lib: [knolleary/pubsubclient#182](https://github.com/knolleary/pubsubclient/issues/182)

Hi Lukas,
I have checked the linked MQTT / PubSubClient issue for having subscribed two topics at the same time. 
This is what thijsdebont added lately:
> First, the delay(1000); in loop() is bad practice. It blocks everything for a long time. 
> Second, the char pointers 'payload' and 'topic' are used as both a global variable for publishing ánd in the mqtt callback. 
> The globals should be given another name, since it is likely it gets partially overwritten. 
> Then the length of the global topic and topic2 variables. 150 is way too much. 
> Set it something that actually matches the max topic length.

I do not know if it really is matching our WDT resets as in both my StackTrace (click it to open) above and the one supplied for 0.2.9 by @dad401 there is this umm_init on the top of the stack which is for memory handling.
Probably we should revisit our use of String class in the ESP8166WebServerTemplates.
Maybe there is a smarter way to do things there, which does not involve creating copies of the Templates in memory before sending out ? 
This would affect the umm Memory Management less and therefor could prevent WDT resets too.
Also note that I do not have MQTT active and therefor you do not see the PubSubClient on my stack.

But given we removed the delay(500); from our loop the bad practice (that thijsdebont mentioned too) should be gone.
The now longer stable runtimes of several hours / days speak volumes!

lumapu, 2022-05-06:
Hallo stefan123t,

das mit dem join und leave klingt sehr gut, werde ich mir anschauen. Macht durchaus Sinn, den AP aktiv zu lassen wenn ein Client angemeldet ist. Aufrufe - aber dann nur auf der Setup-Seite, denn dafür war der AP Modus ja mal gedacht.

Was ich auch noch gerne hätte, wäre eine Liste der in der Nähe verfügbaren WLANs.

lumapu, 2022-05-06:
My uptime was nearly 46h but then I got an interrupt. I don't know the issue because of lacking uart logging. The good point is the the ESP automatically reconnected and has a new uptime of more than 9h. (Version 0.3.6)
I will think about your notice to check how the memory is orgnized / used and where pointers are used. Let us improve all the points we get and the uptime will increase from days to weeks - someday I have to build a second device for not interrupting my long term test.

stefan123t, 2022-05-07:
Wegen der Liste der WLANs in der Umgebung (auch praktisch zur Vorbelegung der Auswahl auf der Setup Seite !) kannst Du Dir gerne mal die [wifi.h](https://github.com/grindylow/ahoy/blob/main/tools/NRF24_SendRcv/wifi.h#L40-L52) aus dem tools/NRF24_SendRcv ansehen. 
Da war das bereits drin um m.W. die Top5 erreichbaren WiFi APs anzuzeigen.

Könnten wir die UART Ausgabe auch irgendwie per remote logging (WLAN) ausgeben/schicken ? 
Manchmal wäre es auch schön bei der Taube auf dem Dach mitzulesen und nicht nur den Spatz in der Hand zu debuggen =^D

Vielleicht kann man das ja per WebSerial verwirklichen. 
Schau Dir doch bitte mal https://randomnerdtutorials.com/esp8266-nodemcu-webserial-library/ an, wie schwierig es wäre das auszutauschen ? 
Die Bibliothek verwendet auch ESPAsyncWebServer und ESPAsyncTCP, vielleicht bekommen wir ja damit sogar das WDT / Reset Problem in den Griff ?

rosch99, 2022-05-12:
Die Stabilität hat sich mit der 0.3.8 bei mir verbessert, trotzdem muss ich den ESP mehrmals am Tag stromlos machen und neu starten, da er irgendwo "hängenbleibt".

lumapu, 2022-05-12:
Bei mir - versorgt von einem USB Netzteil - bootet er einfach neu und ist erreichbar wie zuvor. An den 3.3V habe ich einen Elko und habe das RF24 Modul auf `HIGH` stehen.

stefan123t, 2022-05-17:
Habe gerade die 0.4.3 bei mir installiert und bekomme folgendes Logfile.

Dabei fällt auf:
1. das Programm macht 5 WiFi scan bzw. Connect Versuche mit dem Router
2. dann werden die Serien Nummern meiner (immer noch drei, aber uninteressant für dieses Issue) Wechselrichter Konfigurationen aktiviert
3. das hmRadio Modul wird initialisiert
4. der Access Point wird gestartet, obwohl die 30 Sekunden noch nicht vorbei sind
5. das Programm versucht weitere 5 WiFi scans bzw. Connect Versuche mit dem Router
6. Access Point steht, der 180 Sekunden Ticker wird gestartet, es finden vorerst keine weiteren WiFi scans bzw. Connect Versuche mit dem Router statt.
7. Erst nach 180 Sekunden werden die WiFi scans fortgesetzt
8. wieder nach 5 WiFi scans wird der AP gestartet
9. Nach weiteren 180 Sekunden wird der AP geschlossen
10. Diesmal klappt die WiFi Verbindung beim ersten scan.
11. Nach 10 Minuten klappt die ap_probe_send nicht und die WiFi Verbindung wird wieder geschlossen
12. das Programm verbleibt nun 30 Sekunden im WiFi scan modus und schafft dabei 13 Verbindungsversuche
13. wdt reset das Programm startet neu
14. WiFi Verbindung klappt auf Anhieb
15. hmRadio Modul wird initialisiert
16. dennoch wird der AP aufgemacht
17. nach 180 Sekunden wieder geschlossen
18. das Programm versuch 25 Sekunden lang schafft aber nur 2 (!) Versuche ?
19. AP wird wieder aufgemacht und nach 180 Sekunden geschlossen
20. interessant ist hier die Zeit mit den `AP will be closed in XXX seconds` vergeht linear. Davor aber wenn der AP aufgemacht wird bleibt die Zeit irgendwie fast 20 Sekunden (von `Active for: 180 seconds` bis zum ersten `AP will be closed in XXX seconds`)  stehen ?
21. Danach wieder WiFi scan mit Verbindung beim ersten Versuch
22. regelmäßige und erfolgreiche Verbindung `ap_probe_send_start`

<details>
  <summary>Serial Monitor Log - Click to expand!</summary>

```
22:29:30.733 -> SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
22:29:30.733 -> fpm close 1 
22:29:30.733 -> mode : sta(2c:3a:e8:12:34:56)
22:29:30.733 -> add if0
22:29:32.748 -> connect to network 'YOUR_WIFI_SSID' ...
22:29:32.847 -> ........scandone
22:29:33.579 -> no YOUR_WIFI_SSID found, reconnect after 1s
22:29:33.646 -> .reconnect
22:29:33.746 -> ......................
22:29:35.939 -> ......scandone
22:29:36.505 -> no YOUR_WIFI_SSID found, reconnect after 1s
22:29:36.539 -> .reconnect
22:29:36.638 -> .............................scandone
22:29:39.464 -> no YOUR_WIFI_SSID found, reconnect after 1s
22:29:39.531 -> .reconnect
22:29:39.630 -> ............................scandone
22:29:42.389 -> no YOUR_WIFI_SSID found, reconnect after 1s
22:29:42.455 -> .reconnect
22:29:42.555 -> ............................scandone
22:29:45.346 -> no YOUR_WIFI_SSID found, reconnect after 1s
22:29:45.346 -> .reconnect
22:29:45.446 -> ....scandone
22:29:45.845 -> .
22:29:46.841 -> SERIAL: 11 41
22:29:46.841 -> add inverter: HM-600, SN: 114173123456
22:29:46.841 -> SERIAL: 11 41
22:29:46.841 -> add inverter: HM-700, SN: 114173123456
22:29:46.841 -> SERIAL: 11 41
22:29:46.841 -> add inverter: HM-800, SN: 114173123456
22:29:46.874 -> RF24 Amp Pwr: RF24_PA_LOW
22:29:46.874 -> Radio Config:
22:29:46.874 -> SPI Frequency		= 1 Mhz
22:29:46.874 -> Channel			= 3 (~ 2403 MHz)
22:29:46.874 -> RF Data Rate		= 250 KBPS
22:29:46.874 -> RF Power Amplifier	= PA_LOW
22:29:46.874 -> RF Low Noise Amplifier	= Enabled
22:29:46.874 -> CRC Length		= Disabled
22:29:46.874 -> Address Length		= 5 bytes
22:29:46.874 -> Static Payload Length	= 32 bytes
22:29:46.874 -> Auto Retry Delay	= 250 microseconds
[... 437 lines cut ...]
23:02:03.184 -> bcn_timout,ap_probe_send_start
23:03:19.971 -> bcn_timout,ap_probe_send_start
23:03:32.287 -> bcn_timout,ap_probe_send_start
23:03:59.397 -> bcn_timout,ap_probe_send_start
23:04:15.063 -> bcn_timout,ap_probe_send_start
23:04:36.799 -> bcn_timout,ap_probe_send_start
23:04:47.317 -> bcn_timout,ap_probe_send_start
23:05:02.882 -> bcn_timout,ap_probe_send_start
23:05:17.754 -> bcn_timout,ap_probe_send_start
23:05:36.870 -> bcn_timout,ap_probe_send_start
23:05:48.154 -> bcn_timout,ap_probe_send_start
23:06:46.803 -> bcn_timout,ap_probe_send_start
23:07:03.227 -> bcn_timout,ap_probe_send_start
23:07:18.359 -> bcn_timout,ap_probe_send_start
23:07:33.720 -> bcn_timout,ap_probe_send_start
23:07:51.543 -> bcn_timout,ap_probe_send_start
23:08:07.744 -> bcn_timout,ap_probe_send_start
23:08:35.293 -> bcn_timout,ap_probe_send_start
23:08:48.801 -> bcn_timout,ap_probe_send_start
23:09:04.765 -> bcn_timout,ap_probe_send_start
23:09:22.057 -> bcn_timout,ap_probe_send_start
23:09:39.250 -> bcn_timout,ap_probe_send_start
23:10:31.779 -> bcn_timout,ap_probe_send_start
```
</details>

stefan123t, 2022-05-17:
@lumapu sollen wir analog zu `AP will be closed in XXX seconds` auch eine Meldung für `STA will be scanned for XXX seconds` einführen, dann kann man die beiden Ticker dafür parallel beobachten und ggf. Überschneidungen einfacher feststellen ?
Eventuell wäre das auch für die anderen Ticker (MQTT, etc.) sinnvoll und beim Debuggen hilfreich.

stefan123t, 2022-05-17:
Hallo Lukas,

während ich das Problem mit den WiFi connections näher analysiert und oben beschrieben habe, hat sich der nächste `wdt reset` ereignet.

```
23:26:42.840 -> bcn_timout,ap_probe_send_start
23:26:42.840 -> mac 985
23:26:50.042 -> 
23:26:50.042 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
23:26:50.042 -> 
23:26:50.042 -> wdt reset
23:26:50.042 -> load 0x4010f000, len 3460, room 16 
23:26:50.075 -> tail 4
23:26:50.075 -> chksum 0xcc
23:26:50.075 -> load 0x3fff20b8, len 40, room 4 
23:26:50.075 -> tail 4
23:26:50.075 -> chksum 0xc9
23:26:50.075 -> csum 0xc9
23:26:50.075 -> v00059810
23:26:50.075 -> ~ld
23:26:50.141 -> 
23:26:50.141 -> SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
23:26:50.141 -> fpm close 1 
23:26:50.141 -> mode : sta(2c:3a:e8:12:34:56)
23:26:50.141 -> add if0
```

Kannst Du mal meine SDK Angaben vergleichen ?

SDK:2.2.2-dev(38a443e)
Core:3.0.2=30002000
lwIP:STABLE-2_1_2_RELEASE
glue:1.2-48-g7421258
BearSSL:6105635

stefan123t, 2022-05-17:
The boot messages and modes are explained in the board overview:
https://arduino-esp8266.readthedocs.io/en/latest/boards.html#boot-messages-and-modes
Decoding stack trace is possible with
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

Found this old issue / thread about wdt reset and wrong Pin number assignments should be using e.g. D4, etc. instead of 2 and using some capacitors on external pins
github.com/esp8266/Arduino/issues/1809

rst cause:4, boot mode:(3,6) often happens when you define the pins incorrectly. For example, I was using a Wemos D1 Mini and specified constants to define pins 6 and 7, whereas I should have put in D6 and D7. EDIT: oops, noted above this has already been mentioned.

Works after changing PROGMEM const int l=7,r=6,f=5,b=4,h=3; to PROGMEM const int l=D4,r=D3,f=D2,b=D1,h=D0;

Reason(in my case):
pins RST(reset), EN(enable) and VCC were facing fluctuations in power supplied irrespective of the constant power supply. This was causing a rst of no. 4, i.e. Hardware WDT reset.

Solution:
Add a 47uF capacitor at each of the pin mentioned wrt GND.

stefan123t, 2022-05-29:
Join und Leave messages werden jetzt (positiv getestet in 0.4.13) berücksichtigt.

Dies führt dazu, daß man mit dem AP verbunden der Timer alle 10 Sekunden wieder auf den ursprünglichen Wert 3*60 bzw. 180 Sekunden zurückgesetzt wird. M.E. genügt hier ggf. auch ein Default von 60 Sekunden, wenn man sich vom AP abmeldet muß man sonst 3 Minuten warten, bis er sich wieder mit dem Router verbindet, aber das kann ich ja in der config.h setzen.

stefan123t, 2022-06-02:
@lumapu danke für den https://github.com/grindylow/ahoy/commit/e7d0e4ae66b632b742d1615e0b42383cd81ed3bd das sollte m.E. sowohl den o.g. Interrupt Handler stark vereinfacht und optimiert, als auch das Problem mit dem MQTT Connect gelöst haben.

stefan123t, 2022-06-03:
Mein ESP8266 lief jetzt mit der 0.4.15 seit 14:38 Stunden ohne WDT Reset, das ist doch schon mal was!
Und der Reset gerade eben war wieder in der o.g. Transmit/Receive Endlos-Schleife nach Sonnenuntergang:
```
22:16:19.456 -> Error while retrieving data: last frame missing: Request Retransmit
22:16:19.456 -> Transmit 11 | 15 12 34 56 78 78 56 34 12 83 AD 
22:16:29.086 -> 
22:16:29.086 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
22:16:29.086 -> 
22:16:29.086 -> wdt reset
22:16:29.086 -> load 0x4010f000, len 3460, room 16 
22:16:29.167 -> tail 4
22:16:29.167 -> chksum 0xcc
22:16:29.167 -> load 0x3fff20b8, len 40, room 4 
22:16:29.167 -> tail 4
22:16:29.167 -> chksum 0xc9
22:16:29.167 -> csum 0xc9
22:16:29.167 -> v0005a5d0
22:16:29.167 -> ~ld
22:16:29.167 -> Main::Main
22:16:29.167 -> app::app():Main
22:16:29.167 -> hmRadio.h : HmRadio():mNrf24(CE_PIN: 2, CS_PIN: 15, SPI_SPEED: 1000000)
22:16:29.167 -> app::setup
22:16:29.167 -> Main::setup
22:16:29.167 -> Main::getConfig
22:16:29.312 -> main.h:checkEEpCrc
22:16:29.312 -> main.h:buildEEpCrc
22:16:29.312 -> main.h:checkEEpCrc
22:16:29.312 -> main.h:buildEEpCrc
22:16:29.312 -> Main::setupStation
22:16:31.202 -> connect to network 'YOUR_SSID' ...
```

Andererseits sehe ich wie gesagt einige der Antwort-Pakete bis zu 4 Mal (siehe oben)
Entweder der Code ist jetzt dank optimiertem Interrupt Handler so schnell, dass er mehrere Responses des WR auf dem selben Kanal empfängt, oder es existiert evtl. doch noch eine Race Condition im Interrupt Handler ?

Von den HackRF Aufnahmen meine ich keine Mehrfach Pakete vom Wechselrichter gesehen zu haben.
Das könnte ja evtl. einer der Spezialisten mit der entsprechenden Funkhardware nachprüfen.

lumapu, 2022-06-03:
In #53 diskutieren wir im Prinzip das gleiche Thema, ich denke es liegt an dem setRetries der NRF24 Lib.

stefan123t, 2022-06-05:
Hallo Lukas,

ich dachte in #53 geht es um die fehlenden Pakete (mW bei HM-1000/1200/1500 fünf statt vier Pakete für den HM-600/700/800).

Hier in #15 und #52 geht es um was anderes nämlich dass die WiFi Verbindung teilweise nicht mehr verfügbar und der ESP nicht mehr erreichbar ist. Das liegt mE daran dass er aus der Schleife die die drei/vier/fünf Pakete für eine Payload zusammenbekommen soll nicht mehr rauskommt.
Aktuell ist die RC Routine vor allen anderen priorisiert. Es sollte also dafür gesorgt werden, dass nach einem Empfangszyklus (nach dem TX-Zyklus) auch die anderen „Threads“ abgearbeitet werden. Sonst kommt es zwar ncht mehr zum WDT Reset mehr, da der ESP ja nicht mehr im Interrupt Handler feststeckt, aber die anderen Funktionen wie WiFi, NTP, MQTT, etc können bei Nacht eben auch nicht mehr ausgeführt werden.
Der Reset gestern war vermutlich durch irgendwelche TCP Pakete die nicht abgerufen wurden hervorgerufen. Ich habe nämlich mehrfach erfolglos versucht per HTTP den ESP zu erreichen. Ping ging noch aber der WebServer war nicht zu einer Antwort fähig.
Heute Nacht lief er durch, war bloss nicht zu erreichen (WebServer) und seit heute mirgen 5:15 Uhr geht das wieder. Er empfängt nämlich wieder Daten vom Wechselrichter und kann somit nach der TX und RX Routine auch die Requests beantworten. Uptime ist jetzt bei 1 Tag und 3:40h ohne Reset / WDT ! Das ist bei mir neuer Rekord!

Das Problem mit den Paketen gehört dann evtl doch in das andere Issue #53 (?) aber jetzt habe ich die Logs hier schon oberhalb eingestellt.

Das erklärt zumindest warum ich manche Pakete mehrfach sehe:

```
mNrf24.setRetries(3, 15);
// 3*250us and 15 loops -> 11.25ms
```

void | setRetries (uint8_t delay, uint8_t count)
-- | --
  | Set the number and delay of retries upon failed submit

void 	[setRetries](https://maniacbug.github.io/RF24/classRF24.html#a4c6d3959c8320e64568395f4ef507aef) (uint8_t delay, uint8_t count)
 	Set the number and delay of retries upon failed submit

Es sind also immer 15 Versuche, von denen bei mir mindestens 1-4 Requests aktuell vom WR empfangen *und* auch beantwortet werden bzw. deren Antwort ich auch empfange.

Wenn man wie bei mir zuviele Antworten bekommt, kann man den retry_count evtl. ja wieder sukzessive -1 bis auf min. 1 erniedrigen.
Vielleicht sollte man den Wert für diesen retry_count aber auch sukzessive um +1 anheben, bis man wieder Antworten empfängt oder die max 15 retry_count erreicht sind.

lumapu, 2022-06-06:
Ich denke wir sollten es so lassen wie es ist. Ich habe während der Entwicklung der gesamten Payload gesehen, dass es sehr stark schwankt. Manche frames kommen 1x und andere 4x. Komisch ist, dass man tatsächlich maximal 4 sieht, das sollten wir noch erklären können. Meine erste Vermutung ist, das das Intervall zwischen den Retries zu kurz ist und der WR gar nicht so schnell antworten kann.

stefan123t, 2022-06-07:
Hallo Lukas (@lumapu)
ja, den 2. Punkt können wir mE aufschieben bzw. unbeachtet lassen.

Aber wie sieht es mit dem 1. Punkt aus ?
Wir sollten mE:
* für die Pakete der Reihenfolge nach Retransmits anfordern, nicht mit dem letzten Paket beginnen.
* wenn N Retransmit Anfragen (N = ca. 3-5 ?) fehlgeschlagen sind sollte die komplette Payload abgebrochen und verworfen werden und der nächste WR abgefragt werden, bzw. auch mal Zeit für die WebServer, MQTT und/oder NTP Tasks je nach Bedarf, Priorität und Notwendigkeit eingeplant werden.
* Am Besten könnte man auch nach einem TX/RX Fenster, das ja naturgemäß sehr zeitkritisch ist und direkt auf den TX folgen muss, eine Zeitscheibe für die anderen og Tasks (WebServer, MQTT, NTP) einplanen. Dann erst erfolgt die Retransmit Anfrage und das horchen auf Antworten für diese RX Pakete.
Bis entweder N Retransmits für eines der Pakete 1..M erfolglos waren oder die Payload komplett ist. Auch das Dekodieren der Payload kann wieder nach einer kleinen Lunstpause für WebServer, etc erfolgen da ja die Daten im Pufferspeicher des ESP (CircularBuffer) bereits abgelegt sind.

stefan123t, 2022-06-07:
Ach ja, wie Jan-Jonas bereits angemerkt hat sind 3 Retransmits eines Paketes 3 TX Versuche mit je 15 Retries, also 3x16=48 Pakete on-air. Die Zeiten können wir evtl. berechnen bzw. auf den HackRF Screenshots im Forum ablesen. Dort sind auch 15 Retries zu erkennen, daher ist das vermutlich in Ordnung und wir können doppelte Antwortpakete einfach silently ignorieren.

AsZork, 2022-06-30:
Uptime 3 Tage 02:04:23

AsZork, 2022-06-30:
AHoy-ESP V0.4.19
Uptime: 3 Tage, 02:07:02
Time: 2022-06-30 10:41:10  (Akt. Ortszeit 13:45 also ca 1 Stunde pro Tag verloren)

Ich aktualisiere die Werte von meinen 2 Wechselrichtern nur alle 30 Sekunden. 
(Durch diese Ändrung konnte ich diese lange Laufzeit erreichen) T

Alle anderen Eckdaten sind eigentlich nicht förderlich.
-CS=D8 CE=D3 IRQ=D4   
-mqtt aktiv mit 60 Sekunden
-Wemos D1 V3 und nrf-Modul mit externer Antenne ohne Abschirmung. 
-Entfernung WLAN 5 Meter durch 2 Außenmauern.
-Entfernung Wechselrichter 3 M, bzw 15 m durch eine Außenmauer. 
-Netzteil altes Handy-Netzteil mit 250mA.
-Kein Kondensator verbaut.

stefan123t, 2022-07-06:
+1 für einen täglichen NTP Refresh Job.

lumapu, 2022-07-10:
habe immer noch mehrere Reboots am Tag, daher hat mich die fehlerhafte Zeit noch nie gestört. 
Wie wollen wir das Triggern? Mit einem zusätzlichen Timeout, der alle 24h triggert?
+1 bin aber trotzdem für dieses Feature

stefan123t, 2022-07-10:
@lumapu Ja timeout zum NTP time refreshen finde ich gut.
Mal sehen wieviele Stunden wir mit einem Timer hinbekommen. 
Es wäre bestimmt auch nicht verkehrt das alle 6 Stunden zu aktualisieren, wenn sonst die millis mehrmals überlaufen, müssten wir den Überlauf sonst auch noch zählen. 
Machst Du einen extra Feature Request auf ?

lumapu, 2022-07-11:
nein, ich glaube dass ist eine kleine Sache, dann schließen wir diesen alten Issue - wir haben noch genug andere issues, die sich um die Stabilität kümmern

stefan123t, 2022-07-11:
@lumapu, hier ist halt die meiste Dokumentation zum Thema WDT resets und vor allem auch zu den evtl. geplanten State Machine vorhanden, oder haben wir das anderweitig (z.B. #78) hinreichend beschrieben ?
@andreas5232 sind die WiFi Verbindungsabbrüche bei Dir jetzt behoben, wenn ja seit welcher Version ?

stefan123t, 2022-08-04:
Lukas wie machen wir das in Zukunft wenn auch noch Device Control Commands (Active Power Limit) und weitere Device Info Commands (AlarmData und AlarmUpdate) dazukommen. Wir müssen uns etwas zum Schedulen überlegen. Sollten wir da so wie Thomas B arbeiten mit den EVERY_* Timer Makros ?
Auch blicke ich nicht ganz durch bei den Aufgaben von main.cpp, app.cpp und esp8266.cpp. Das sollten wir mal refaktorisieren. 
Ist es u.a. nötig dass alle WebSeiten showXxx Handler über app.cpp und main.cpp verteilt sind. Es ist vielleicht sogar egal ob der WebServer nur einmal läuft aber sowohl über STA als auch AP mode WiFi Stack erreichbar ist.
Wenn man die Debug Meldungen in der platformio.ini aktiviert dann sieht man dass er zwei Interfaces if0 und if1 aufmacht.
Anscheinend kann er die sogar parallel offen haben/lassen.

stefan123t, 2022-08-12:
@andreas5232 bitte mit der v0.5.9 überprüfen und ggf. als gelöst schließen. Danke
Das größere Thema mit dem Scheduler und einer Command Queue sollte im Rahmen von #78 gelöst werden.

andreas5232, 2022-08-12:
Sorry, dass ich hier zwischenzeitlich ausgestiegen bin und auf das letzte Mention nicht reagiert habe.
Vielen vielen Dank für eure tolle Arbeit! :+1: 
Aus meiner Sicht ist das Issue gelöst.

closed 2022-08-12 by andreas5232