lumapu/ahoy #19: esp8266 (lumapu) Watchdog Resets

dad401, 2022-04-29:
Ich hatte gestern die V0.2.9 länger laufen lassen und irgendwann hat mehrmals der Watchdog zugeschlagen:

`19:24:51.744 -> HM-400/ch0/U_AC: 234.900 V
19:24:51.744 -> HM-400/ch0/Freq: 49.990 Hz
19:24:51.744 -> HM-400/ch0/P_AC: 34.400 W
19:24:51.744 -> HM-400/ch0/I_AC: 0.150 A
19:24:51.744 -> HM-400/ch0/Temp: 18.200 °
19:24:56.456 -> 
19:24:56.456 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
19:24:56.456 -> 
19:24:56.456 -> Soft WDT reset
19:24:56.456 -> 
19:24:56.456 -> >>>stack>>>
19:24:56.456 -> 
19:24:56.456 -> ctx: sys
19:24:56.456 -> sp: 3fffecd0 end: 3fffffb0 offset: 01a0
19:24:56.456 -> 3fffee70:  40000f3d 00000023 00000000 4024d518  
19:24:56.456 -> 3fffee80:  00000009 40236dbd 60000200 40100bdc  
19:24:56.456 -> 3fffee90:  40007424 3fffeee0 3ffe8070 3ffe96c4  
19:24:56.456 -> 3fffeea0:  3ffe96b0 00000000 4bc6a7f0 40224419  
19:24:56.456 -> 3fffeeb0:  00000010 4023c215 3ffe8070 4021c32c  
19:24:56.489 -> 3fffeec0:  40236813 4023ae2e 3ffe99a0 3ffe8070  
19:24:56.489 -> 3fffeed0:  00000064 60000600 3ffe8070 4021c3a5  
19:24:56.489 -> 3fffeee0:  00000000 4023af36 4023af27 4021c427  
19:24:56.489 -> 3fffeef0:  00000000 40240c9b 3ffeed60 00000001  
19:24:56.489 -> 3fffef00:  40240cf8 3ffe99a4 3ffeed60 479d5488  
19:24:56.489 -> 3fffef10:  40105b91 16610177 3ffeed60 60000600  
19:24:56.489 -> 3fffef20:  40245c79 3ffe99a4 3ffeed60 479d3f32  
19:24:56.522 -> 3fffef30:  40245cbe 3fffdab0 00000000 3fffdcb0  
19:24:56.522 -> 3fffef40:  3ffeed80 3fffdad0 3ffef4fc 4021025a  
19:24:56.522 -> 3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
19:24:56.522 -> 3fffef60:  40000e19 00057825 00000000 00000005  
19:24:56.522 -> 3fffef70:  3fffefc0 aa55aa55 000000ed 40105539  
19:24:56.522 -> 3fffef80:  4010553f 00000000 00000005 40100b8c  
19:24:56.522 -> 3fffef90:  4010000d fe8e7600 00057825 401000ac  
19:24:56.522 -> 3fffefa0:  00000000 3fffef3c 00000000 3ffffea8  
19:24:56.555 -> 3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
19:24:56.555 -> 3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.555 -> 3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.555 -> 3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.555 -> 3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.555 -> 3ffff000:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.555 -> 3ffff010:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.588 -> 3ffff020:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.588 -> 3ffff030:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.588 -> 3ffff040:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.588 -> 3ffff050:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.588 -> 3ffff060:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.588 -> 3ffff070:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.588 -> 3ffff080:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.622 -> 3ffff090:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.622 -> 3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.622 -> 3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.622 -> 3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.622 -> 3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.622 -> 3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.622 -> 3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.655 -> 3ffff100:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.655 -> 3ffff110:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.655 -> 3ffff120:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.655 -> 3ffff130:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.655 -> 3ffff140:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.655 -> 3ffff150:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.655 -> 3ffff160:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.688 -> 3ffff170:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.688 -> 3ffff180:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.688 -> 3ffff190:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.688 -> 3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.688 -> 3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
19:24:56.688 -> 3ffff1c0:  feefeffe feefeffe feefeffe feefeffe  
[... 540 lines cut ...]
20:25:15.337 -> <<<stack<<<
20:25:15.337 -> 
20:25:15.337 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
20:25:15.370 -> 
20:25:15.370 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:25:15.370 -> 
20:25:15.370 -> load 0x4010f000, len 3460, room 16 
20:25:15.370 -> tail 4
20:25:15.370 -> chksum 0xcc
20:25:15.370 -> load 0x3fff20b8, len 40, room 4 
20:25:15.370 -> tail 4
20:25:15.370 -> chksum 0xc9
20:25:15.370 -> csum 0xc9
20:25:15.370 -> v00057830
20:25:15.404 -> ~ld
20:25:20.444 -> wait for network
20:25:20.544 -> ...................................................
20:25:25.651 -> ....................................................................................................
20:25:36.564 -> 
20:25:36.564 -> ---------
20:25:36.564 -> AP MODE
20:25:36.564 -> SSDI: ESP AHOY
20:25:36.564 -> PWD: esp_8266
20:25:36.564 -> ---------
`

lumapu, 2022-04-29:
duplicate #15 
fix: https://github.com/lumapu/ahoy/commit/b3bdf2811ec36edd0fbba435d313d3b12b968a40

lumapu, 2022-04-29:
übersetzter Stacktrace:
`
Decoding stack results
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100b8c: umm_init() at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 476
0x401000ac: app_entry_redefinable() at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 325
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x4021e57c: pbuf_try_get_at at core/pbuf.c line 1392
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x4021907d: __hexnan at /workdir/repo/newlib/newlib/libc/stdlib/gdtoa-hexnan.c line 145
0x4021993a: __b2d at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 691
0x40217fc3: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c line 498
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x4021993a: __b2d at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 691
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40214c2b: _printf_float at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_float.c line 205
0x40218655: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c line 808
0x40219c0c: __any_on at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 1035
0x4021493d: isspace at /workdir/repo/newlib/newlib/libc/ctype/isspace.c line 49
0x40219c0c: __any_on at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 1035
0x40215481: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 163
0x40219c0c: __any_on at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 1035
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x402155b0: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 209
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x402098a6: WiFiServer::available(unsigned char*) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiServer.cpp line 115
0x40207bf0: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 274
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40207bd4: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 274
0x40207d62: esp8266webserver::ESP8266WebServerTemplate ::begin() at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 372
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100348: delayMicroseconds(unsigned int) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 209
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40211efd: cleanupFunctional(void*) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\FunctionalInterrupt.cpp line 32
0x4020c1d3: PubSubClient::loop() at C:\Users\usr\Documents\Arduino\libraries\PubSubClient\src\PubSubClient.cpp line 373
0x40203cf6: app::loop() at C:\github\ahoy\tools\esp8266/app.cpp line 160
0x401001c8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x401001e9: esp_schedule() at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 128
0x4021041e: init_done() at C:\Users\usr\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 265
`

stefan123t, 2022-05-06:
@dad401 hier ist der Commit für die 0.2.9 bei dem der o.a. Stacktrace mutmaßlich aufgetreten ist:
https://github.com/lumapu/ahoy/commit/a9b7fd05999147f2853de43fc75cfccbc1fb709a

M.E. ist er außer in app::loop (app.cpp Zeile 160) auch in PubSubClient::loop und dort beim cleanupFunctional, etc.

Mir ist nicht ganz klar wie das esp8266webserver::ESP8266WebServerTemplate auf den Stack kommt, aber mein [StackTrace in #15](https://github.com/grindylow/ahoy/issues/15#issuecomment-1115118899) hat auch den umm_init() Aufruf in umm_malloc.cpp relativ weit oben auf dem Stack. Ich habe daher die Vermutung daß evtl. die String Klasse im Constructor / Deconstructor sehr viel (oder zu viel?) Zeit verbringt. Bei den Templates müssen ja mehrere String Objekte angelegt werden und dann die {VARIABLEN} ausgetauscht werden. Dabei werden i.d.R. mehrere Instanzen erzeugt, die danach wieder aus dem Speicher entfernt werden müssen. Da die String Objekte (=ESP8266WebServerTemplates) relativ groß sind, dürfte das auch einige Zeit in Anspruch nehmen

Vielleicht versucht er auch WebServer und PubSubClient gleichzeitig zu handlen und schafft es daher nicht.

stefan123t, 2022-05-16:
Ich habe mir mal angesehen wie es denn andere Projekte (hier [Tasmota](https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota.ino#L581-L590)) machen.

```
void SleepDelay(uint32_t mseconds) {
  if (!TasmotaGlobal.backlog_nodelay && mseconds) {
    uint32_t wait = millis() + mseconds;
    while (!TimeReached(wait) && !Serial.available() && !TasmotaGlobal.skip_sleep) {  
      // We need to service serial buffer ASAP as otherwise we get uart buffer overrun
      delay(1);
    }
  } else {
    delay(0);
  }
}
```

closed 2022-05-17 by lumapu