lumapu/ahoy #20: esp8266: no packets received

petit-miner, 2022-04-30:
Hallo,
erstmal vielen Dank für das tolle Projekt und den Code!
Habe hier einen ESP8266 mit einem nrf24l01+ PA verbunden und habe das Problem das ich scheinbar keine Pakete erhalte.
Die Seriennummer habe ich von der Rückseite des Umrichters abgeschrieben, die Nummer habe ich mehrmals überprüft.
Initial hatte ich den ESP falsch mit dem NRF Modul verbunden, welches ich durch die Fehlermeldung auf der Indexseite sehen konnte. Dies konnte ich bereits beheben. Gibt es einen DebugModus oder andere Möglichkeiten zu testen warum ich keine Messwerte erhalte? 
Vielen Dank im voraus!
Gruß Marcel

petit-miner, 2022-04-30:
Der IRQ Pin lag nicht an einem Pin der Interruptfähig ist. Nach dem ich den Fehler behoben habe läuft auch alles.

closed 2022-04-30 by petit-miner

lumapu, 2022-05-01:
Super, dass du es beheben konntest. Was mich auch freut, dass die Fehlermeldung auf der Indexseite zur Problembehebung beitragen konnte.

stefan123t, 2022-05-06:
Hallo Lukas (@lumapu) vielleicht sollten wir die Auswahl der Interrupt PINs in der Setup Seite auf die Interrupt-fähigen PINs einschränken.
Marcel (@petit-miner), welcher PIN war denn bei Dir der Falsche, bzw. welchen ESP8266 verwendest Du und wie war er verschaltet/verdrahtet ?

stefan123t, 2022-05-07:
Ich habe folgende [Anmerkung](https://randomnerdtutorials.com/interrupts-timers-esp8266-arduino-ide-nodemcu/) hierzu gefunden:
> The ESP8266 supports interrupts in any GPIO, except GPIO16.

crbble, 2022-07-18:
Moin, da ich über das gleiche Problem gestoßen bin, auch beim Arduino MEGA 2560 sind nur die Pins 2, 3, 18, 19, 20, 21 interrupt fähig.