lumapu/ahoy #28: Wie OTA Update von 0.3.3 auf aktuelle Version?

Wolf6660, 2022-05-13:
Hallo,

erstmal SUPER Arbeit DANKE. Habe die Version 0.3.3 installiert. Wie kann ich das Update auf die aktuelle Version machen? Welche Dateien muss ich uploaden. Konnte hierzu keine Anleitung finden. 

Danke

stefan123t, 2022-05-13:
Hi @Wolf6660 
für das OTA Update müssten wir vermutlich auch Binary Releases in github bauen, die man dann direkt per WLAN hochladen kann.
Aktuell ist das noch nicht in github umgesetzt, so dass die CI/CD Pipeline das automatisch erledigt.

Evtl. kannst Du die Binaries auch mit der Arduino IDE selber generieren.
Unter `File > Preferences > Settings` schaltet man mit `Show verbose output during [x] compilation` zusätzliche Infos zum Compiler Lauf ein.

Dann steht nach dem Verifizieren / Upload des Sketches auch der Pfad unter dem das Binary File generiert wurde:
Bei mir unter Linux liegt das dann unter `/tmp/arduino_build_<pid>/esp8266.ino.elf`

Unter der Ahoy Homepage gibt es unten einen Link `Update Firmware`, der verweist bei mir auf http://192.168.178.XX/update hier kann man dann Firmware und FileSystem separat auswählen und aktualisieren.
Das Filesystem sind die Teile des ESP8266 Flash-Memory die dafür unter `Tools > Flash Size` ausgewählt wurden.
Man kann die Firmware also auch ohne Filesystem bauen und dann die Daten im FS bestehen lassen.
Ich vermute das ist der Bereich in dem aktuell auch die Konfiguration des WiFi SSID / Password und der Wechselrichter Seriennummern abgelegt sind.

DanielR92, 2022-05-17:
Hallo zusammen,

bei einem OTA Update werden alle gespeicherte Daten überschrieben/gelöscht. 
Ich hab nicht im Source Code geschaut, aber werden die Daten nicht im ROM gesichert und wieder abgerufen?

Nicht das nach dem Flashen beim ersten Start alles gecleared wird. :)

lumapu, 2022-05-17:
@DanielR92 nein, die werden nicht gelöscht. Ich persönlich habe mir allerdings die Seriennumer nochmal gesondert gespeichert, damit ich im Notfall noch dran komme.
Problematisch wird es immer, wenn wir neue Speicherstellen einführen - da kann es zu Verlusten kommen. Wir haben noch kein professionelles Speichermanagement, aber wir haben ja auch noch eine sehr frühe Version (0.4.1)

closed 2022-05-17 by lumapu