lumapu/ahoy #33: esp8266 keine Pakete mit HM1500

TomMeyer90, 2022-05-15:
hallo.

Ich bekomme keine Pakete gesendet mit der esp8266 v0.3.6.

mit dem code NRF24_SendRcv kann ich Daten empfangen

TomMeyer90, 2022-05-15:
17:04:28.042 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
17:04:28.088 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
17:04:28.135 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
17:04:33.070 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4

Bekomme das im Seriellen Monitor

stefan123t, 2022-05-15:
Hallo Tom,
das sieht danach aus als wenn die config mit der WR Serial No. noch nicht stimmt oder er einfach anderweitig keinen Empfang hat. Bitte probiere es doch nochmal entweder mit der python software oder dem ESP8266 Code v0.3.9. Lukas arbeitet aktuell an einer neuen Version, die auch die Pakete zu einer Payload ordentlich zusammensetzt. Das dürfte dann wohl v0.4.0 werden ... mal sehen wann er die commitet ?
Ggf. kann man auch die RF24_PA_LOW auf RF24_PA_MAX stellen, sollte dann aber auf ausreichende Stromzufuhr (Netzteil) achten bzw. ggf. sogar einen 100uF Elko (Polung!) zwischen +3.3V und GND des nRF24L01+ Moduls klemmen um evtl. auftretende Stromabfall beim Senden vorzubeugen.

TomMeyer90, 2022-05-16:
Hallo Stefan,

die v0.3.9 habe ich auch schon probiert, selbes Ergebnis.
LOW bis MAX und Kondensator habe ich auch alles durch, sogar 1 m vom WR entfernt passiert nichts.

Dann werde ich mal auf die neue Version warten.
Nur komisch das bei paar Nutzern es reibungslos klappt mit dem HM1500.

stefan123t, 2022-05-16:
Hallo Tom,
habe gerade meinen ESP8266 neu gestartet und bekomme ebenfalls die o.g. "Fehlermeldung".

```
06:51:08.945 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
06:51:08.990 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
```

Es bedeutet wohl dass er keine Antwort vom Wechselrichter empfängt. Bei mir sind die Paneele noch nicht sauber ausgerichtet / installiert und es dauert eine ganze Weile bis der HM-600 genug Strom / Energie bekommt. Solange die LED am WR noch nicht grün/rot blinkt kann ich ihn auch nicht erreichen. Das ist auch der Grund warum die Entwicklung sich meist auf die Tageslichtstunden beschränkt.

TomMeyer90, 2022-05-16:
dann lass ich den mal laufen, vielleicht kommt ja noch was. Aktuell bekomme ich gerade 60W rein.

Hatte es die letzten Tage auch getestet und mit 600W Einspeisung kam auch nichts an.

Wenn du einen Neustart des ESP´s durchführst kommen bei dir gleich werte nach  oder dauert das paar Minuten? (Unterm Tag bei voller Leistung)

stefan123t, 2022-05-16:
Er verbindet sich bei mir nicht immer direkt mit dem WLAN (Dachgeschoss und Router im Keller), daher spannt er meist erst den AP auf und hat dann so seine Probleme beim Neustart. Die Meldung mit `-> missing: N` kommt aber definitiv aus dem hmRadio, d.h. hier wird auf jeden Fall versucht an den WR zu senden.

Nach ca. einer Stunde hatte er dann heute morgen auch Daten bekommen, wie gesagt nicht immer, und vor allem nicht immer alle Pakete, aber es wird langsam besser:

Manche haben auch davon berichtet, dass es je nach Ausrichtung der Antenne bzw. nach Belastung der WLAN Frequenzen zu unterschiedlichen Empfangserfolgen kommt. Da sowohl BLE / nRF24 als auch WLAN auf dem selben 2,4 GHz Band funken.

Hier mal die Daten aus dem Serial Log von meinem HM-600, wie gesagt noch nicht voll und korrekt angeschlossen, daher nur geringe Werte:

```
07:59:41.984 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
07:59:42.017 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
07:59:50.944 -> HM-600/ch0/YieldTotal: 0.327 kWh
07:59:50.944 -> HM-600/ch0/U_AC: 0.200 V
07:59:50.944 -> HM-600/ch0/Temp: 16.700 °
08:00:00.966 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:00:00.966 -> HM-600/ch0/U_AC: 0.200 V
08:00:00.966 -> HM-600/ch0/Temp: 16.700 °
08:00:10.955 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:00:10.955 -> HM-600/ch0/U_AC: 0.200 V
08:00:10.955 -> HM-600/ch0/Temp: 16.700 °
08:00:11.951 -> 0 (0000) 5 (2003) 3 (0030) 0 (0000)  -> missing: 2
08:00:11.984 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
08:00:12.017 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
08:00:20.943 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:00:20.943 -> HM-600/ch0/U_AC: 0.200 V
08:00:20.943 -> HM-600/ch0/Temp: 16.700 °
08:00:30.965 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:00:30.965 -> HM-600/ch0/U_AC: 0.200 V
08:00:30.965 -> HM-600/ch0/Temp: 16.700 °
08:00:41.020 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:00:41.020 -> HM-600/ch0/U_AC: 0.200 V
08:00:41.020 -> HM-600/ch0/Temp: 16.700 °
08:00:41.950 -> 0 (0000) 0 (0000) 3 (0003) 0 (0000)  -> missing: 3
08:00:41.983 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
08:00:42.016 -> 0 (0000) 0 (0000) 0 (0000) 0 (0000)  -> missing: 4
08:00:50.943 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:00:50.943 -> HM-600/ch0/U_AC: 0.200 V
08:00:50.943 -> HM-600/ch0/Temp: 16.800 °
08:01:00.965 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:01:00.965 -> HM-600/ch0/U_AC: 0.200 V
08:01:00.965 -> HM-600/ch0/Temp: 16.800 °
08:01:10.954 -> HM-600/ch0/YieldTotal: 0.327 kWh
08:01:10.954 -> HM-600/ch0/U_AC: 0.200 V
08:01:10.954 -> HM-600/ch0/Temp: 16.800 °
```

TomMeyer90, 2022-05-16:
Endlich, es funktioniert einwandfrei.
Wahrscheinlich wars das channel hopping. Hatte ich immer auskommentiert gehabt.
Also am besten nichts configurieren, alles so lassen wie es ist.

Empfang ist gut mit dem NRF24L01+PA+LNA, ca. 20 meter  durch eine Wand und in die Holzhütte.

14:44:38.930 -> Hoymiles/ch1/I_DC: 5.270 A
14:44:38.930 -> Hoymiles/ch1/P_DC: 136.000 W
14:44:38.930 -> Hoymiles/ch1/YieldDay: 578.000 W
14:44:38.930 -> Hoymiles/ch1/YieldTotal: 33.937 kW
14:44:38.930 -> Hoymiles/ch3/U_DC: 26.100 V
14:44:38.930 -> Hoymiles/ch2/I_DC: 5.530 A
14:44:38.930 -> Hoymiles/ch2/P_DC: 142.500 W
14:44:38.930 -> Hoymiles/ch2/YieldDay: 604.000 W
14:44:38.930 -> Hoymiles/ch2/YieldTotal: 35.650 kW
14:44:38.930 -> Hoymiles/ch3/I_DC: 4.330 A
14:44:38.930 -> Hoymiles/ch3/P_DC: 113.000 W
14:44:38.930 -> Hoymiles/ch3/YieldDay: 445.000 W
14:44:38.930 -> Hoymiles/ch3/YieldTotal: 23.091 kW
14:44:38.977 -> Hoymiles/ch4/I_DC: 4.300 A
14:44:38.977 -> Hoymiles/ch4/P_DC: 112.100 W
14:44:38.977 -> Hoymiles/ch4/YieldDay: 447.000 W
14:44:38.977 -> Hoymiles/ch4/YieldTotal: 23.740 kW
14:44:38.977 -> Hoymiles/ch0/U_AC: 237.900 V
14:44:38.977 -> Hoymiles/ch0/I_AC: 2.010 A
14:44:38.977 -> Hoymiles/ch0/P_AC: 478.500 W
14:44:38.977 -> Hoymiles/ch0/Freq: 49.990 Hz
14:44:38.977 -> Hoymiles/ch0/Pct: 99.900 %
14:44:38.977 -> Hoymiles/ch0/Temp: 46.300 °
14:44:38.977 -> Hoymiles/ch0/YieldDay: 2074.000 
14:44:38.977 -> Hoymiles/ch0/YieldTotal: 116.418 k
14:44:38.977 -> Hoymiles/ch2/U_DC: 25.800 V
14:44:38.977 -> Hoymiles/ch4/U_DC: 26.100 V

closed 2022-05-16 by TomMeyer90

lumapu, 2022-05-16:
ich habe heute festgestellt, das es sehr stark von den 3,3V abhängt. Erst nach der externen Versorgung mit 5V zu 3,3V DCDC habe ich Pakete empfangen können. Davor nur sehr selten oder spärlich, wenn ich über den FTDI Adapter versorge