lumapu/ahoy #35: ESP8266: Feature Request

stefan123t, 2022-05-17:
Hallo Lukas,

kannst Du den Feature Request noch aufnehmen, d.h. die Daten mit  Transmit und Received per Serial (ggf. auch WebSerial oder MQTT o.a.) im "Unified Format" mitzuloggen, damit man die Daten auch mit dem Python Code nachverarbeiten könnte ?
Es gibt zwar ein paar Zeilen, die bereits die `sent packet: #` und `RAW` / `CMD` im entsprechenden Format ausgeben.
Diese sind jedoch meist inaktiv und man muss die immer von Hand einschalten vor dem Kompilieren, damit man die Rohdaten sieht.

Transmit
https://github.com/grindylow/ahoy/blob/main/tools/esp8266/hmRadio.h#L239-L240
Received
https://github.com/grindylow/ahoy/blob/main/tools/esp8266/app.cpp#L142-L151

lumapu, 2022-05-17:
wurde in 0.4.0 eingebaut

closed 2022-05-17 by lumapu

stefan123t, 2022-05-17:
Another job well done! Thanks a lot for anticipating this in 4.0.0 already!