lumapu/ahoy #36: ESP8266: Discussion

stefan123t, 2022-05-17:
Hallo Lukas, et. al.

Bzgl. der blauen LED auf dem NodeMCU / Wemos D1 mini ist es evtl. keine so gute Idee gewesen den Anschluss D4 (GPIO2) für CE des nRF24 Moduls zu verwenden. Sobald CE low wird leuchtet die blaue LED. Das passiert wohl auch im Falle eines TX per Serial IO.

Soll ich die Fritzing Layouts nochmal anpassen und gibt es eine empfohlene Verdrahtung, bzw. sollten noch andere GPIOs evtl. nicht / 
anders verwendet werden ?
https://www.computerhilfen.de/info/esp8266-blaue-led-ausschalten-oder-blinken-lassen.html

Aktuell ist die Belegung in der getting started ESP8266 Dokumentation:
https://github.com/grindylow/ahoy/blob/main/doc/getting-started-ESP8266.md

Wire Connections
```
    +-----------+          +-----------+
    |  ESP8266  |--colour--| nRF24L01+ |
    |           |          |           |
    |       GND |---black--|[GND]      |
    |     +3.3V |----red---| VCC       |
    |        D4 |---grey---| CE        |
    |        D8 |--purple--| CSN       |
    |        D5 |---blue---| SCK       |
    |        D7 |---green--| MOSI      |
    |        D6 |---brown--| MISO      |
    |        D3 |--yellow--| IRQ       |
    +-----------+          +-----------+
```

@Jan-Jonas, ich habe auch schon Fritzing Layouts für den Raspberry Pi angelegt. Hier ist die IRQ Leitung nicht belegt. Offenbar unterstützt die RF24 Python Bibliothek für den RaspberryPi per pigpio auch IRQs aber das ist default nicht aktiv bzw. noch in Arbeit ?

Sollen wir das Getting Started auch von der Readme.md im Hauptarchiv verlinken, dann kommen evtl. weniger Fragen bzgl. Verkabelung ?

lumapu, 2022-05-17:
Ich bin da komplett offen, meine Hardware steht und ich bin damit zufrieden. Wahrscheinlich gibt es hier andere, die sich mehr um die Hardware kümmern, ich sehe mich eher in der Software Ecke.

Argafal, 2022-05-20:
I second this suggestion. I did not receive any communication from the inverter when using D4 and D3. As soon as I changed these two pins to D2 and D1 the communication started to work. I am using a Wemos D1 mini V3.

stefan123t, 2022-06-06:
@Argafal I have 0.4.15 with the improved and optimized IRQ/Interrupt Handler running since almost 2 days without resets.
I would therefor prefer to close this issue and leave documentation as it is.
Would you kindly double check your cabling with the latest version of the ESP code ?

Argafal, 2022-06-10:
Good to hear of the improvement! I am afraid I cannot change the physical cabling right now, since I cannot access the hardware at the moment. 

How is it for @TobiDD79, would you be able to check if the new code works for you with the original cabling (CE=D4, IRQ=D3)? It looks like we are using the same hardware and have the same problem.

stefan123t, 2022-06-10:
@Argafal I have a WeMos D1 mini Pro around but I can't seem to find the nRF24L01+ modules with external PA Antenna, which I bought for this purpose at the moment. So my current setup is using a Lolin NodeMCU v0.1 and its working fine. I did not even add a capacitor to the module nor did I shield it using aluminium foil.

@TobiDD79 wäre interessant zu wissen, ob die Original Schaltung bei Dir mit der 0.4.15/0.4.17 auch funktioniert ?
Ich konnte bisher keine Begründung finden, warum die nicht tun sollte.
Daher vermute ich es hängt eher mit den fehlerhaften Settings nach dem Erase EEPROM o.a. in der Zwischenzeit behobenen TX/RX Problemen zusammen.

@lumapu was ist denn der Default in der 0.4.17 ?
Ich habe das hier in der [defines.h](https://github.com/grindylow/ahoy/blob/main/tools/esp8266/defines.h#L6-L11) gefunden:
```
//-------------------------------------
// PINOUT (Default, can be changed in setup)
//-------------------------------------
#define RF24_CS_PIN         15
#define RF24_CE_PIN         2
#define RF24_IRQ_PIN        0
```

stefan123t, 2022-06-21:
Hier noch zwei Referenzen für die Anbindung der ESP8266 GPIOs generell

ESP8266 Pinout Reference: Which GPIO pins should you use?
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

Und einzelner Boards im Speziellen

ESP8266 und ESP8285 Module Anleitung - Module und Boards
http://stefanfrings.de/esp8266/

Ich muß das mal ordnen und mir genauer ansehen, welche PINs / GPIOs am sinnvollsten sind.

Speziell #79 weist auch auf das Problem hin, daß an D4 / GPIO2 sowohl die blaue LED als auch der Serial Port beim Booten hängt:
> Beim Booten serieller Ausgang, darf dabei nicht auf LOW gezogen werden. Kann nach dem Booten als normaler I/O Pin verwendet werden. Ist mit der blauen LED verbunden, die bei LOW Pegel leuchtet. Hat 12 kΩ Pull-Up Widerstand.

stefan123t, 2022-06-24:
Version 0.4.19 braucht wieder mal / immer noch die Basic Settings:

```
02:26:46.624 -> I: connect to network 'YOUR_WIFI_SSID' ...
02:26:46.723 -> ...............................
02:26:51.474 -> I: RF24 Amp Pwr: RF24_PA_MIN
02:26:51.474 -> I: Radio Config:
02:26:51.474 -> SPI Frequency		= 1 Mhz
02:26:51.474 -> Channel			= 0 (~ 2400 MHz)
02:26:51.474 -> RF Data Rate		= 1 MBPS
02:26:51.474 -> RF Power Amplifier	= PA_MIN
02:26:51.474 -> RF Low Noise Amplifier	= Disabled
02:26:51.474 -> CRC Length		= Disabled
02:26:51.474 -> Address Length		= 2 bytes
02:26:51.474 -> Static Payload Length	= 32 bytes
02:26:51.474 -> Auto Retry Delay	= 250 microseconds
02:26:51.507 -> Auto Retry Attempts	= 0 maximum
02:26:51.507 -> Packets lost on
02:26:51.507 ->     current channel	= 0
02:26:51.507 -> Retry attempts made for
02:26:51.507 ->     last transmission	= 0
02:26:51.507 -> Multicast		= Disabled
02:26:51.507 -> Custom ACK Payload	= Disabled
02:26:51.507 -> Dynamic Payloads	= Disabled
02:26:51.507 -> Auto Acknowledgment	= Disabled
02:26:51.507 -> Primary Mode		= TX
02:26:51.507 -> TX address		= 0x0000000000
02:26:51.507 -> pipe 0 (closed) bound	= 0x0000000000
02:26:51.507 -> pipe 1 (closed) bound	= 0x0000000000
02:26:51.540 -> pipe 2 (closed) bound	= 0x00
02:26:51.540 -> pipe 3 (closed) bound	= 0x00
02:26:51.540 -> pipe 4 (closed) bound	= 0x00
02:26:51.540 -> pipe 5 (closed) bound	= 0x00
02:26:51.540 -> W: WARNING! your NRF24 module can't be reached, check the wiring
02:26:51.540 -> I: 
02:26:51.540 -> 
02:26:51.540 -> ----------------------------------------
02:26:51.540 -> I: Welcome to AHOY!
02:26:51.540 -> I: 
02:26:51.540 -> point your browser to http://192.168.X.Y
02:26:51.540 -> I: to configure your device
02:26:51.540 -> I: ----------------------------------------
```

http://192.168.X.Y/setup

Setup

Pinout (Wemos)
CS: D8 (GPIO15)
CE: D4 (GPIO2)
IRQ: D3 (GPIO0)

Reboot device after successful save [x]
Save

lumapu, 2022-06-28:
> 
>     +-----------+          +-----------+
>     |  ESP8266  |--colour--| nRF24L01+ |
>     |           |          |           |
>     |       GND |---black--|[GND]      |
>     |     +3.3V |----red---| VCC       |
>     |        D3 |---grey---| CE        |
>     |        D8 |--purple--| CSN       |
>     |        D5 |---blue---| SCK       |
>     |        D7 |---green--| MOSI      |
>     |        D6 |---brown--| MISO      |
>     |        D4 |--yellow--| IRQ       |
>     +-----------+          +-----------+

mit dieser Belegung klappt es auch bei mir und war so frei es gleich als neuen Standard (Vorgabe nach Reset, kann individuell geändert werden) in die Firmware zu integrieren.

stefan123t, 2022-07-02:
Die Änderung in defines.h muss noch dokumentiert werden:

https://github.com/grindylow/ahoy/blob/main/tools/esp8266/defines.h#L11-L16

lumapu, 2022-07-04:
Ich weiß nicht, ob meine letzte Änderung hier zu mehr Verwirrung geführt hat. Ich hatte nur den Kommentar von @TobiDD79 gelesen und meinen Aufbau von einem ESP-07 auf einen Wemos-D1-mini umgebaut. Ich habe es gleich so verdrahtet und es hat auch bei mir auf Anhieb geklappt - während das alte Pinout nicht bei allen klappt.

stefan123t, 2022-07-04:
Hallo Lukas,
wir müssen entweder die ganzen Schematics anpassen und die Platinen von Holger u.a. "umlöten" / umkonfigurieren oder die beiden Zeilen in der defines.h wieder ändern...
Weiß auch nicht was weniger Irritationen und Aufwand verursacht. Ist halt so mit breaking changes / news 😃 
Bei mir läuft es auf dem NodeMCU mit der alten Konfiguration schon mehrere Tage lang, ich kann mich an ca. 5 Tage erinnern, bevor ich eine aktuellere Version geflasht habe und rebooten musste.

lumapu, 2022-07-04:
Ich würde ein revert auf die alte config vorschlagen. Das macht viel mehr Sinn, ich hatte nicht auf den Schirm, dass schon so viel davon abhängig ist.
Manchmal sind meine Scheuklappen aber auch nicht schlecht ;-P

Argafal, 2022-07-05:
Do they have to resolder? Isn't it sufficient to just update the configuration to the pins actually used? Whereas anyone who actually HAS problems with the default pin layout actually DOES have to physically change the connections. 

What I dislike about the old defaults is that they cause problems in a number of documented cases. This might make it hard for newcomers to get started: imagine you just put the hardware together, flash the software the first time, and now you don't get any response from the inverters. It will take a long time to identify that the cause of the problem is the default pin layout, and not anything YOU got accidentally wrong with the inverter serials, version of the RF module, etc.. I remember that I nearly gave up after three hours when trying it the first time, until I re-soldered from the default pins to an alternative pin layout and everything just magically worked all of a sudden. I feel that this initial hurdle could have been easily avoided with a different default. Am I wrong?

Alternative suggestion, could there at least be a sentence in the documentation? Something on the lines of "If you do not get any response from the inverters, try swapping pins D4 and D3 for D2 and D1, both physically and in the software configuration. Using pins D4 and D3 caused problems for some users in the past depending on their individual hardware choices."

lumapu, 2022-07-05:
Everyone has the ability to contribute even to the documentation ;-)

I think we should discuss here a little bit the pros and cons before changing anything. I soldered only once using the changed Pinout. It started working immediately so I decided to make it the new default.

CS - D8 (GPIO 15)
CE - D3 (GPIO 0)
IRQ - D4 (GPIO 2)

permissionBRICK, 2022-07-08:
> 

I had the same issue, switching to D1 CE and D2 IRQ fixed it. Also the default pinout in the graphic is CE D4, IRQ D3, but the default config in the code is switched CE D3, IRG D4

billy0xff, 2022-07-09:
Ich habe alle Pinouts durchprobiert, mit keiner ist die Kommunikation zum Inverter erfolgreich.

das zuletzt getestete Pinout: 
CS -> D8 (GPIO15)
CE -> D1 (GPIO5)
IRQ -> D2 (GPIO4)

Inverter am Netz angeschlossen und via Labornetzgerät am DC-Eingang versorgt -> grüne LED blinkt im 4-Sekunden Takt für "keine Kommunikation zum DTU".

Statistics:

Receive success: 0
Receive fail: 104
Frames received: 0
Send Cnt: 310

Inverter 'HM400_0' is not available and is not producing
MQTT: not connected

 
Serielle Konsole:

.....

I: [NTP]: 2022-07-09 14:23:21
I: Inverter #0 I: no Payload received! (retransmits: 0)
I: Requesting Inverter SN 1121xxxxxxxxx
I: Transmit 27 | 15 80 13 35 19 78 56 34 12 80 0B 00 62 C9 8F 59 00 00 00 05 00 00 00 00 99 67 AF
E: while retrieving data: last frame missing: Request Retransmit
I: Transmit 27 | 15 80 13 35 19 78 56 34 12 80 0B 00 62 C9 8F 59 00 00 00 05 00 00 00 00 99 67 AF
E: while retrieving data: last frame missing: Request Retransmit
I: Transmit 27 | 15 80 13 35 19 78 56 34 12 80 0B 00 62 C9 8F 59 00 00 00 05 00 00 00 00 99 67 AF
I: Inverter #0 I: no Payload received! (retransmits: 2)
.... etc.

Was könnte das noch sein?

stefan123t, 2022-07-10:
@billy0xff 
>  I: Requesting Inverter SN 1121xxxxxxxxx
> I: Transmit 27 | ... 80 13 35 19 ...
Deine Seriennummer dürfte also 112180133519 sein.
Es ist ungewöhnlich eine Seriennummer mit 80 am Anfang zu haben. 
Ich habe in unserer Liste bisher nur Seriennummern mit 112163..112174 für die 1121-Modelle.
Ich meine irgendwo im Sourcecode gelesen zu haben, daß die ersten Stellen der Seriennummer dem Baujahr entsprechen ?
Vielleicht hast Du auch eine neuere Firmware auf dem Wechselrichter ?
>  E: while retrieving data: last frame missing: Request Retransmit
Er bekommt keine Pakete vom Wechselrichter als Antwort übermittelt.
Stimmt das Präfix 1121 oder hat Dein HM-400 ein anderes Präfix ?

@permissionBRICK yes, we know that there is a mismatch between documentation and code right now (since v0.4.20-0.4.22 at least). Could you specify the Wemos D1 model you are using *and* could you retry with the D3/D4 connection but matching config in the settings ?

stefan123t, 2022-07-10:
> Are there different versions of Wemos D1 mini?

@lumapu yes there are at least three different models (mini, lite and pro) and several versions according to [Stefan Frings ](http://stefanfrings.de/esp8266/) page:

Das Wemos D1 Mini Board gibt es in folgenden Versionen:

Version | ESP Chip | Flash MByte | LED an GPIO2 | Antenne | Lithium Laderegler
-- | -- | -- | -- | -- | --
Wemos D1 Mini Lite v1 | ESP8285 | 1 | ja | Leiterschleife | nein
Wemos D1 Mini v2 | ESP-12 Modul (=ESP8266) | 4 | nein | Leiterschleife | nein
Wemos D1 Mini v3 | ESP8266 | 4 | ja | Leiterschleife | nein
Wemos D1 Mini Pro v1 (oder -16) | ESP8266 | 16 | ja | Keramikantenne und ext. Antennenanschluss | nein
Wemos D1 Mini Pro v2 | ESP8266 | 16 | ja | Keramikantenne und ext. Antennenanschluss | ja

> Die oberen vier Boards haben die gleichen Abmessungen, das letzte mit Laderegler ist etwas länger.
> Alle Versionen haben einen USB-UART (CH-340) und einem Low-Drop Spannungsregler. Alle freien I/O Pins des ESP
> Chips sind herausgeführt, aber nicht der CHIP_EN Pin.

permissionBRICK, 2022-07-10:
@stefan123t [stefan123t](https://github.com/stefan123t)
I did that originally, but no matter how I set the config it didnt work. 
[pic](https://i.imgur.com/fLDAlO9.jpeg)
its a v2... i think?

NDS234, 2022-07-10:
ESP8266 bei mir 4 MB Flash

stefan123t, 2022-07-10:
@permissionBRICK 
> I did that originally, but no matter how I set the config it didnt work. [pic](https://i.imgur.com/fLDAlO9.jpeg) its a v2... i think?

I can barely read the fine print on the chip picture. 
Does it say EPS8266 or ESP8285 ? 
Are you setting 4MB flash in the IDE and it compiles and flashes fine ?

permissionBRICK, 2022-07-10:
It says 8266, so according to the table above it should be 4mb?

stefan123t, 2022-07-10:
> Also the default pinout in the graphic is CE=D4, IRQ=D3, but the default config in the code is switched CE=D3, IRQ=D4
reverted the default settings so it matches the pinout diagrams for the time being.

laserrichi, 2022-07-11:
ich nutze seit eh und je ESPs mit mind. 4MB Flash und erachte das quasi als Standard und min. Meistens habe ich die D1 mini Pro mit 16Mb

billy0xff, 2022-07-11:
@stefan123t
ja, exakt das ist die s/n, der Inverter ist vom Mai, da hat ihn auch der Händler erst bekommen.
Präfix 1121 passt.

@permissionBRICK
I wired the communication module directly to a ESP12-F without any carrier board:

D1 mini		ESP12F		NRF24L01+  SMD

D1		GOIO5		CE	   3
D2		GPIO4		IRQ	   8
#D3		GPIO0		IRQ	   8
#D4		GOIO2		CE	   3
D5		GPIO14		SCK	   5
D6		GPIO12		MISO	   7
D7		GPIO13		MOSI	   6
D8		GPIO15		CS	   4
3V3		Vcc		Vcc	   1
G		GND		GND	   2

GPIO15	-> Pulldown ca. 10k
EN	-> Pullup ca. 10k
GPIO0	-> Pulldown ca. 10k (optional)

Flashen:

GPIO0 -> GND
RST   -> HI->LO->HI
... flashen 
RST   -> HI->LO->HI

billy0xff, 2022-07-17:
Hier sind alle Signale am ESP12F zum NRF, Verdrahtungsfehler sollte keiner vorliegen:
[image]
[image]

stefan123t, 2022-07-17:
Hallo @billy0xff anscheinend kommuniziert der ESP8266 ja mit Deinem NRF24 Chip was sich an den MISO/MOSI Daten auf Deinem SPI erkennen lässt. Vielleicht versuchst Du mal die TX/RX Channel wie von Ziyat T. vorgeschlagen anders einzustellen ? Offenbar haben manche Wechselrichter eine (temporäre?) Präferenz für bestimmte Sende/Empfangskanäle.

Da Du schon den ganzen Messaufbau mit Oszi/Logic Analyzer vorbereitet hast, könntest Du evtl. auch nochmal eine Messung mit D3/D4 unternehmen ? Da wir in diesem Issue der Anschluss-Kombination D3/D4 am Wemos D1 mini (Pro/Lite) und den Meldungen nach Problemen mit dieser Kombination nachgehen.
Herzlichen Dank im Vorraus.

billy0xff, 2022-07-17:
Hallo @stefan123t; kann ich gerne am kommenden WE machen; Vor der Umlöterei schaue ich mir noch die Kanäle laut deinem Tip an; sehr klar ist mir das nicht, wie das Channelhopping implementiert ist und ob es reicht, einfach alle 5 Kanäle durchzuprobieren. Übrigens, das Diagramm zeigt nur die Kommunikation zum NRF, der Inverter war da gar nicht in Betrieb; ich wüsste auch nicht wie man in dieser Dauerkommunikation eine erfolgte Antwort erkennen könnte bzw. auf was man triggern könnte.

stefan123t, 2022-07-17:
@billy0xff bisher werden bei der Initialisierung der NRF24 Bibliothek die retries mit 15 angegeben. D.h. er versucht alle Requests 16 mal zu schicken. Ich meine auf Seite 2/3 des Forum Threads erste HackRF Aufnahmen von einer DTU Lite o.a. gesehen zu haben bei der ebenfalls 15/16 Retries zu sehen waren. D.h. man sollte bei einigermaßen schnellem Channel Hopping die Responses auch sehen. In der Hoymiles DTU-PRO werden auch entsprechende LOCAL_TIMEOUTs gesetzt je nachdem welches Kommando bzw. welche Antwort erwartet wird. Das liegt m.W. so zwischen 1-2 Sekunden idR. Es gibt aber auch einzelne Kommandos ab 300 ms, das ist m.W das absolute Minimum eines Sende / Empfangsversuches.

stefan123t, 2022-08-07:
Peter S. schrieb im Beitrag [#7152099(https://www.mikrocontroller.net/topic/525778#7152099)]:
> Hi Danke für all Eure tolle Vorarbeit, ich stamme aus der Hardware Ecke 
> und mir ist da etwas aufgefallen:
> Irgendwann verabschieden sich meine beiden dauer-laufenden 
> Installationen gerne mal mit einem watchdog reset (Ursache noch 
> unbekannt) und bleiben hängen. Auch ein manueller Reset behebt dann den 
> Hanger nicht, erst nach einem Power Reset läuft alles wieder.
> Wenn ich in diesem Fall den Serial Output mit 74880 Bd angucke, dann 
> sehe ich einen falschen Bootloader Mode:
>  ets Jan  8 2013,rst cause:2, boot mode:(1,6)
> verursacht durch einen LOW Pegel auf GPIO0 = D3 = IRQ vom NRF-Modul. Man 
> sollte also den IRQ besser woanders hinlegen, meine ich.

Hallo Peter S.,
danke für den Hinweis, wir diskutieren das Problem schon seit geraumer Zeit hier.

Eigentlich war die ursprüngliche Intention die Pins D1/D2 für I2C oder andere zukünftige Anwendungen (ModBus485, etc.) freizuhalten.

Bisher hatte ich als Grund für die [WDT Resets](https://github.com/grindylow/ahoy/issues/24#issuecomment-1132654422) in den dekodierten Exceptions meist ein Problem in umm_malloc ausgemacht. Ich vermute das hängt mit der Nutzung der String-Implementierung im ESP8266 zusammen. 

<details><summary>Debug Output</summary>
<p>

```text
16:01:55.100 > Request: /setup
16:01:55.100 > Arguments: 
16:01:55.100 > final list of key/value pairs:
16:01:55.103 > [String] Reallocating large String(3933 -> 3946 bytes) '<!doctype  ... dy></html>'
16:01:55.116 > [String] Reallocating large String(3943 -> 4028 bytes) '<!doctype  ... dy></html>'
...
16:01:56.467 > [String] Reallocating large String(2036 -> 2067 bytes) '<label for ... value="16"'
16:01:56.477 > [String] Reallocating large String(7185 -> 9253 bytes) '<!doctype  ... dy></html>'
16:01:56.492 > [String] Reallocating large String(9253 -> 9377 bytes) '<!doctype  ... dy></html>'
```
</p>
</details>

Weiteren Debug Output dazu hier

Gibt es nicht eine einfache Möglichkeit den GPIO0 / D3 beim Booten auf High zu ziehen -> RC-Glied ?

stefan123t, 2022-08-12:
Das Problem wurde durch einen besseren Interrupt Handler weitgehend behoben.
Die Probleme beim Pinout / Verkabelung rühren i.d.R. nicht vom Pin D3/D4 her.
Ich schließe das Issue bis auf Weiteres.

Wer einen Fehler entdeckt, bitte mit Debug Logs wie oben und in https://github.com/grindylow/ahoy/pull/107#issuecomment-1211090337 beschrieben als neues Issue einstellen. Danke

closed 2022-08-12 by stefan123t

Sprinterfreak, 2022-10-17:
> @Sprinterfreak, ich habe auch schon Fritzing Layouts für den Raspberry Pi angelegt. Hier ist die IRQ Leitung nicht belegt. Offenbar unterstützt die RF24 Python Bibliothek für den RaspberryPi per pigpio auch IRQs aber das ist default https://github.com/nRF24/RF24/issues/818 bzw. noch in https://github.com/nRF24/RF24/pull/819 ?

Ich hab einen großen Rewrite von dem ganzen python Modul in der Pipe und gehe von RF24 weg, weil das einfach pain ist zum laufen zu bringen. Ich konzentriere mich da mehr auf Circuitpython's rf24l01. Kann man einfach via pip mit installieren. Werde die Entwicklung aber in ein eigenes Repo puschen weil das sonst nicht mehr maintainbar ist. Und Zeit brauche ich...

stefan123t, 2022-11-14:
Hallo Jan-Jonas, gib Bescheid wenn Du das Repo hochgeladen hast, dann setzen wir gerne einen Link auf Dein Repo.

Helmer2000, 2022-12-11:
> ich nutze einen wemos d1 mini und einem nrf modul mit externer antenne und habe mit obriger verkablung keinerlei rückmeldungem vom wr bekommen. nach tausch der pinbelegung von IRQ und CE bekomme ich nun endlich Datenpakete vom wr zurück.
> 
> ```
> +-----------+          +-----------+
> |  ESP8266  |--colour--| nRF24L01+ |
> |           |          |           |
> |       GND |---black--|[GND]      |
> |     +3.3V |----red---| VCC       |
> |        D3 |---grey---| CE        |
> |        D8 |--purple--| CSN       |
> |        D5 |---blue---| SCK       |
> |        D7 |---green--| MOSI      |
> |        D6 |---brown--| MISO      |
> |        D4 |--yellow--| IRQ       |
> +-----------+          +-----------+
> ```

Herzlichen Dank @TobiDD79 - Tauschen der Verdrahtung und Anpassung im Interface brachte auch bei mir den ersehnten Empfang .

martinriedel, 2023-01-19:
@DerSelbermacherBlog ich hatte das gleich Problem und habe es nun folgendermaßen gelöst:
Gelb und Grau getauscht, also Gelb auf GPIO2 und Grau auf GPIO0. Danach musste ich aber in den Einstellungen (System Config) die Pinbelung ändern -> CE auf D3 und IRQ auf D4. Danach kamen Daten an.

cibis8, 2023-02-21:
> Danach musste ich aber in den Einstellungen (System Config) die Pinbelung ändern -> CE auf D3 und IRQ auf D4

Bei mir werden auch keine Daten vom WR ausgelesen. Ich habe bereits die Kabel vertauscht "gelb" und "grau", es kommen jedoch weiterhin keine Daten vom WR an.
Wie komme ich an die System Config und wo finde ich diese Einstellungen zum Ändern der Pinbelegung?
Ich habe den Wemos Mini v3.0.0 ESP8266 mit 4MB und dem WLAN Modul nRF24L01+ mit ext. Antenne.
Vielen Dank...

martinriedel, 2023-02-21:
@cibis8 
Hat der wirklich USB-C? Sieht für mich nach Micro-USB aus. Im Dropdown unter System wäre die Frage ob unter Radio `is connected` oder `is not connected` steht. Wenn du aber die Micro-USB Variante hast musst du nichts umstellen

cibis8, 2023-02-21:
> Hat der wirklich USB-C? Sieht für mich nach Micro-USB aus. Im Dropdown unter System wäre die Frage ob unter Radio `is connected` oder `is not connected` steht. Wenn du aber die Micro-USB Variante hast musst du nichts umstellen

Vielen Dank für die Rückantwort.
Nein hat kein USB-C sondern Micro USB. Zugriff habe ich über das Webinterface. Unter "System" "Radio" steht: is conntected, ich habe aber nicht den nRF24L01+ mit der aufgeätzten Antenne, sondern einen mit einer aufgeschraubten etwa 8cm langen "Gummi-Antenne". Ich habe das so angeschlossen wie bei Ahoy gefordert (fest verlötet auch Farbcodierung soweit möglich beibehalten), dann hier im Forum gelesen, dass man D3 und D4 tauschen sollte bei ext. Antenne - das hat bei mir aber nix gebracht. Vermutlich muss man dann in dieser "Config" dann auch eine Anpassung vornehmen, da ich aber das erste mal mit so einem Arduino Projekt zu tun habe, weiss ich nicht wie ich das machen soll. Hast du da eventuell ein How-To-"Config" Link oder so was, wie ich da am besten dran komme? Bissl programmieren kann ich, aber dieser Arduino ist mein erster Gehversuch.

martinriedel, 2023-02-21:
@cibis8 mach mal ein extra Issue auf, ich will das Issue hier nicht zumüllen, dann können wir uns gezielter deinem Problem widmen ;)

cibis8, 2023-02-21:
> @cibis8 mach mal ein extra Issue auf, ich will das Issue hier nicht zumüllen, dann können wir uns gezielter deinem Problem widmen ;)

erledigt: https://github.com/lumapu/ahoy/issues/704

JuKi1994, 2023-02-22:
Hallo zusammen,

bei mir hatte die Verkabelung zuerst auch nicht funktioniert, ich habe den CE und IRQ Anschluss beliebig zwischen D1-D4 variiert und die System Config entsprechend angepasst...funktioniert hat es dann aber erst mit RX (GPIO3) und TX (GPIO1). 

```
+-----------+          +-----------+
|  ESP8266  |--colour--| nRF24L01+ |
|           |          |           |
|       GND |---black--|[GND]      |
|     +3.3V |----red---| VCC       |
|        RX |---grey---| CE        |
|        D8 |--purple--| CSN       |
|        D5 |---blue---| SCK       |
|        D7 |---green--| MOSI      |
|        D6 |---brown--| MISO      |
|        TX |--yellow--| IRQ       |
+-----------+          +-----------+
```

[image]

Vielleicht hilft das auch wem weiter...
VG,
Julian

rhobin242, 2023-02-28:
Hallo,
ich verwzeifele an der Kommunikation zw. WR und dem nFR-Modul. Ich hab mittlerweile (fast) alle hier beschriebenen Verdrahtungen und Konfigurationen ausprobiert. Aktuell läuft die von JuKi1994 oben beschriebene Verkabelung.
Leider kommen immer wieder folgende Fehlermeldungen (gilt übrigens auch für die Original-Verdrahtung):
*************************
13:02:52 I: (#0) sendTimePacket
13:02:52 I: sendTimePacket 5
13:02:52 I: TX 27B Ch75 | 15 81 81 22 07 85 66 28 95 80 05 00 63 fd ed 67 00 00 00 00 00 00 00 00 b1 09 47 
13:02:54 W: while retrieving data: last frame missing: Request Retransmit
13:02:54 I: (#0) sendTimePacket
13:02:54 I: sendTimePacket 5
13:02:54 I: TX 27B Ch3 | 15 81 81 22 07 85 66 28 95 80 05 00 63 fd ed 67 00 00 00 00 00 00 00 00 b1 09 47 
13:02:57 W: while retrieving data: last frame missing: Request Retransmit
13:02:57 I: (#0) sendTimePacket
13:02:57 I: sendTimePacket 5
13:02:57 I: TX 27B Ch23 | 15 81 81 22 07 85 66 28 95 80 05 00 63 fd ed 67 00 00 00 00 00 00 00 00 b1 09 47 
********************
Verbindung ist also vorhanden, das Tauschen von CE und IRQ funktioniert auch nicht. :(((

Wer weiß Rat?
Danke
Rho

martinriedel, 2023-02-28:
@rhobin242 Welchen ESP benutzt du?

rhobin242, 2023-02-28:
...den esp8266...

martinriedel, 2023-02-28:
Ok falsch ausgedrückt ;)
Den? https://funduinoshop.com/media/image/a7/ef/b0/Wemos-D1-Mini-Node-Mcu-LUA-WIFI-ESP-8266-Entwicklungsboard-01.png
Oder den? https://www.wemos.cc/en/latest/d1/d1_mini_3.1.0.html

rhobin242, 2023-02-28:
...den ersten...

Argafal, 2023-02-28:
Ich glaube, was @martinriedel versucht zu sagen: wir brauchen mehr Details. Beschreib doch bitte genau die Schritte, die du probiert hast. Nur wenn wir uns vorstellen können, wie bei dir die Hard- und die Software konfiguriert sind, haben wir eine Chance den Fehler zu sehen. Ansonsten ist das nur ein Ratespiel und davon wird keiner glücklich :)

martinriedel, 2023-02-28:
In dem Fall bleibt die Verdrahtung bei https://ahoydtu.de/img/fritzing/AhoyWemos_Steckplatine.jpg und in den Einstellungen bleibt es bei CS D8, CE D4 und IRQ D3. Amplifier Power Level ist LOW?
Weitere Fehlerquelle: Benutzt du ein Netzteil mit mindest 1.5A oder besser 2A?

Argafal, 2023-02-28:
Und den Elko nicht vergessen ;)

rhobin242, 2023-02-28:
@martinriedel: 
Ja, mit der Verkabelung bin ich ursprünglich angefangen, aber ich rüste gerne wieder darauf zurück. Power Level ist LOW! Der Elko kommt heute nachmittag (wenn Nachbar wieder daheim ist .-) ). Ich betreibe den ESP an einem USB-Port, da ich das WebUI von Ahoy verwende. 
An der Fehlermeldung stört mich erstmal gar nix, aber ich bekomme keine Daten vom WR...

@Argafal 
Sorry, Du hast natürlich Recht. Ich versuche das Ahoy-Projekt mit meiner neuen PV-Anlage (Hoymiles HM800, nrf24l01+, esp8266), gemäß dieser Anleitung https://ahoydtu.de/getting_started/  ans Laufen zu bringen. Im WebUI sehe ich, dass der INverter "connected" ist, aber auf der "LIVE"-Seite stehen alles nur Nullen - obwohl draussen so schön die Sonne scheint! :).
Brauchst Du sonst noch Infos??

rhobin242, 2023-02-28:
@martinriedel 
das mit dem USB-Port ist natürlich Blödsinn - wie mir gerade auffällt. Das Ding hängt ja im WLAN, werde gleich mal ein ordentliches Netzteil dranhängen

martinriedel, 2023-02-28:
Das könnte nämlich schon der ganze Trick sein 😉

rhobin242, 2023-02-28:
....keine Änderung - gleiche Fehlermeldungen und keine Daten vom WR...

martinriedel, 2023-02-28:
Unter `System `steht bei `Radio nrf24l01+` `is connected`?

martinriedel, 2023-02-28:
Kein Thema ;)

rhobin242, 2023-02-28:
@martinriedel 
Hallo Martin, leider hat sich nach etwa 2 Stunden die Kommunikation wieder verabschiedet :((
Ich habe jetzt erstmal die Verkabelung wieder auf die ursprüngliche Konfiguration (mit D3 und D4) zurückgelötet und auch den Kondensator eingelötet. Netzteil mit 2,4 A.
Alles bislang ohne Erfolg, der WR ist "connected", die gleichen Fehlermeldungen wie heute mittag schon. 
Hast Du noch eine Idee?

cibis8, 2023-03-01:
Hallo rhobin242,
ich hatte ähnliche Probleme wie du und hab mit diversen PIN Konfigurationen eine bei mir dauerhaft sehr gut funktionierende Lösung gefunden. CS auf D8, CE auf D4, IRQ auf D1 (nur auf D1 hatte ich Erfolg!). Entsprechende Anpassungen natürlich auch in der "Settings" - "System Config" durchgeführt. Es konnte mir keiner sagen, warum dies nur bei D1 funktioniert und nicht an einem andern PIN. Vielleicht klappt das bei dir ja auch. Netzteil habe ich ein Standard USB Netzeil 5V 1A. Den Elko (100uF 25V) habe ich nachträglich eingefügt. Aktuelle Entfernung von der Ahoy DTU ca 2-3m bei einem Testaufbau/Testbetrieb meiner Anlage. In Zukunft habe ich hier etwa 14-15m zu überbrücken.

cibis8, 2023-03-01:
Hast du denn einmal IRQ auf PIN D1 gelegt? ich wäre auch fast daran verzweifelt, dass gar nix übertragen wurde außer diese Fehlermeldungen. Ich vermute einmal das der IRQ an den anderen PINs Fehler verursacht.. warum auch immer (Störung des IRQs?!?) Nach dem ich den auf D1 legte läuft das Teil nun schon seit dem 20. Februar ohne Probleme. Scheinbar gibt es da doch Unterschiede bei den Baugruppen, da viele ja gar keine Probleme hatten oder haben.

rhobin242, 2023-03-02:
Kann ich gerne mal ausprobieren....
Im Moment ist die Situation so, dass etwa alle 30 Minuten die Daten einmalig korrekt übertragen werden und dann wieder nur diese Fehlermeldungen. Über das MQTT-Protokoll sehe ich tatsächlich, dass immer dann, wenn die Fehlermeldungen erscheinen, ein unvollständiger Satz an Daten übertragen wird. Also bspweise die Werte für Spannung und Leistung, aber kein A-Wert...

rhobin242, 2023-03-03:
@cibis8 
IRQ auf D1 hats auch nicht gebracht :(((
Dafür steht unter dem Menüpunkt "System"/Unterpunkt "Radio":
nrf24l01+ is connected
Datarate 250 kbps
Power Level HIGH
RX success 0
RX fail 0
RX no answer 16
RX frames received 0
TX count 96

Auffällig ist imho der "RX no answer"-Wert von 16 und die 0 bei der empfangnen Frames.
Ansonsten wie gestern: unregelmäßig, höchstens einmal pro Stunde werde Daten empfangen. Die Verbindung steht also es werden keine vollständigen Datensätze übertragen.
Noch jemand 'ne Idee?

[15 more replies]