lumapu/ahoy #38: HM1200 Define anpassen

lumapu, 2022-05-17:
@Sprinterfreak 

In folgender Zeile muss der Index `6` sein, der WR hat nur 2 DC-Spannungen, aber 4 DC-Ströme;
https://github.com/grindylow/ahoy/blob/6871bf4247e283c163f220d973862198c670fb1a/tools/rpi/hoymiles/decoders/__init__.py#L584

Zusätzlich gibt es noch den Powerfactor mit dem Index `56`, `2 Byte` und Teiler `10`

Sehr gut finde ich, dass jetzt die Definitionen sehr gut vergleichbar sind:
[Python](https://github.com/grindylow/ahoy/blob/main/tools/rpi/hoymiles/decoders/__init__.py)
[ESP](https://github.com/grindylow/ahoy/blob/main/tools/esp8266/hmDefines.h)

lumapu, 2022-05-17:
Mir ist gerade noch was aufgefallen, im Kommentar über den Definitionen steht die Anzahl der MPPT, der HM1200 und auch der HM1500 haben nur 2 und nicht 4.

closed 2022-05-19 by lumapu

Sprinterfreak, 2022-05-21:
@tbnobody

commit 8238e909 (2022-05-19): * fix #38 current assignment for 4-channel inverters
* added last received timestamp in /hoymiles livedata web page #47
* improved style.css
* improved NTP as described in #46