lumapu/ahoy #39: bugs 0.4.1

t0nym0n0, 2022-05-17:
Hallo Leute,

in der version 0.4.1. passt bei meinem HM600 der Wert YieldDay auf dem CH2 nicht mehr. Es wird nur 1.0 Wh angezeigt.
Ebenfalls wird bei meinem HM350 der CH2 im Webinterface angezeigt...

Gymnae, 2022-05-17:
Bei mir sendet YieldDay auf CH1 abwechselnd den Wert, der auch im Web interface gezeigt wird oder einen viel kleineren Wert, den ich im Web interface nicht sehe

lumapu, 2022-05-17:
ok, wie sieht euere Payload auf der Seriellen Konsole aus? Kommt da was?
@t0nym0n0 Ich glaube die Bytezuordnung ist falsch - das kann gut sein, da ich keinen der Inverter zum testen habe.
@Gymnae Das klingt komisch, auch hier wäre interessant was die Serielle Konsole spricht. Deine Seriennnummer fängt mit 1141 an?

AsZork, 2022-05-17:
Kleiner Zuordnungsfehler in

    { FLD_YD,  UNIT_WH,  CH2, 24, 2, 1    },
    { FLD_YD,  UNIT_KWH, CH2, 18, 4, 1000 },

closed 2022-05-17 by lumapu

Gymnae, 2022-05-17:
Works perfectly, thank you

commit 5acce759 (2022-05-17): * fix #39 Assignment 2-Channel inverters (HM-600, HM-700, HM-800)