lumapu/ahoy #43: bugs 0.4.2

t0nym0n0, 2022-05-17:
Hallo Lukas,
besten dank für den schnellen Fix, leider passen die Werte für den HM600 noch nicht.
Für den HM350 wird noch der CH2 angezeigt.

Bitte um kurze Anleitung wie per Remote auf die Konsole zugegriffen werden kann, über Putty bekomme ich keine Verbindung.

Vielen Dank!

anbei Bilder des Webinterfaces...

[image]
[image]

lumapu, 2022-05-17:
Putty auf serial, den richtigen COM Port wählen und die Baudrate auf 115200 stellen. Dann in Web Interface des ESP noch unter serial den RF24 debug einschalten.
Jetzt sollten Werte kommen

lumapu, 2022-05-17:
duplicate #41

t0nym0n0, 2022-05-17:
Ok, remote über netzwerk können die Daten nicht über putty ausgegeben werden?

lumapu, 2022-05-17:
noch nicht, am besten einen neuen Issue mit feature request oä öffnen

closed 2022-05-17 by lumapu

t0nym0n0, 2022-05-17:
in Tasmota kann die Konsole im Webinterface aufgerufen werden, vielleicht kann das übernehmen.

Hier mal ein Bild aus der Tasmotaconsole meines APSystems YC Wechselrichters.
Hier wurde Tasmota so angepasst das die Komunikation mit dem YC600 Wechselrichter möglich ist allerding arbeitet der Wechselrichter mit einem speziellen Zigbee Protokoll. 

[image]
[image]

commit a35cb414 (2022-05-17): * fixed #41 HM800 Yield total and Yield day were mixed around. Found issue while comparing to Python version
* fixed #43 HM350 channel 2 is displayed in Live-View
* added #42 YieldTotal and YieldTotal Day for HM600 - HM800 inverters