lumapu/ahoy #48: Version 0.4.4

RobertoBee, 2022-05-19:
0.3.6: Alles funktionierte: 
0.4.4: Original:Nix funktioniert.

HM 1500: 
Failed Payload: 35
Send Cnt: 114

Pinnen gut eingestellt.
Serielle Output: Nog nicht kontroliert

Danach Libraries geupdated (alle), nochmals kompiliert und ge-upload, und sehe, geht doch.

Vielleicht im code minimumversionen für die Libraries kontrolieren?

lumapu, 2022-05-19:
Gute Idee, allerdings habe ich sowas noch nie in Arduino Code gesehen, sondern nur bei PlatformIO.
Bei mir sind die libraries immer Recht aktuell - die IDE nervt ja sonst auch ständig diesbezüglich.

lumapu, 2022-05-24:
Habe die Versionen im Readme.md hinterlegt:
https://github.com/grindylow/ahoy/tree/main/tools/esp8266#readme