lumapu/ahoy #52: v4.8 WiFi loss connection and weird serial connection with ESP

DanielR92, 2022-05-23:
Besitze ein HM-1500 (4 Kanal). Wie bei [µController](https://www.mikrocontroller.net/topic/525778#7073670) geschrieben.

Habe meins nun auch geupdated 0.4.4 auf 4.8.
Aktuell habe ich damit nun immense WiFi probleme.

Ping wird ausgeführt für 192.168.10.149 mit 32 Bytes Daten:
Antwort von 192.168.10.149: Bytes=32 Zeit=46ms TTL=255
Antwort von 192.168.10.149: Bytes=32 Zeit=85ms TTL=255
Antwort von 192.168.10.149: Bytes=32 Zeit=99ms TTL=255
Antwort von 192.168.10.149: Bytes=32 Zeit=2ms TTL=255

Firefox (v100.0.2) lädt sich ins Nimmerleinstag. =(
ESP Serial Verbindung bleibt leer, irgendwann schmeist er mir paar logs 
entgegen.

lumapu, 2022-05-23:
gerne nochmal flashen und dann mal im Setup auf "Erase Settings (not Wifi)" drücken, evtl. ist es 'einfach' nur eine Einstellung z.B. MQTT

lumapu, 2022-05-23:
Ich denke ich habe den Fehler gefunden, wenn man für die MQTT Adresse was nicht vorhandenes eingibt, dann hängt bei mir der ESP auch, praktisch nicht mehr bedienbar.

DanielR92, 2022-05-23:
Richtig, ich nutze aktuell kein MQTT

lumapu, 2022-05-23:
hab jetzt mal eine 0.4.9 gepuht, bitte mal prüfen ob die besser geht, je nach Licht auch erst morgen ;-)

DanielR92, 2022-05-23:
Also ich habe das ganze am Labornetzteil dran. Meine PV werden noch geliefert. 
Aktuell geht weiterhin nichts, habe noch alle Konfigs neu gemacht. Keine Besserung.

homeautomation2022, 2022-05-23:
Ich kann auch sagen, dass ich ebenfalls kein MQTT nutze und frisch kompiliert und geflasht hatte(mit Erase All Einstellung), danach sind viele Felder mit FFs gefüllt, daher hatte ich als nächstes auch Erase Settings gemacht(aber kein Factory Reset). Zu dem Thema würde ich denken, dass bei der 4.8 keine IP drin war(Feld leer) und bei der 4.4 ist 0.0.0.0 drin.

Zumindest wie im Forum geschrieben, schmiert die Verbindung kurz nach dem der Wechselrichter erkannt wurde ab, muss also etwas damit zu tun haben. Das mit MQTT könnte aber natürlich eine zusätzliche Fehlerquelle gewesen sein.

HFarryranz, 2022-05-24:
Gerade getestet:  MQTT mit fehlerhaften Login-Daten getestet. (Leerzeichen hinten angefügt)
Der ESP versucht sich ca. alle 15-20 Sekunde am MQTT anzumelden und wird abgelehnt
WebIF so gut wie nicht nicht mehr erreichbar bzw. nach ca 50 Sekunden wird das WebIF auch nicht mehr aktualisiert.

Nach x Versuchen sollte die Nutzung zu MQTT deaktiviert  werden.
iM ist auch kein OTA mehr möglich.

Kann kann die seriellen Daten auf das WebIF umleiten?

rosch99, 2022-05-24:
0.4.9 läuft bei mir seit heute früh. Aus meiner Sicht deutlich stabiler als ALLE vorherigen Versionen. Stürzt zwar immer noch regelmässig einige Minuten nach der vollen Stunde(!) ab, aber zumindest restartet der ESP nach kurzer Zeit selbsttätig und published wieder per MQTT. Bei den früheren Versionen musste ich ihn dann oft mehrfach stromlos machen (WLAN-Steckdose) um ihn dann wieder zum Leben zu erwecken.

Danke für eure unermüdlichen Bemühungen zur Optimierung!!!
rosch

lumapu, 2022-05-24:
@HFarryranz Ja die Umleitung der Seriellen Daten steht noch auf der Wunschliste

@rosch99 schön, dass ich nicht der einzige bin, bei dem es läuft. @stefan123t hat schon Ideen wie wir das System stabiler machen können - ich hoffe in Zukunft mehr Schritte nach vorne zu machen als zurück

stefan123t, 2022-05-24:
Hallo @lumapu,
den Punkt von Daniel R. dass die Software sehr oft requests anfordert habe ich auch gesehen. Im Firefix/Chrome F12 Developer Tools > Network lädt er ca. jede Sekunde die Daten von /livedata. /time und anderen endpoint URLs neu. So schnell kann mein ESP meist nicht antworten und die Requests werden mE vom ESP8266WebServer gequeued.
Kann man das Verhalten im JavaScript der Seiten nicht nur hinauszögern (15/30 Selunden statt 1000/2000ms) und dabei irgendwie so gestalten dass es nicht so viele Aufrufe parallel startet ?
Nichtsdestotrotz habe ich auch mit der 0.4.5 Version viele Abstürze ca. 52 in 24 Stunden laut meinem Log. Die Stack Traces lade ich mal noch hoch, habe ich extrahiert und decoded.

stefan123t, 2022-05-25:
@lumapu eventuell liegt es ja an der "neuen" Payload Decodierung. Die sammelt ja alle einzelnen Pakete zusammen und versucht auch ggf. noch einen Retransmit, bevor sie die Payload überprüfen kann. 
Kannst Du dort einige `yield()`'s an für den Funkverkehr unkritischen Stellen einbauen, damit ggf. das WiFi Interface bedient werden kann ?

lumapu, 2022-05-25:
@turbolift sehr interessante Grafik. Da lohnt sich nochmal einen Diff der Versionen zu machen um evtl. das Problem einzugrenzen. Leider hatten wir noch nie eine stabile Version.

Ich habe noch andere ESPs am laufen mit anderer Firmware, aber auf der gleichen Basis als RGB LED Controller und in 3x Sonoff POW R2. Alle haben Laufzeiten von Wochen und nicht Stunden.

@stefan123t Der Retransmit läuft asynchron ab, ein `yield` kann ich einbauen, nur ist das für mich keine produktiv Lösung. Ein Delay in Code löst zwar manchmal das Problem, fällt einem später wieder auf die Füße.
Schon wäre es ohne solche Funktionen auszukommen.

stefan123t, 2022-05-25:
Habe bei mir mal folgendes in Main:loop() eingebaut um die Laufzeiten besser zu monitoren:

main.h:
```
class Main {
...
    protected:
...
        uint32_t mLastLoop;
```

main.cpp
```
Main::Main(void) {
...
    mLastLoop = 0;
...
void Main::loop(void) {
    if(millis() - mLastLoop > 1000) {
        DPRINTLN("Main::loop: " + String( millis()));
        mLastLoop = millis();
    }
...
```
Zusätzlich habe ich noch unter `Tools > Debug Level > HTTP_SERVER` aktiviert. Dabei scheint mir die Generierung der drei o.g. Seiten manchmal sehr lange zu brauchen, dabei wird die Serial Monitor Ausgabe auch relativ lange **blockiert**. Im Anschluß an einen Request wartet der ESP8266WebServer besonders lange (bevor das ACK Paket per TCP kommt) im Status `wait-close`:
```
06:32:05.384 -> http-server loop: conn=1 avail=0 status=wait-close
```

DanielR92, 2022-05-30:
Hallo zusammen,

die neuste Version 4.13 läuft nun ohne probleme bei mir. Danke. :) 👍

commit 01f1fab2 (2022-05-23): * try to fix #52
closed 2022-05-30 by DanielR92