lumapu/ahoy #56: Version 0.4.10 esp2866 stuck in boot loop

Gymnae, 2022-05-25:
Tried uploading the sketch via OTA and then directly connected to my machine. 
The eso2866 gets stuck in a boot loop. Prior versions were fine, other sketches load fine as well:

```
09:53:33.806 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
09:53:33.806 -> 
09:53:33.806 -> load 0x4010f000, len 3460, room 16 
09:53:33.844 -> tail 4
09:53:33.844 -> chksum 0xcc
09:53:33.844 -> load 0x3fff20b8, len 40, room 4 
09:53:33.844 -> tail 4
09:53:33.844 -> chksum 0xc9
09:53:33.844 -> csum 0xc9
09:53:33.844 -> v0005a050
09:53:33.844 -> ~ld
09:53:33.918 -> Settings not valid, erasing ...
09:53:33.918 -> erase: 0x71 - 0xb1
09:53:33.918 -> erase: 0xb1 - 0xf1
09:53:33.918 -> erase: 0xf1 - 0x131
09:53:33.918 -> erase: 0x131 - 0x171
09:53:33.918 -> erase: 0x171 - 0x192
09:53:33.994 -> ... restarting ...
09:53:34.209 -> 
09:53:34.209 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

lumapu, 2022-05-25:
ok I think I know the problem. Can you remove the erase function call inside main.cpp in setup()

Gymnae, 2022-05-25:
I'm sorry, I don't find what you want me to delete...

lumapu, 2022-05-25:
I'm sorry it is here:

https://github.com/grindylow/ahoy/blob/main/tools/esp8266/main.cpp#L128

Gymnae, 2022-05-27:
Ah yeah, that fixed it :) 
Now it looks like this:
```
    else {
        /*mApActive = true;
        memset(mStationSsid, 0, SSID_LEN);
        memset(mStationPwd, 0, PWD_LEN);
        memset(mDeviceName, 0, DEVNAME_LEN);

        // erase application settings except wifi settings */
        snprintf(mStationSsid, SSID_LEN,    "%s", FB_WIFI_SSID);
        snprintf(mStationPwd,  PWD_LEN,     "%s", FB_WIFI_PWD);
        snprintf(mDeviceName,  DEVNAME_LEN, "%s", DEF_DEVICE_NAME);
    }
```

stefan123t, 2022-05-27:
@lumapu upgraded to 0.4.10 tonight and also ran into this issue.

Please fix line 136 & 140 and not 128 (128 is inside a multiline comment already):
https://github.com/grindylow/ahoy/blob/main/tools/esp8266/main.cpp#L136-L140
Otherwise the reboot continues indefinetly, somehow the eraseSettings() does not fill in zeros or does not clear the EEPROM area used for settings so it will continue after a Reset / ESP.reset().

I removed some settings by hand and used the "Erase all settings" button on the setup page to get all empty or "0".

I had to fill in my serial number and name again and add the new setting for each panel / channel max Wpp.
Why can I not set these things in config.h or how would I do that ?

lumapu, 2022-05-29:
hopefully fixed in 0.4.11

closed 2022-05-29 by lumapu

stefan123t, 2022-05-29:
Still have this after flashing it again.
```
22:38:50.459 -> ... restarting ...
22:38:50.658 -> 
22:38:50.658 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
22:38:50.658 -> 
22:38:50.658 -> load 0x4010f000, len 3460, room 16 
22:38:50.658 -> tail 4
22:38:50.658 -> chksum 0xcc
22:38:50.658 -> load 0x3fff20b8, len 40, room 4 
22:38:50.658 -> tail 4
22:38:50.658 -> chksum 0xc9
22:38:50.658 -> csum 0xc9
22:38:50.658 -> v0005a820
22:38:50.658 -> ~ld
22:38:50.725 -> Settings not valid, erasing ...
22:38:50.725 -> erase: 0x71 - 0xb1
22:38:50.725 -> erase: 0xb1 - 0xf1
22:38:50.725 -> erase: 0xf1 - 0x131
22:38:50.725 -> erase: 0x131 - 0x171
22:38:50.725 -> erase: 0x171 - 0x192
22:38:50.824 -> ... restarting ...
```
unfortunately this is all rinse and repeat.
I will try again with an empty / original config.h ...

stefan123t, 2022-05-29:
Okay needed to call buildEEpCrc() after eraseSetting() or saveValues().
Fixed and can be closed.

stefan123t, 2022-05-29:
Hmm maybe still something wrong.
```
connect to network '⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮'
```
It now tries the empty / erased settings for its FB_WIFI_SSID and FB_WIFI_PWD.
Not exactly what we expected from setting this in config.h.
I will try to set it manually in /setup and save settings in EEprom.

Gymnae, 2022-05-29:
Version 0.4.13 works fine for me, updated via OTA

stefan123t, 2022-05-29:
Here are pictures after connecting to the Access Point.
[image]
[image]
I had to setup everything new, even the wiring.

This is probably only necessary after the EEprom data structure changed.
After forcibly resetting the ESP the Crc checksums seemed to match according to the Debug line.
```
CRC RD: 1c79 CRC CALC: 1c79
23:40:06.329 -> CRC RD: 2593 CRC CALC: 2593
```

Nevertheless, the Wiring, MQTT and SSID should be saved with sane defaults after eraseSettings.

stefan123t, 2022-05-29:
Die [sicheren Defaults in app.cpp](https://github.com/grindylow/ahoy/blob/247ae3e1db7c602dcea2548bf25b87d73c020b3b/tools/esp8266/app.cpp#L491-L504) werden offenbar nicht gesetzt sondern mit den "falschen" Defaults aus eraseSettings() und saveValues() ersetzt.

```
        html.replace(F("{MQTT_PORT}"), "1883");
```

lumapu, 2022-05-30:
ja der `else` Zweig kann jetzt nicht mehr erreicht werden, da die Settings jetzt immer valide sind (CRC wird beim erase aktualisiert). Hier brauchen wir ein neues Konzept um zu detektieren, dass frisch erased wurde.

commit 247ae3e1 (2022-05-29): fix #56