lumapu/ahoy #60: Installationsanleitung rpi

setje, 2022-05-27:
Hallo,

leider steige ich nicht durch die eigentliche Installation der Pakete. Die Entsprechenden Vorbereitungen habe ich schon durchgeführt, sprich Python und die Treiber sind schon installiert. Nur wie gehe ich nun mit den hier abgelegten Dateien um?
Ich wäre über eine Anleitung sehr dankbar! Danke :-)

grindylow, 2022-05-29:
Schau mal in die [README.md](https://github.com/grindylow/ahoy/blob/main/tools/rpi/README.md), dort ist ein Beispiel-Aufruf drin:

```$ sudo python3 -um hoymiles --log-transactions --verbose --config /home/dtu/ahoy.yml```

Sprinterfreak, 2022-05-30:
Hallo setje,
was grindylow schreibt. Das Kommando musst Du im Verzeichnis ahoy/tools/rpi/ ausführen. Wir sind noch nicht soweit, dass sich packaging wirklich lohnt, ist noch sehr WIP. Später wird es denke ich aber mal einen pip installer mit console_script geben, der Grundstein dafür ist gelegt.

Welcher Punkt in der Anleitung passt Dir denn nicht oder wo hängt es bei Dir?
Ich persönlich hatte lange mit dem NRF24 Treiber zu kämpfen bis ich rausgefunden habe, dass man Kommandos im richtigen cwd ausführen muss. (crosscompiler not found)
RF24 ist aber eigentlich nicht unsere Baustelle, drum in unserer Anleitung nur der Verweis auf die originale RF24-Anleitung.

rosch99, 2022-06-03:
Wie kann man das tool per cron automatisch beim reboot starten.
Ich bekomme es nicht hin, weil er beim Start das hoymiles module (-um hoymiles) nicht findet.
Kann man den Pfad irgendwie mitgeben, wie macht ih das?

Den "manuellen" Start aus der shell mache ich mit:
nohup python3 -um hoymiles --config /home/dtu/ahoy.yml > /dev/null 2>&1 &
Das muss allerdings auch im passenden Verzeichnis ausgeführt werden.    

rosch

Sprinterfreak, 2022-06-04:
```crontab
@reboot cd ahoy/tools/rpi && python3 -m hoymiles --config /home/dtu/ahoy.yml > /dev/null 2>&1 &
```

Sprinterfreak, 2022-06-07:
Alternativ ginge das auch per systemd

/etc/systemd/system/ahoy.service
```ini
[Unit]
Description=Hoymiles Inverter Gateway

[Service]
Type=simple
WorkingDirectory=/root/ahoy/tools/rpi
Restart=always
RestartSec=30
ExecStart=/usr/bin/python3 -m hoymiles --config /home/dtu/ahoy.yml --verbose --log-transactions

[Install]
WantedBy=multi-user.target
```

Quick usage
```bash
# Systemd's Config-Files neu einlesen
systemctl daemon-reload

# ahoy.service beim boot automatisch laden
systemctl enable ahoy.service

# ahoy.service jetzt starten
systemctl start ahoy.service

# Output ansehen
journalctl -fu ahoy.service
```

setje, 2022-06-21:
> Hallo setje, was grindylow schreibt. Das Kommando musst Du im Verzeichnis ahoy/tools/rpi/ ausführen. Wir sind noch nicht soweit, dass sich packaging wirklich lohnt, ist noch sehr WIP. Später wird es [...]

Danke! Mit der neuen Anleitung die zuletzt die letzten RF24 Probleme behoben hat, läuft nun alles!

DanielR92, 2022-06-22:
Kann dann geclosed werden denke ich? @grindylow @setje

Sprinterfreak, 2022-06-22:
Meinetwegen. Kann ich nicht machen.

setje, 2022-06-22:
@DanielR92 Von mir aus ja 👍

closed 2022-06-22 by setje