lumapu/ahoy #67: Version 0.4.17 MQTT topic

sude22, 2022-06-10:
@lumapu  Danke für die neue Version!
In Bezug auf [hier](https://github.com/grindylow/ahoy/issues/53#issuecomment-1151623400) erst mal ein dickes Lob. 
Mein Setup mit jetzt 4 HM-1500 empfängt die Daten sehr sauber. Fast bei jedem Durchlauf wird eine Payload erkannt. Top.
Damit ist diese Version definitiv für den Multi-Inverter Einsatz geeignet ;)

Eine Kleinigkeit ist mir aufgefallen:
app.cpp Zeile 134 snprintf(mqttTopic, MQTT_TOPIC_LEN, "inverter");
Das setzt das MQTT Topic fest auf "inverter", ohne die Zeile funktioniert es wie gehabt.

lumapu, 2022-06-10:
Sehr schön, das freut mich. Ich habe bei mir einfach einen zweiten Inverter 'erfunden' und eingetragen. Das Ergebnis deckt sich mit seinem Test: 614: fail stehen 613 success gegenüber.
Also trotz falschem / nicht vorhandenem Inverter wird der andere fehlerfrei abgefragt.

DanielR92, 2022-06-22:
Moin zusammen,
der Pull request wie ich sehe hat geklappt [https://github.com/grindylow/ahoy/pull/69](Pull).
Könnte einer von euch das issue hier bitte dann closen? :)

closed 2022-06-22 by sude22