lumapu/ahoy #68: Mqtt - keine Übertragung nach Nachtabschaltung 0.4.15

TobiDD79, 2022-06-10:
Nachdem bei nachtabschaltung die Übertragung der Mqtt Nachrichten eingestellt wird, werden diese auch am nächsten Tag bei Produktionsbeginn weiterhin nicht mehr übermittelt. 

Desweiteren reagiert der esp auf alle Anfragen in der weboberfläche sehr träge.

erst nach einem reboot sind die antwortzeiten auf der weboberfläche wieder normal und auch die Mqtt Übertragung funktioniert wieder.

lumapu, 2022-06-10:
das klingt interessant, ich hatte den Zustand wohl noch nie, da mein ESP nicht so stabil läuft. Ich hoffe dass bald nachvollziehen zu können.

mslookup, 2022-06-10:
hatte das problem mit der 0.4.15 auch, hab soeben die 0.4.17 draufgeschmissen - mal sehen ob sich was verändert hat - ich werde berichten.
@lumapu: wenn du noch ein NodeMCU Lolin V3 Module ESP8266 ESP-12F + nRF24L01+ zum entwickeln / testen brauchst, kann ich dir gern eines kostenfrei zusenden. pn im mikrocontroller board an mslookup mit adresse genügt ;)

TobiDD79, 2022-06-11:
Das Problem besteht auch mit der 0.4.17 Version. Der HM produziert am Morgen wieder und Mqtt steht nach der Nachtabschaltung auf Not connectet, nach Neustart wieder io

mslookup, 2022-06-11:
Hallo zusammen,
ich kann vielleicht mal eine temporäre Rückmeldung geben: Bei mir hat sich nach ersten Erkenntnissen das Problem mit 0.4.17 normalisiert. Ich habe zwar noch keine Langzeitdaten, aber nach einem Tag schaut bisher die Uptime gut aus. Auch die Nachtphase hat mein ESP8266 gut überstanden. Habe dazu mal nen Screenshot angehängt: Er "produziert" und Werte nach MQTT werden auch geschrieben. Ich werde weiter berichten...
V
[image]
G

lumapu, 2022-06-11:
@mslookup danke fürs Angebot, bin gut ausgestattet :-)

Freut mich zu hören, dass die Version so stabil läuft, ich habe leider immer wieder reboots und vermute, dass die Ursache Netzwerk Probleme sind. Heute um 16 Uhr ist mein ESP komplett abgestürzt und war nicht mehr erreichbar - ein Powercycle hat's gerichtet.

TobiDD79, 2022-06-13:
Bei mir läuft die mqtt Übertrag jetzt auch nach der Nachtabschaltung mit der 0.4.17 seit 2 Tagen

mslookup, 2022-06-15:
@TobiDD79: Läuft deines immernoch stabil? Bei mir komme ich einfach nicht mehr über 24 Stunden uptime, ohne dass irgendwas zu hängen anfängt.

TobiDD79, 2022-06-15:
Hab aktuell 1tag und 2h uptime, davor wie gesagt die 2 Tage uptime. Aber irgendwas bringt den esp nach wie vor immer mal zum hängen bzw. Man muss manuell neustarten

Oliver-ger, 2022-06-16:
Hier ein kleiner workaround, der aber funktioniert (erspart den manuellen Neustart)
Änderung der void reconnect in mqtt.h:
void reconnect(void) {
            //DPRINTLN(F("mqtt.h:reconnect"));
            int count = 0;
            String mqttId = "ESP-DTU";
            while(!mClient->connected()) {
                if((strlen(mUser) > 0) && (strlen(mPwd) > 0))
                  if (mClient->connect(mqttId.c_str(), mUser, mPwd)){
                    Serial.println("MQTT connected");
                  }
                  else {
                    Serial.println("MQTT failed");
                    delay(2000);
                    count = count + 1;
                  }
                  if (count >=3){
                    ESP.restart();
                  }
            }
        }

lumapu, 2022-06-16:
siehe auch PR #75 das dürfte auf den ersten Blick überstimmen

TobiDD79, 2022-06-17:
Bin aktuell bei 3 Tagen uptime mit der 0.4.17 und ohne Mqtt Ausfälle nach der Nacht, denk damit kann der bug als gelöst abgeschlossen werden

lumapu, 2022-06-17:
Beneidenswert. Wie weit bist du von WR entfernt und welches NRF24 Level hat du eingestellt?
Hast du ein NRF24 mit externen und internen Antenne?

TobiDD79, 2022-06-17:
Nrf Level steht auf min, esp ist nur nen Meter vom wr entfernt, muss nur durch ne Glasscheibe funken und das Nrf Modul hat eine externe Antenne

DanielR92, 2022-06-22:
Super!

@TobiDD79 könntest du dann auch dieses Issue closen? :)