lumapu/ahoy #74: rpi: High CPU usage with interval: 5

berni2288, 2022-06-14:
Hallo,

- Tool: rpi
- Chip: NRF24L01+
- Inverter: HM-800
- Platform: Raspberry PI 4
- interval: 5

I'm running the rpi tool with NRF24L01+ and HM-800 inverter.
You guys really did a great job and I'm forwarding a lot of data to my InfluxDB and displaying it in my Openhab.

However there is one thing I've noticed, the CPU usage is always quite high, around 12%. Well it's not ridiculous high, but still I wonder if this can be optimized somehow? Since all this tool does is sending and receiving data and forwarding it.

Danke! :+1:

lumapu, 2022-06-14:
how did you read the CPU usage? In general the ESP8266 will never become a battery powered MCU.
I think temperature isn't a problem here, the chip isn't noticeable hot.

More interesting than the CPU usage will be the view from a scheduler, how the functions are aligned together.

stefan123t, 2022-06-14:
@lumapu IMHO this is for the python version under Raspberry Pi 
@berni2288 did you check the pigpio settings (default off) of your nRF24 library are built in so you could connect the modules IRQ interrupt too ?
Other than that I would expect the python code to poll the nRF until something is received.

@Sprinterfreak may know some more details too?

> ich habe auch schon Fritzing Layouts für den Raspberry Pi angelegt. Hier ist die IRQ Leitung nicht belegt. Offenbar unterstützt die RF24 Python Bibliothek für den RaspberryPi per pigpio auch IRQs aber das ist default https://github.com/nRF24/RF24/issues/818 bzw. noch in https://github.com/nRF24/RF24/pull/819 ?

Sprinterfreak, 2022-06-15:
@stefan123t python does not use interrupts. Does not make sense because everything is synchronous in current transactions.
@berni2288 with an interval of 5 sek. for 5 inverters, You end up continously communicating. Each inverter transaction takes about 3-4 seconds at best. Limited by the ESB protocol. That should not be a problem tho. I guess You will end up around ~12 sec interval per inverter.
Great to hear that it obvously works with multi-inverters. Can't test that because I only have one.

I can confirm the usage around ~12% single core on a pi 2b using TMRh20's RF24 lib. I guess thats coming from the relatively high receive buffer poll rate and channel hopping frequency in the receive loop. This could be even sightly higher if the RF24 module ist not configured to use bcm hardware acceleration for spi.

Bad news so far, I am currently in lack of time and being in the middle of the process, rewriting the code to support circuitpython's rf24l01 library, as this is purely pip installable. Also transforming the transaction handlers to support MI transactions.
My goal is here to be able to finaly split the python code into it's own repository to be able to build and upload this to pypi.org

circuitpython's rf24l01 on my dut does bump up the cpu usage even up to 20% single core. I don't think it will go down in the meantime and that is also not at the very top of my priority list. I'm fine with my pi idling around 5% total cpu usage. Optimizations eventually come, after the protocol is well understood and settled. I don't want to waste much time on tweaking code which eventually has to be replaced again because of new discoveries.

stefan123t, 2022-06-15:
Yes, maybe a safe / fool proof default minimum would be well above 30-60 seconds. Having it poll every 300-900 seconds (5-15 minutes) would be reasonable as you store and discard/rotate your data later anyway.

closed 2022-06-15 by berni2288

Sprinterfreak, 2022-06-15:
@stefan123t just, no. That would mean to have poor quality data, You have to interpolate to get readable graphs out of. If I set an interval, I need that interval for every measurement. Interval is not to confuse with delay.
If the interval is set smaller as the communications take, it will not wait and try to get vacant data asap. That's at least required, especially if You plan to use this data in closed loop for zero export.

What I did see tho is, there is continuous querying configured in the ahoy.yml template (interval: 0) which could be set to like 10 seconds. But leave it up to users how they require their data.

I will change the template soon.

stefan123t, 2022-06-16:
@Sprinterfreak I dunno maybe we could use the 0x15 or 0x09 command to query the inverters internal history.
According to the documentation there is a sub command byte[10] and a gap byte[19] which defines the interval between these historic readings:
[image]

Maybe we can use the sync-time to get old data from the inverter too ?

### Sub Command byte[10]
* 00: upload all frames
* nn: upload the N-1th frame (nn=>01)

### Gap (Collection interval time)  byte[19]
* 00: 5min
* 01: 10min
* 02: 15min