lumapu/ahoy #77: Use the 0x15 or 0x09 command to query the inverters internal history

Sprinterfreak, 2022-06-16:
@Sprinterfreak I dunno maybe we could use the 0x15 or 0x09 command to query the inverters internal history.
According to the documentation there is a sub command byte[10] and a gap byte[19] which defines the interval between these historic readings:
[image]

Maybe we can use the sync-time to get old data from the inverter too ?

### Sub Command byte[10]
* 00: upload all frames
* nn: upload the N-1th frame (nn=>01)

### Gap (Collection interval time)  byte[19]
* 00: 5min
* 01: 10min
* 02: 15min

_Originally posted by @stefan123t in https://github.com/grindylow/ahoy/issues/74#issuecomment-1158106737_

Sprinterfreak, 2022-06-16:
@stefan123t You can't mess around with the sync time because this resets daily counters. We could try of course, if this also applies for 0x80 0x11 event log request instead of 0x0b presumably set time request.
By my observations, this is not the case. Entries get overwritten when log is full. Important entries seem to overflow to 0x12.
The table shown above is for MI inverters but is mostly similar to HM's.

I guess HM's use "command" different (Nordic calls it "Preamble" [1]). The command table from MI inverters does not apply. At least returns mostly none or weird data. subcommand can have multiple bytes on HM's. 0x80 0x0b or 0x80 0x11 to name the known ones.

5min inteval? useless. Also I think, this does only apply to MI inverters.

[1] https://www.sparkfun.com/datasheets/Wireless/Nordic/nRF24L01P_Product_Specification_1_0.pdf

stefan123t, 2022-06-16:
@Sprinterfreak 
you are right, `0x80 XX` where `XX` is the Sub Command can be found in `usart_nrf3.c/.h`.
`uint8_t UsartNrf3_Send_PackDataCommand(uint8_t *target_adr, uint8_t *rout_adr, uint8_t DataType, uint8_t DataMode, uint8_t Gap, uint8_t *Password)`

Here are the Sub Commands I found in usart_nrf3.h:
```
typedef enum
{
    InverterDevInform_Simple = 0, // 0x00
    InverterDevInform_All = 1, // 0x01
    GridOnProFilePara = 2, // 0x02
    HardWareConfig = 3, // 0x03
    SimpleCalibrationPara = 4, // 0x04
    SystemConfigPara = 5, // 0x05
    RealTimeRunData_Debug = 11, // 0x0B
    RealTimeRunData_Reality = 12, // 0x0C
    RealTimeRunData_A_Phase = 13, // 0x0D
    RealTimeRunData_B_Phase = 14, // 0x0E
    RealTimeRunData_C_Phase = 15, // 0x0F
    //RealTimeRunData_Password = 16, // 0x10
    AlarmData = 17, // 0x11
    RecordData = 18, // 0x12
    InternalData = 19, // 0x13
    ExternalData = 20, // 0x14
} DataType;
```

The Sub Command is handed over in `DataType`
Especially the `0x0B` rings a bell with me and our practice so far!
`0x11` is the `AlarmData` and `0x12` the `RecordData` as you already hinted earlier too.

While `DataMode` can be `CurRecLostPackageNO`, `0xFF` or `0`.

If you look closely you can see that the implementation always uses Gap = 0 in `UsartNrf3_Send_PollVersionCmd` that IMHO is the last five minute interval.

stefan123t, 2022-08-12:
Das Issue sollte mit #145 implementiert werden.
SubCmd 0x11 AlarmData und 0x12 AlarmUpdate sind ja Teil der Daten die man mit dem DevInform MainCmd 0x15 REQ_ARW_DAT_ALL von den HM-Gen3 Wechselrichtern bekommen kann.

Für MainCmd 0x09 REQ_A_DAT sollten wir ein eigenes Issue als Unterpunkt von #145 für die MI-Gen2 Wechselrichter aufmachen.
Ziyat T. und Jörg R. haben dafür bereits einiges implementiert (siehe Forum).

closed 2022-08-16 by aschiffler