lumapu/ahoy #79: Turn off the blue wemos led

MubiTec, 2022-06-17:
Would it be possible to turn off the onboard led?

Thanks a lot for this great project!

DanielR92, 2022-06-19:
Hello,

try it with this code:

# disable built-in LED (inverted logic > pin D4)
pinMode(GPIO02, OUTPUT);
digitalWrite(GPIO02, HIGH);

But good idea, it is possible to build in the webinterface for a "status".
Example: LED blink > error?
LED off > just fine (or invert led)
LED on > Init

closed 2022-06-23 by MubiTec

stefan123t, 2022-06-23:
@DanielR92 it would be possible to use D2 as a signal LED as you mentioned.
Currently we use it as Chip Enable for the nRF24 chip, i.e. when there is no communication between the ESP and nRF the line goes down and we see the blue light. Actually this is the time when the ESP has free cpu for non-nRF things like WiFi, MQTT, etc. So in principal it acts as an indicator whether the ESP is reachable. Not too bad IMHO.