lumapu/ahoy #81: Fehler beim Kompilieren des ESP8266 Programms in der Arduino IDE

ocxjuergen, 2022-06-25:
Hallo, 
Bekomme beim Kompilieren diesen Fehler in der Arduino IDE
Habe den Ordner ganz normal Runtergeladen, entpackt, alle Libraries installiert.

"conversion from 'const __FlashStringHelper*' to 'const StringSumHelper' is ambiguous"

ist in der app.cpp ganz unten

Liegt Hier das Problem bei mir oder ist hier etwas anderes?

Danke

stefan123t, 2022-06-27:
Hallo @ocxjuergen,
ich vermute mal Du hast noch nicht das richtige Board "ESP8266" Wemos D1 o.a. im Tools Menü konfiguriert.
Die FlashStringHelper sind ein Macro, das m.W. Bestandteil von Arduino/ESP8266 ist und daher bei Auswahl der richtigen Boards vorhanden sein sollte.
Hast Du die URL für den ESP8266 unter dem Board Manager eingetragen und die Arduino IDE neu gestartet ?

ocxjuergen, 2022-06-27:
Hallo Stefan,

Boards sind eigentlich alle Installiert, trotzdem Vielen Dank für die tolle Anleitung. Hab beim Kompilieren ja wie laut Anleitung das „Generic 8266 Module“ ausgewählt sowie auch die speichergröße vom Flash. Hmmm werde es einfach nochmal mit einen anderen Board(z.B Wemos D1) probieren und dann kompilieren und schauen was passiert.

stefan123t, 2022-06-27:
Hallo Jürgen, ich verwende bei mir ein NodeMCU v1.0 Board mit ESP8266-12E/F chip.  
Den Wemos D1 habe ich immer noch nicht ausprobiert aber zumindest schon mal die Pin-Header drangelötet.
Kannst Du ggf ein Logfile anhängen mit etwas mehr Kontext Deiner Fehlermeldung ?

ocxjuergen, 2022-06-28:
Hallo Stefan, Okay werde mal mit den NodeMCU Board probieren.

Anbei das Logfile: (für Pastebin zu groß)

https://zerobin.net/?7784faab35de51ad#/QIlaV9wD/mVrGFNzKBdIHbCVxexIpQd80m2eB/J5oc=

ocxjuergen, 2022-06-28:
Hallo Stefan,
jetzt lässt sich es  kompilieren - habe zusätzlich noch die ESP32 Boards runtergeladen. Laut Arduino Forum (finde den Link leider nicht mehr) gibt es irgendwie Probleme mit den SSL Files zwischen der ESP8266 2.7.4 Version und der 3.0.2 - bei der neueren Fehlen eben die SSL Files - die jedoch wiederrum bei den ESP32 Boards dabei sind. =)

Trotzdem Danke für die Hilfe!
LG

stefan123t, 2022-06-29:
Ich glaube ich habe den Artikel gefunden:
https://forum.arduino.cc/t/f-question-syntax/271595/3
Wir sollten also den Code optimieren und String() concatenation / casts vermeiden.

ocxjuergen, 2022-07-01:
@billy0xff habe die esp8266 v3.0.2 upgedatet, danach die esp32 boards runter geladen, danach ist es bei mir ganz normal zum kompilieren gegangen… was schreibt er dir den für einen Fehler beim Kompilieren? Ja klar, kann dir gerne die Files schicken!

billy0xff, 2022-07-02:
@ocxjuergen: danke, jetzt hat es geklappt, ich hatte nicht die esp8266 boards auf 3.0.2 aktualisiert gehabt.

DanielR92, 2022-07-04:
Super das alles geklappt hat, wen möglich dann diesen Thread schließen.

closed 2022-07-04 by ocxjuergen