lumapu/ahoy #87: Reset Yield total and Store yield today ?!

eeprom23, 2022-07-03:
Hello,
It is possible to testet Yield total ?
is yield today Stored in the esp for Dome days ?

NDS234, 2022-07-04:
It is possible to reset Yield total ?  --->  The value ist stored in the inverter. Can not be reset until today

eeprom23, 2022-07-04:
> It is possible to reset Yield total ? ---> The value ist stored in the inverter. Can not be reset until today

It is possible to make a Little history for yield Today Which Save the Last Month?

thinklerde, 2022-07-04:
You need an MQTT server. A database. And something to show that off nicely.

For this I use a RaspberryPI on which ioBroker runs with MQTT, as the database influxDB, and I display this graphically in Grafan. In Grafana you can then display everything possible. Day/Week/Year/ Last 5 minutes/... and much more

[image]

lumapu, 2022-07-04:
I think like @thinklerde mentioned. You should do this kind of post processing in an environmet where you have resources to store a lot of data. 
My sight to the ESP module is more like a bridge between inverter and your home automatisation.

stefan123t, 2022-08-12:
@eeprom23 das YieldTotal kann m.W. nicht überschrieben werden. 
Die DTU-Pro überschreibt nur ihr eigenes YieldDaily auf dem SPIFFS der DTU, nicht das im WR!
Daher bitte das Issue als nicht lösbar schließen. Danke.

closed 2022-08-12 by lumapu