lumapu/ahoy #88: MQTT Server  erreichbar per  IP Adr. und / oder DNS, ON / OFF  Feature Request

NDS234, 2022-07-04:
Kopie von https://www.mikrocontroller.net/topic/525778?goto=7116390#7116209

Ist nur ein Vorschlag:

Es spielt doch gar keine Rolle wor ein MQTT Server steht !
Das kann intern oder extern sein.

Mein Vorschlag für die Programmierer wäre:

MQTT ein/aus  wählbar

Wenn MQTT ein, dann

A) IP Addr und Timeout in ms  angebbar
B) DNS Adr. und Timeout in ms  angebbar

Reihenfolge zur Erreichbarkeit Auswahl:
(folgende Optionen nur wählbar, wenn A) bzw. B) auch ausgefüllt)

Nur A)
Nur B)
Erst A) und wenn nicht erreichbar bis Timeout dann B)
Erst B) und wenn nicht erreichbar bis Timeout dann A)

So was in der Art würde vielleicht Sinn machen.

thinklerde, 2022-07-04:
Es wäre ja schon schön, wenn man anstatt einer IP Adresse eine Webadresse eingeben könnte.  Z:B: "xyz.ddns.net"

lumapu, 2022-07-04:
ja das stimmt, guter Feature-Request. Wie lange URLs sollen wir zulassen - ist immer eine Speicherfrage und die Geschäcker sind bekanntlich verschieden. Die Vorgabe kann natürlich dann wieder in der `config.h` geändert werden - aber der Default sollte schon für den Mainstream passen finde ich.
Genügen 32 Zeichen oder lieber gleich 64? Es geht natürlich jede beliebige Zahl - ich denke nur gerne in 2er-Potenzen.

NDS234, 2022-07-05:
wenn man z.B. mal nimmt:  mqtt.meinhaus.meinestadt.net wären das 28 Zeichen. Ich denke, 32 Zeichen reichen da für den DNS Namen. Ein vielfaches von 2 oder vielleicht besser 4 würde ich vielleicht nehmen wegen dem Alignment im Speicher.
Bevor nun die Frage nach Defaults für die TimeOuts kommen:
MQTT Server Erreichbarkeit TimeOut für DNS viellecht auf 500 ms als Default setzen ?
MQTT Server Erreichbarkeit TimeOut für IP vielleicht auf 150 ms als Default setzen ?
Jemand, der damit was anzufangen weiss, kann es ja dann ändern. Ich denke, die beiden Werte sollten auf der Sicheren Seite sein, für die, die damit nicht so richtig was anfangen können.

stefan123t, 2022-07-07:
@lumapu & @NDS234 ja 32 sollten grade so reichen z.B. auch für die verbreiteten "123456789ABCDEF.myfreez.net" Adressen, meine hat  zumindest nur 28 chars. Eintragen in der config.h wäre m.E. ausreichend.

stefan123t, 2022-07-10:
Haha den DNS Eintrag in PR #98 wollte @thinklerde als Feature, vielleicht kann er es ja mal testen ?
Müssen wir jetzt eigentlich auch einen DNS Server (1.1.1.1 oder 8.8.8.8, etc.) konfigurierbar machen oder reicht der, den der ESP über DHCP bekommen hat ?

NDS234, 2022-07-10:
Da es kein Zwang ist, per DHCP einen DNS Server bekannt zu machen, wäre es optional mal irgendwann evt. mal sinnvoll, einen von Hand eintragen zu können. Für die, die eine Fritzbox ihr Eigen nennen ist ja normalerweise die Fritzbox der DNS Server. Bei anderen Routern wird es sicher ähnlich sein. Erst mal würde ich da nix machen. Gibt noch Wichtigeres zu tun.

stefan123t, 2022-07-10:
Ja, wenn man einen anderen (externen) DNS im ESP angibt dann findet er evtl. lokale Namen / Adressen nicht mehr. Es wird also ggf. beliebig kompliziert. Ich denke wir verlassen uns erstmal auf DHCP und schliessen diesen Feature Request wenn @thinklerde es getestet hat.

closed 2022-07-11 by lumapu

thinklerde, 2022-07-11:
das scheint soweit zu funktionieren. Allerdings aktualisiert er nur noch alle 90sek. lässt sich auch nicht anders einstellen. General Intervall auch nicht kleiner 30sek?

commit 10771d8f (2022-07-11): Merge pull request #98 from stefan123t/main

fix #88 and #93, clean up #63
MQTT was tested using IP address