lumapu/ahoy #90: "Inverter is available and is not producing" Error when ESP is not connected to Wireless-Network

oxylog, 2022-07-04:
Wenn Ahoy ohne Wlan-Zugriff (also nur mit "eigenem" Ahoy-DTU-Netzwerk)  gestartet wird, bekommt man auch bei korrekt eingetragener Seriennummer "Inverter is available and is not producing". Sobald eine Verbindung zu einem Router besteht und die Zugangsdaten eingegeben wurden funktioniert es wieder tadellos. 

Gäbe es eine Mäglichkeit dies zu beheben und die Werte auch über das "Ahoy-DTU"-Netzwerk abfragbar zu machen?

lumapu, 2022-07-04:
Das hätte ich eigentlich erwartet. Komisch dass es nicht geht. Bei Gelegenheit prüfe ich das mal bei mir, evtl. ist bisschen Debugging nötig um herauszufinden wo es klemmt.
Eine Idee habe ich - evtl. weigert der ESP sich ohne aktuelle Zeit eine erste Anfrage zu stellen. Man müsste wahrscheinlich für die Stand-Alone Lösung hier eine weitere Bedingung einfügen.

oxylog, 2022-07-05:
Falls es zum Troubleshooting dienlich ist: In 0.4.14 funktioniert es, wenn auch nur kurzzeitig. Nach ein paar Minuten ist der D1mini nicht mehr erreichbar und benötigt einen liebevollen Reset

oxylog, 2022-07-05:
Noch eine Info: Es scheint mti der Zeitabfrage zu tun zu haben. Unterbinde ich den Internetzugriff und die NTP-Abfrage, so funktioniert quasi nichts - nur die Statusseite lässt sich extrem langsam aufbauend darstellen.

lumapu, 2022-07-05:
Das bestätigt meine obige Annahme. Ich denke folgende Zeile ist dafür verantwortlich:

https://github.com/grindylow/ahoy/blob/main/tools/esp8266/app.cpp#L315

Wenn ich am PC sitze baue ich das noch eine Bedingung für Stand Alone ein.

stefan123t, 2022-08-12:
@oxylog bitte mit der v0.5.9 überprüfen und ggf. als gelöst schließen. Danke

oxylog, 2022-08-12:
Problem leider weiterhin persistent - geprüft auf einem frisch geflashten ESP8266 - Ebenfalls wird, wenn man kein Limit einträgt, dieses als 0 gesetzt - dank falscher Zeit aber nicht gesendet. 
Weiterhin bleibt die Meldung, dass der Inverter erreichbar ist aber nicht produziert. 

Check mit einer 0.4.22  und Zeitsynchronisation -> alles "im lot "

lumapu, 2022-08-12:
Frage: hat du eine extra Version kompiliert mit `#define AP_ONLY` oder die Standard Firmware verwendet?
Ich denke es geht nur mit einer gesondert kompilierten Version, da sonst die Zeit bei 0 (01.01.1970) stehen bleibt

oxylog, 2022-08-12:
> Frage: hat du eine extra Version kompiliert

Ich hatte mir mit Downgit das aktuelle Repo runtergeladen und selbsttätig so kompilieet wie es heruntergeladen war. Ich habe aber keine Änderungen im Code vorgenommen. 

Was mir ebenfalls aufgefallen ist - es werden weder Pakete gesendet noch empfangen - wird dann ei n Payload überhaupt generiert?
Bin leider kompletter Novize auf dem Gebiet,...

lumapu, 2022-08-12:
> Ich habe aber keine Änderungen im Code vorgenommen.

dann würde ich dich bitten vor dem kompilieren mal diese Zeile einzukommentieren und dann nochmal zu prüfen:
https://github.com/grindylow/ahoy/blob/e1c713fb003f2114171110417755191adc1b88ef/tools/esp8266/config.h#L18

stefan123t, 2022-08-14:
@lumapu wäre es eine Idee die Zeit anstelle über NTP (zusätzlicher Button zum aktualisieren im Setup ?) auch manuell zu stellen ? Dann könnte man ggf auch eine Warnung ausgeben so lange die Zeit noch auf 1.1.1970 steht die auf die fehlende Uhrzeit hinweist.
Der restliche Programmcode scheint ja so weit korrekt zu sein, nur eben bisher keine Möglichkeit die Zeit von Hand zu stellen.

lumapu, 2022-08-14:
ja und MQTT hat glaube ich auch eine Prüfung auf AP mode.
Im Setup könnte man das vorsehen, allerdings driftet die Zeit doch relativ schnell ab oder? Die Frage, die sich mir stellt: wie viele Installationen gibt es wirklich im AP only mode und was bringt diesen wenigen eine genaue Zeit?

stefan123t, 2022-08-14:
@lumapu ich weiss es auch nicht wie viele Nutzer es gibt aber ein Input Field für Datum + Uhrzeit sollte in JS den Timestamp als Unix Epoch übergeben können und das Problem ist für die meisten Nutzer gelöst/behoben.

Ich habe mal nach ESP8266 und clock drift gesucht:
https://www.instructables.com/TESTED-Timekeeping-on-ESP8266-Arduino-Uno-WITHOUT-/
TLDR:
> After about 3 days, on both my arduino and wemos the clock was about 2 to 2,5 sec fast. that is 274 seconds a year off or just over four and a half minutes.

Es ist also gar nicht so schlecht wie böse Zungen oder Hersteller von RTC Modulen behaupten.

closed 2022-08-17 by aschiffler