lumapu/ahoy #91: Plausability checks

eingemaischt, 2022-07-05:
Today, ahoy on my esp8266 reported one wrong state for yield total: Over 16.000kWh. 

Would it be a good idea to add a small plausability check für transmitted values to get sure that they are not corrupted? yield total should not jump over thousands of kWh between two transmissions...

lumapu, 2022-07-06:
It would be nice if you share the type of inverter you are using.

I think this problem can occur only for 2 channel inverters because they are using two frames to transmit their data and one field is split over these frames.
From the implementation side I see no problem, the data is only processed if the CRC is correct.

It would be helpful to see the raw date for such a broken message.

stefan123t, 2022-07-10:
Philipp @eingemaischt we are missing your feedback to progress this issue.
Can you share the serial logs with transmitted and received packets for us to verify and possibly fix your issue ?

hismastersvoice, 2022-07-10:
Today we found an issue with implausible values 0 or X thousand watts. (openDTU)

If you have an ESPxxxx and an org. DTU from Hoymiles that are active at the same time, such values happens.
As soon I disconnect the org. DTU the values are not any longer implausible, and direct after I plug again the org. DTU the implausible values came back.

It is not intended that two DTUs are connected to one inverter. :(

stefan123t, 2022-07-10:
@lumapu do we have a check that response packets are actually intended for our fake DTU serial number or do we take everything coming from the inverter as being granted ?

lumapu, 2022-07-10:
we do not check the fake DTU serial. Everything which seems too be valid is accepted. Could be a new feature request. Do you think that this issue can be related to another request which is using the same frame IDs?

stefan123t, 2022-07-10:
@lumapu yes feature request.
But worthwhile imho +1 as we should verify the answer was intended for us.
Only in "promiscuous mode" which I suggested in response to Denny S. query on the forum today we should accept any packet from the inverter, regardless of the destination DTUs serial number.

eingemaischt, 2022-07-11:
I had the problem a second times - and several crashed on the esp8266. Sorry, it was not connected to my RPi during the second "wrong" transmission because I thought that the "low power port" on the Pi would be a possible cause for crashes.

I am not using an official DTU in parallel.

stefan123t, 2022-07-11:
So the problem occurs on the ESP8266 but you use the Raspberry Pi for Building and Flashing your ESP ?
Thanks for clarification. Please upload the logs once you are connected to the RPi again and have something in your serial monitor to report. You can surely keep the issue open if this reproduces several times a week.

eingemaischt, 2022-07-11:
nope, I am using my notebook for building and flashing - the rpi is my homeassistant server and I would only use it to save energy.

But I now tried OpenDTU (i had several spare esp32) and will see if it also produces error transmissions. Thank you for your help and sorry that I could not provide error logs.

closed 2022-07-11 by eingemaischt