lumapu/ahoy #92: Fehlerhafte MQTT Datenpunkte und Werte

HFarryranz, 2022-07-06:
Im Einsatz 0.4.20 auf NodeMCU lose verkabelt. IoBroker mit MQTT-Adapter: 

Bei mir kommt es immer wieder zu reboots, mehr wie ein Tag geht nicht. Daten werden aber geliefert, nach dem ich einen passenden Standort für die NodeMCU gefunden habe, damit WLAN und Funk zuverlässig funktionieren,
Receive fail ist max  1.

Im MQTT tauchen aber immer wieder mal fehlerhafte Datenpunkte auf:
mqtt.0.inverter.HM-600.0!_inve                     Wert: rter/
mqtt.0.inverter.HM-600.c0&_inv                    Wert: erter/H
mqtt.0.inverter.HM-600!_invert                      Wert : er/HM

Löschen der Datenpunkt geht, aber sie kommen nach einigen Tagen wieder.
Alle anderen Datenpunkte passen.

lumapu, 2022-07-06:
habe die letzten Tage schon ähnliches gelesen und beobachte das auch in meiner Installation.

Es wird aktuell die Lib `PubSubClient` verdächtigt.

Hans-JR, 2022-07-10:
Ich habe jetzt seit mehreren Tagen die 0.4.22 auf dem gleichen System wie der Themenstarter im Einsatz und habe auch die reboots sowie die selben fehlerhaften Datenpunkte.

fila612, 2022-07-25:
Also ich habe bisher keinerlei solche Themen feststellen können. Nutze jedoch HA und nicht IOBroker. tritt das ganze denn auch mit der 4.25 auf?
sind die Fehler auch in einem MQTT Tool ersichtlich? Kann der IOBroker aus Ursache ausgeschlossen werden und warum?

stefan123t, 2022-08-04:
Aktuell können wir das Problem nicht / schwerlich reproduzieren.

Wir hatten in der reconnect() Methode einen Zeiger auf einen temporären String übergeben.
Eventuell hängt das hier beobachtete Problem (dangling pointer) damit zusammen ?

Bitte einmal mit Version 0.4.26 überprüfen und falls das Problem damit gelöst ist, diese Issue bitte schließen.

stefan123t, 2022-08-12:
@HFarryranz bitte mit der v0.5.9 überprüfen und ggf. als gelöst / nicht reproduzierbar schließen. Danke

aschiffler, 2022-08-16:
closed due to inactivity

closed 2022-08-16 by aschiffler