lumapu/ahoy #94: Wifi password gets overwritten on setup

martin-at-schorsch, 2022-07-07:
On  0.4.23 ESP8266 version, when saving the setup, the Wifi password seems to be overwritten by a 5 char password.

Setting the password correctly every time before saving is a workaround.

BR from Austria
Martin

lumapu, 2022-07-10:
I can't imagine that this was changed in that version. For me I had to update the WiFi settings while I switched from Arduino IDE to Visual Studio Code and PlatformIO.

stefan123t, 2022-07-10:
I set the password every time and/or better rely on the password configured in config.h. But with some of the changes we made to eeprom I had to reconfigure it. @martin-at-schorsch but I do not recall resetting it *every* time. Indeed the browser only shows the 5 star password but it does not require setting it again.
@lumapu if we require 4MB flash, would it be worthwhile to switch to SPIFFS filesystem and store the config in a flat file with key=values ? 
I dunno how Thomas handles it in the OpenDTU ? I believe the Hoymiles DTU stores a lot in files on the SPIFFS, it even stores some of the daily values in plain files and writes the OTA / firmware updates for the inverters into SPIFFS before uploading / flashing it to the targets. Also the Grid Profile is stored in SPIFFS.

lumapu, 2022-07-10:
I think execept for password it could be an easy way to store settings. The whole `saveValues` function isn't needed any more, but we need to better check the values for overflow / wrong type and so on.

stefan123t, 2022-07-10:
Yes we could have two files one for WiFi / SSID settings and one for all the other stuff. But this is a larger change which needs some more love =^D

martin-at-schorsch, 2022-07-11:
I just retryied it and you are right, it looks like a problem only occurring after the first setup.

My steps:
- Hit the "erase flash" in Platform.io
- Upload and Monitor
- ESP is connecting to my Wifi using the credentials from config.h
- I access the setup webpage
- Set the GPIO to match my wiring
- press "save"
- ESP is NOT connecting to my WiFi but starts its own AP after some time
- Accessing this AP allows me to set the correct Wifi Password
- After this, it works fine, including saving the settings without entering the Wifi password again.

here the serial monitor log:

`======================================================================= [SUCCESS] Took 54.96 seconds =======================================================================
--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time    
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I: connect to network 'Gtec-IoT' ...
...............................
................................
I: Settings not valid, erasing ...
I: ... restarting ...

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0005cc70
~ld
I: connect to network 'Gtec-IoT' ...
...............
I: RF24 Amp Pwr: RF24_PA_MIN
I: Radio Config:
SPI Frequency           = 1 Mhz
Channel                 = 0 (~ 2400 MHz)
RF Data Rate            = 1 MBPS
RF Power Amplifier      = PA_MIN
RF Low Noise Amplifier  = Disabled
CRC Length              = Disabled
Address Length          = 2 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 0 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 0
Multicast               = Disabled
Custom ACK Payload      = Disabled
Dynamic Payloads        = Disabled
Auto Acknowledgment     = Disabled
Primary Mode            = TX
TX address              = 0x0000000000
pipe 0 (closed) bound   = 0x0000000000
pipe 1 (closed) bound   = 0x0000000000
pipe 2 (closed) bound   = 0x00
pipe 3 (closed) bound   = 0x00
pipe 4 (closed) bound   = 0x00
pipe 5 (closed) bound   = 0x00
W: WARNING! your NRF24 module can't be reached, check the wiring
I:

----------------------------------------
I: Welcome to AHOY!
I: 
point your browser to http://192.168.30.17
I: to configure your device
I: ----------------------------------------

I: [NTP]: 2022-07-11 13:44:37
I: Serial debug is I: off

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0005cc70
~ld
I: connect to network 'Gtec-IoT' ...
...............................
....................................................................................................
I: RF24 Amp Pwr: RF24_PA_MIN
I: Radio Config:
SPI Frequency           = 1 Mhz
Channel                 = 3 (~ 2403 MHz)
RF Data Rate            = 250 KBPS
RF Power Amplifier      = PA_MIN
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 0 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 0
Multicast               = Disabled
Custom ACK Payload      = Disabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = Disabled
Primary Mode            = RX
TX address              = 0xe7e7e7e7e7
pipe 0 (closed) bound   = 0xe7e7e7e7e7
pipe 1 ( open ) bound   = 0x1234567801
pipe 2 (closed) bound   = 0xc3
pipe 3 (closed) bound   = 0xc4
pipe 4 (closed) bound   = 0xc5
pipe 5 (closed) bound   = 0xc6
I:

----------------------------------------
I: Welcome to AHOY!
I:
point your browser to http://192.168.1.1
I: to configure your device
I: ----------------------------------------

I: connect to network 'Gtec-IoT' ...
...............................
....................................................................................................
I: 
---------
AP MODE
SSID: AHOY-DTU
PWD: esp_8266
Active for: 60 seconds
---------

I: AP will be closed in 49 seconds`

stefan123t, 2022-07-11:
@martin-at-schorsch thanks for verifying it time and again and providing the extensive logging.
I can see that it is working as intended, you were using erase flash in the IDE, hence the EEPROM storage area was wiped.
So first it erases that area again, which is maybe superfluous (@lumapu) in order to write default/empty settings to the area and calculate a matching CRC Checksum.
After reboot into AP mode for the second time you can configure the NRF24 chip connections and everything seems alright.

Lukas, could we clean up the code a bit and consider not requiring the reboot here after erasing the EEPROM settings in the first place ?
Martin, would you prefer to close this issue as the problem seems to be explained / solved ?

closed 2022-07-11 by martin-at-schorsch