lumapu/ahoy #96: TOR Einstellung ändern

martin-at-schorsch, 2022-07-07:
Die Konfiguration der passenden TOR-Einstellungen wäre nötig damit der Betrieb in Österreich oder Deutschland erlaubt ist.
Dies ist wohl über die DTU problemlos möglich. Es gibt Profile für Österreich (AT_TOR_Erzeuger_default) und Deutschland die über den Standort der Anlage ganz einfach festgelegt werden können.

Daher wäre es perfekt wenn diese Feature eingebaut werden könnte damit man auch für die Installation keinen DTU benötigt.

stefan123t, 2022-07-07:
Hallo @martin-at-schorsch ich vermute das sind die GridProfile's die man mit der offiziellen DTU einstellen kann.

Soweit ich es verstanden habe wird dabei der Wechselrichter abgefragt und liefert 1200 Datenpunkte seiner Inverterkurve, praktisch die komplette Sinus-Kurve einmal als Fensterfunktion. 

Die Daten liegen (/lagen früher) unter `Download/GridProfilesList.txt` bzw `/Download/GridProfilesList.hex` und `/Config/Grid_Profiles_Cfg_Data.dat` auf dem fatfs der DTU. Ich vermute man kann über den USB Port einen Stick mit den Dateien anschließen, analog zum DTU / MI Program upgrade `/Download/DTUProgram.hex` bzw. `/Download/MIProgram.hex`.

Schau Dir doch mal die Methoden `UsartNrf_Send_PackUpdateGridOnProFile()` bzw. `UsartNrf3_Send_PackUpdateGridOnProFile` in `UsartNrf3_Process_DevInform_GridOnFile()` und `UsartNrf3_Process_DevInform_RecordWave()` an.

`RecordData` `0x13` und `GridOnProFilePara` `0x02` sind beides `SubCmd` für das `MainCmd` `REQ_ARW_DAT_ALL` `0x15`.

martin-at-schorsch, 2022-07-08:
da ich keinen DTU haben kommen ich an diese Daten leider nicht.

stefan123t, 2022-07-08:
@martin-at-schorsch es gibt ein gitee repository (iotloves) mit dem vermeintlichen original source code der hoymiles-DTU-PRO software auf dem stm32f4xx.

martin-at-schorsch, 2022-07-11:
@stefan123t leider finde ich mit deinem Hinweis kein Repository. zZ habe ich den Wechselrichter auch noch nicht im Betrieb, bin da noch in Vorbereitung.

stefan123t, 2022-07-11:
@martin-at-schorsch im Forum Thread mal nach "iotloves" oder "gitee" suchen ... das hat ein Leser in China entdeckt und den/die Links gepostet. Wie gesagt das iotloves ist aktuell das interessanteste der drei Repos.

unhold, 2022-07-19:
The repo with the DTU firmwares is here: https://gitee.com/iotloves/projects

unhold, 2022-07-19:
If you don't want to register at Gitee you can directly clone it from https://gitee.com/iotloves/hoymiles-DTU-PRO.git

stefan123t, 2022-08-12:
@martin-at-schorsch hast Du in der Zwischenzeit etwas zu den Grid Profiles herausgefunden. 
Bitte erwähne doch noch GridProfile im Titel dieses Issues. Danke

MainCmd: REQ_ARW_DAT_ALL 0x15 und SubCmd: GridOnProFilePara 0x02 oder 
MainCmd: DOWN_DAT 0x0a und SubCmd: Type_Init 0xff
Dürften hierfür relevant sein / werden...

martin-at-schorsch, 2022-08-14:
Ich hab das Git-Repo runtergeladen und dann bin ich ins stocken gekommen.
Ich finde mich allerdings auch im Code nicht zurecht.

stefan123t, 2022-08-14:
Hallo @martin-at-schorsch 
hast Du das gitee oder das github Repo heruntergeladen ?
Im gitee repo gibt es zwei/vier Dateien usart_nrf[3].c/h Darin ist im Prinzip die ganze Kommunikation mit den Wechselrichtern programmiert. Es gibt Funktionen/Methoden um die Pakete zum Wechselrichter zu erstellen (alles mit Usart3_Send*) und die Methoden um die Antworten zu parasen (alles mit Usart3_Process*)
Ich habe auch für die meisten Befehle / Pakete bereits eine FAQ angelegt in der die Kommunikation exemplarisch mit Frage/Antwort Paketen gezeigt wird. 
Für die og GridOnProFilePara und DOWN_DAT Kommandos habe ich aber glaube noch nichts aufgeschrieben. Klahus1 / Ichirou im Discord Chat wollten evtl auch diese Kommandos noch exemplarisch aufzeichnen damit wir sie decodieren können.

closed 2022-09-04 by aschiffler

mozzbozz, 2022-10-03:
@aschiffler why did you close this issue as completed? Has there been a PR to implement this feature - or was it on accident?