lumapu/ahoy #100: 0.4.24 Setup.

costa2, 2022-07-11:
Im Setup kann ich das Intervall nicht mehr ändern, es bleibt bei 30s.
Ist das so gewollt?

thinklerde, 2022-07-11:
Als schnelle hilfe: 
ändere in der config.h den Wert 
"#define MIN_SEND_INTERVAL       30"

auf:
"#define MIN_SEND_INTERVAL       5"

stefan123t, 2022-07-11:
@thinklerde ja ich habe die [Default Minimum Werte](https://github.com/grindylow/ahoy/pull/98/commits/d4ba302307b2b6995c2df0a83e4d02ff45bbd08d) mal etwas höher angesetzt um Probleme vor allem bei neuen Nutzern auszuschließen. Wer seinen Inverter häufiger abfragen will/muss, kann das ja wie Du beschrieben hast in der config.h jederzeit anpassen.

stefan123t, 2022-07-12:
@mslookup ja prinzipiell gebe ich Dir recht, wenn man nur 3 WR hat, dann kann man das `MIN_SEND_INTERVAL` auch auf 6-12 Sekunden runtersetzen. Im Hoymiles Code werden teilweise zwei Sekunden als max. Wartezeit auf die Antwortpakete definiert.

Es gab bis jetzt noch keinen `SERIAL/MQTT/SEND_INTERVAL` default Wert. Habe das mal so eingebaut ... also die MIN Werte in die `defines.h` verschoben und in der `config.h` die Default Werte vorgegeben. Ich habe mal auf 15 Sekunden `SEND_INTERVAL` gestellt. Das sind dann `4x60*24=5760` Werte/Abfragen am Tag, da ist der Wechselrichter besser überwacht als so mancher Server. Bei 5 Sekunden Intervall wären das `17280` Abfragen pro Wechselrichter!

stefan123t, 2022-07-14:
@costa2 können wir das Issue schließen ?

closed 2022-07-15 by costa2