lumapu/ahoy #103: Uninizialisierter Speicher für WIFI Daten bei inbetriebnahme

Qowy, 2022-07-13:
Hi,
super Programm, vielen Dank.
Mir ist aufgefallen, dass FB_WIFI_SSID und FB_WIFI_PWD komplett ignoriert werden.
Über den Hotspot kam ich dann rein, und wurde mit absoluten UTF-8 Müll in den Feldern SSID Password und Device Name begrüßt.

Am auffälligsten war vor allem, dass  DeviceName voll mit sehr vielen "ÿ" war. 
Das sollte UTF-8 0x00FF sein.
Da das als "vorhandene WIFI Daten" akzeptiert wird werden wohl auch die Sachen aus Config.h ignoriert.

Nach löschen und neu eingeben der entsprechenden Felder geht es dann aber.

Wird hier aus irgendwelchen gründen speicher einfach garnicht initialisiert? 
Oder Fälschlicherweise mit 0x00FF stat 0x00 o.Ä.?

stefan123t, 2022-07-13:
@Qowy das wird in `eraseSettings()` mit `uint8_t buf[64] = {0};` und `mEep->write(addr, buf, (end-addr));` entsprechend initialisiert.

https://github.com/grindylow/ahoy/blob/c35c86210f8507be7352ead73c02e5855f3f9d63/tools/esp8266/main.h#L74-L88

Vielleicht kann @lumapu noch was dazu sagen, warum das evtl. in der GUI als UTF-8 interpretiert (richtig) und als 0x00FF bzw `ÿ` (falsch) dargestellt bzw. gespeichert wird/wurde.

Qowy, 2022-07-13:
Ja das habe ich auch gesehen, mein bauchgefühl sagt mir dass das einfach nie passiert.

Ich kann mich zumindest nicht erinnern jemals `DPRINTLN(DBG_INFO, F("Settings not valid, erasing ..."));` in der Konsole gesehen zu haben

`DPRINTLN(DBG_INFO, F("Welcome to AHOY!"));` aber schon also ist das Debug level schon info

stefan123t, 2022-07-13:
Wenn die CRC Checksum nicht passt sollte er das EEPROM beim ersten Start erasen und dann neu starten. 
So habe ich das in der Vergangenheit zumindest beobachten können. Ich weiß nur nicht ob er das mit `eraseSettings(false)` oder `eraseSettings(true)` macht. Ich nehme an dass er es da `eraseSettings(true)` setzt und die WiFi Passwörter eben nicht überschreibt bzw. verschont.

@lumapu, hier sollte vielleicht doch einmal initial überschrieben werden, bis die Checksumme auch für die WiFi Credentials passt ?

Qowy, 2022-07-13:
Ich denke auch das Wifi einfach nie überschrieben wird

das tut ja wie du gesagt hast nur `eraseSettings(true);`
Das einzige mal wo das im code vorkommt ist aber in `void Main::showFactoryRst(void)`

Also wird Wifi nie überschrieben beim ersten setup oder?

Qowy, 2022-07-13:
Also nach noch etwas stöbern werden die wifi einstellungen initial gecheckt in `bool Main::getConfig(void) {`

Da schaut das hier schonmal verdächtig aus `mWifiSettingsValid = checkEEpCrc(ADDR_START, ADDR_WIFI_CRC, ADDR_WIFI_CRC);` 

`bool checkEEpCrc(uint32_t start, uint32_t length, uint32_t crcPos) {`

die length der daten = die addresse des CRC werts? Das kann doch nicht stimmen oder?

Qowy, 2022-07-13:
Wenn ich es richtig verstehe, sollte `bool Main::getConfig(void) {` eigentlich einen CRC check machen.

Kann es sein, dass der CRC für den unitialisierten speicherbereich einfach zufällig passt?

Der Fehler kommt dann definitiv durch

```
if(mWifiSettingsValid) {
        mEep->read(ADDR_SSID,    mStationSsid, SSID_LEN);
        mEep->read(ADDR_PWD,     mStationPwd, PWD_LEN);
        mEep->read(ADDR_DEVNAME, mDeviceName, DEVNAME_LEN);
    }

```

Denn es sind seeeeehr lange strings dann im webUI also liest er einfach bis leng ohne null terminator.

Also bleibt die frage warum ist  `mWifiSettingsValid` true

stefan123t, 2022-07-13:
Die Frage nach den beiden Parametern in `mWifiSettingsValid = checkEEpCrc(ADDR_START, ADDR_WIFI_CRC, ADDR_WIFI_CRC);` ist/war m.E. gar nicht so verkehrt. Jetzt ist der Kommentar oben geändert.

Die Werte werden in defines.h berechnet aufgrund der Feldlängen. Aber warum wird für Länge und CRC Position das selbe verwendet ?
Außerdem hast Du doch gesagt es wäre der ganze Speicher mit 0xFF initialisiert und evtl. ist die CRC Checksum von 0xFF FF FF FF .. .FF immer 0xFF ?

@lumapu was meinst Du ?

Qowy, 2022-07-13:
Den kommentar habe ich zurückgenommen, weil mir aufgefallen ist, dass ADDR_START auf 0 definiert ist. Dann ist die länge bis zur CRC adresse tatsächlich gleich der CRC addresse

Qowy, 2022-07-13:
Also hier mal der Output eines fabrikneuen D1

Wie man sieht ist der CRC tatsächlich identisch, also entweder weil die initialisierung so läuft, oder ein bug in der crc berechnung

```
--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
;l␀d��|␀�l�|␃␌␄␌�␌$�␄c|��␃�␓�s�c�␌b��o'�dog���␄c␜p��${l{lx�o�␘␃␌␄�␄l␄��␌␄␄c␄g�|␂�␌��␄c��'o�␀$��l`␃�␛␓ng␄�␏l␇␃or�ۓo␌␄r{�`␃x�o�␐␃␄␄s�ܜ�␜��␃␌#␌'�|␃␄�␄c��'o�␀␌�l ␃�␓␛gn␌�␇d␏␃'{���g␄␌{␒d`␃␏;��'␌␄;␓$`␃�␂␄␌d��<�␄$`␃��n�␇lV: Main::Main
V: app::app():Main
V: hmRadio.h : HmRadio():mNrf24(CE_PIN: 2, CS_PIN: 15, SPI_SPEED: 1000000)
V: app::setup
V: Main::setup
V: Main::getConfig
V: main.h:checkEEpCrc
D: start: 0, length: 112
V: main.h:buildEEpCrc
D: CRC RD: 4167 CRC CALC: 4167
V: main.h:checkEEpCrc
D: start: 114, length: 415
V: main.h:buildEEpCrc
D: CRC RD: 4e53 CRC CALC: 4e53
V: Main::setupStation
I: connect to network '72c6d785d137ead7a838e6a11f19f01c����������������������������������������������������������������␁␁␁' ...
...............................
....................................................................................................
V: mqtt.h:setup
V: hmSystem.h:setup
V: hmRadio.h:setup
```

stefan123t, 2022-07-13:
Das vor Main::main ist der ESP bootloader, der schreibt mit 9600 baud auf die Serial Console. Dafurch dass wir 115200 baud verwenden ist Deine Serial Console nicht in der Lage die Daten anzuzeigen.
Das mit dem passenden CRC ist genau was Du und ich vermutet haben. Evtl müssen wir statt CRC einen anderen Hash Algorithmus verwenden oder die beiden von Dir genannten CRCs blacklisten als initial leer. Aber das würde sich bei einer Änderung der EEPROM Struktur auch wieder ändern.
Hmm, gute Frage wie wir die leeren Konfigdaten erkennen können ... ?

Qowy, 2022-07-14:
ich mache mir noch etwas sorgen um diesen Debug output: `D: CRC RD: 4167 CRC CALC: 4167`

CRC RD sollte doch eigentlich im speicher stehen als unitialisierter müll, wie kann es dann sein dass genau das bei crc calc rauskommt.

Da stimmt doch was nicht oder?

stefan123t, 2022-07-14:
@Qowy mE müsste man in buildEepCrc() ein paar Debug Optionen vor oder nach Zeile 57 einbauen. 

```
DPRINTLN(DBG_DEBUG, F("buf: ") + String(buf, HEX) + F(", len: ") + String(len) + F(", crc: ") + String(crc, HEX));
String(buf, HEX)
crc = crc16(buf, len, crc);
DPRINTLN(DBG_DEBUG, F("crc16: ") + String(crc, HEX));
```

und dann den Test wiederholen. Dann können wir dir Funktion nachvollziehen.

https://github.com/grindylow/ahoy/blob/c35c86210f8507be7352ead73c02e5855f3f9d63/tools/esp8266/main.h#L48-L62

lumapu, 2022-07-14:
ich habe die CRC Funktion zusammen mit der `buildEepCrc` Funktion getestet. Über einen Speicherbereich, gefüllt mit 0xff bekommt man für jede Länge einen anderen CRC, es geht also auch mit uninitialisierem Speicher.

Ich denke, der ESP versucht daraufhin zu löschen und überschreibt den CRC, bootet neu und genau hier fangen eure Untersuchungen an.

Irgendwas geht scheinbar beim löschen des Speichers schief, evtl. darf man nicht zu früh neu starten nach dem Löschen.

Qowy, 2022-07-14:
D.h. dass die `eraseSettings `Methode beim ersten starten was falsch macht?
Aber die wird ja außer beim expliziten aufrum über webui nie mit (true) aufgerufen, sollte also den bereich vor `ADDR_START_SETTINGS` nicht anfassen

Qowy, 2022-07-14:
Na endlich ich habe es denke ich gefunden.

Beim start mit `mSettingsValid = false` wird das hier ausgeführt:

https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/app.cpp#L172-L182

eraseSettings fasst die Wifisettings nicht an da es nicht mit true aufgerufen wird.
`saveValues(false)` landet irgendwann bei diesem Code:

https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/app.cpp#L780-L782

https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/main.cpp#L261-L282

und wie man sieht wird hier schön `updateCrc();`  aufgerufen, was dann eine gültige CRC für schmarn schreibt.

https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/main.cpp#L295-L302

stefan123t, 2022-07-14:
@Qowy ich denke Du hast Recht! vielleicht kannst Du mal Zeile 873 & 874 auskommentieren ?
@lumapu warum schreibt er hier einen neuen CRC und committed das Ergebnis ins EEprom obwohl er eigentlich einen Fehler meldet ?
https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/app.cpp#L872-L877

lumapu, 2022-07-15:
@stefan123t ich glaube wir hatten hier vorher eine reboot loop.
Das Problem ist dass Auslesen des Speichers. Um sicher zu gehen, das bei falschem CRC kein uninitialisierter Speicher gelesen wird, habe ich den Speicher gelöscht, den CRC aktualisiert und neu gestartet. Evtl. kann man diesen reboot einsparen, indem man das Eeprom zwar zurücksetzt, aber keinen CRC aktualisiert. Man muss sicher gehen, das alle lesenden Speicheroperationen erst nach dem zurücksetzen stattfinden.

stefan123t, 2022-08-12:
@lumapu das ist im Prinzip analog zu dem Reset to Defaults in #114.
Wir sollten m.E. nur eines der beiden Issues behalten. Danke

lumapu, 2022-08-12:
dann nehmen wir den neueren

closed 2022-08-12 by lumapu