lumapu/ahoy #104: MQTT HomeAssistant Autodiscover nicht durchgängig / Erweiterung um Info zu Hersteller, Modell möglich?

fila612, 2022-07-14:
Hallo,

Ich habe voller Freude die Änderung in #85 gesehen und konnte es schon nutzen.
Leider gibt es derzeit damit mehrer "Problemchen", aber erstmal als Frage formuliert, da ich nicht ausschließen kann, dass der Fehler evtl. vorm Rechne sitzt.

1) ich habe mehrere MQTT Geräte, auch von Topic HomeAssisten (Autodiscover). Der Anteile Ahoy ist aber nicht mehr in der Ordnerstruktur von HA zu finden, sondern außerhalb und ohne Ordnername:
[image]
=> Zu Beginn erschien der HM-700 auch im Bereich "homeassistant".

2) Die Werte in HA sind sporadisch als nicht verfügbar dargestellt, auch kann ich die Abfragezeiten im SetupTeil der Ahoy GUI nicht mehr ändern (Aktuell bei 4.25 ist dies fest auf 75s eingestellt). 

3) Nice to Have:
Könnte man die MQTT Daten um die Werte Hersteller und Modell (idealeweise aus den Feldern im Setup) ergänzen. Aktuell steht da bei mir im HA jeweils "unknown".

Falls ich etwas vergessen haben sollte oder noch logs, etc. gebraucht werden hänge ich diese gern an.

Besten Dank schon mal vorab

Und: Tolle Arbeit an alle Beteiligten :) DANKE

fila612, 2022-07-14:
Ich habe nun den WemosD1 als auch den MQTT-Broker einmal rebootet.
Jetzt bekomme ich die Infos doppelt, ich habe jedoch KEIN Topic in dem Setup eingetragen:
[image]

Ist das so gewollt bzw. richtig oder mach ich hier etwas falsch?
Ich vermute, dass der untere Teil (ohne Hauptordner) derjenige ist, welche ohne HA Autodiscover published wird.
Der Anteil im Ordner "homeassistant" ist derjenige, welcher ab #85 möglich ist.
Sollte man hier nicht eine entweder oder Funktion in der GUI hinterlegen?
Interessant sind auch die unterschiedlichen Quantitäten an Meldungen. (siehe die Werte in Klammern).

fila612, 2022-07-14:
Nach über 2h werden wieder keine "HA-Relevanten" MQTT Werte übertragen. auch im HA werden die Entitäten dann als nicht verfügbar angezeigt, hier mal am Beispiel der AC-Leistung:
[image]
.
Die Lücke basierte auf dem ursprünglich beschriebenen Verhalten, der gerade Strich rechts ist der Wegfall nach 2h von eben.

stefan123t, 2022-07-14:
@KG3RK3N Kai hatte das Feature implementiert. Vielleicht kann er etwas zu Deinem Problem beitragen, bzw. Tipps für das weitere Debugging geben ?

fila612, 2022-07-15:
Das wäre super.

was mir ergänzend noch aufgefallen ist:
Die Messages für den HO-Autodiscover werden aus meiner Sicht nur einmal an den MQTT-Server geschickt (kann man durch Reboot triggern).
Ich habe neben der AHOY auch noch Zigbee2Mqtt am laufen, welchen ebenfalls eine HA-Integration besitzt.

Wenn ich nun mit einem MQTT-Tool auf den Broker schaue, sehe ich den Zigbee2Mqtt Anteil für HA sofort, aber den AHOY-Anteil für HA nur dann, wenn ich (während das Tool mit dem Broker verbunden ist) Den Wemos neustarte.
Irgendwie muss es da unterschiede geben. -ich bin aber technisch leider nicht begabt um da selbst zu korrigieren.

KG3RK3N, 2022-07-17:
@fila612 

1. Das macht für mich so keinen Sinn, der "Ordner" (HM-700) könnte nur dann im homeassistant Ordner sein wenn auch der MQTT Topic entsprechend gesetzt wurde.
2. Mir ist auch schon aufgefallen, das es in der neusten Version nicht mehr möglich ist das Interval anzupassen übers Setup. Ich habe es so implementiert, das die Werte nach Interval + 5s ablaufen. Dort habe ich allerdings auch schon festgestellt, das es sehr knapp bemessen ist. Hier würde ich das auf Interval * 2 umstellen, das sollte deutlich unproblematischer sein.
3. Jap, werde ich die Tage mal ergänzen.

> Ich habe nun den WemosD1 als auch den MQTT-Broker einmal rebootet. Jetzt bekomme ich die Infos doppelt, ich habe jedoch KEIN Topic in dem Setup eingetragen: [image]
>
> Ist das so gewollt bzw. richtig [...]

Richtig, der Teil ohne Ordner ist der Teil ohne das auto discovery, das habe ich bewusst so erhalten damit bestehende Konfiguration weiterhin funktionieren. Das die Messages ohne Hauptordner kommen liegt daran, weil keine Abfrage implementiert ist ob dieser Wert gesetzt ist. Sollte man durchaus abfangen.

Auf dem folgenden Bild siehst du eine Discovery Config Message die im home assistant Ordner abgelegt wird. Diese werden nur einmalig verschickt damit der Broker nicht unnötiger weise zugespamt wird und es auch den ESP etwas entlastet. Zusätzlich dazu kann man sehen, das der state auf den anderen Ordner verweist, wo dann der tatsächliche Wert ist wodurch das erneute Senden der Config entfällt und die tatsächlichen Werte nicht mehrfach herausgegeben werden. Home Assistant greift dadurch dann ebenfalls auf die bereits vorhandene MQTT Implementierung zu.
[image]

> Nach über 2h werden wieder keine "HA-Relevanten" MQTT Werte übertragen. auch im HA werden die Entitäten dann als nicht verfügbar angezeigt, hier mal am Beispiel der AC-Leistung: [image] . Die Lücke [...]

Da diese Werte wie vorher beschrieben nicht über die Discovery Config übertragen werden, würde ich es jetzt erstmal noch nicht auf die Auto Discovery funktion schieben. Vlt. könntest du ja mal schauen, wann die letzte Message von den tatsächlichen Werten übertragen wurde. War der Broker vlt. kurzzeitig nicht erreichbar? Mehrere Benutzer haben hier #106 bereits Probleme bestätigt.

Leider habe ich meinen ESP noch nicht im Dauereinsatz, ordentliches Netzteil kommt aber Anfang der Woche hoffentlich und dann sehen wir mal welche Krankheiten der kleine noch hat 😄

stefan123t, 2022-07-17:
@KG3RK3N Hallo Kai,
> Mir ist auch schon aufgefallen, das es in der neusten Version nicht mehr möglich ist das Interval anzupassen übers Setup. Ich habe es so implementiert, das die Werte nach Interval + 5s ablaufen. [...]

Das Abfrage Interval habe ich mit 30 Sekunden so angepaßt, damit es nicht zu niedrig angesetzt wird (>15 s) und evtl. mehrere WR parallel antworten, was dann den ESP8266 beim Zusammensetzen der Antwort (Payload) durcheinander bringen würde. 
In der Hoymiles DTU-PRO ist das Intervall für die Leistungsbegrenzung übrigens auch auf 15 Sekunden vorbelegt. 

Es sollte sich aber nach wie vor anpassen lassen über das Setup ? 
Oder ist das, was Du meinst das MQTT Intervall, das i.d.R. an das aktuelle Intervall der WR gekoppelt ist (mehr Datenpunkte zu senden als vom WR empfangen macht ja keinen Sinn), daher hat das Lukas dementsprechend ebenfalls gekoppelt ?

KG3RK3N, 2022-07-17:
@stefan123t Ich müsste es nochmal prüfen, hab's nicht mehr exakt in Erinnerung wie es war. Ich denke Mal der Fehler war vor dem Bildschirm und ich hab einfach einen zu kleinen Wert versucht weshalb es nicht geklappt hat. 😄 

Ne, macht schon Sinn das es gekoppelt ist.

fila612, 2022-07-18:
Hallo @KG3RK3N,

Danke für deine ausführlichen Antworten.
Ich habe nun selbst in den letzten Tagen etwas weiter beobachten und getestet.

1. den Hersteller & Modell habe ich in der app.cpp (in 4.25 ab Zeile 900) im Bereich "deviceDoc" ergänzt:
` deviceDoc["mf"] = "Hoymiles";
   deviceDoc["mdl"] = iv->name;'
   
2. Was mir aufgefallen ist: Wenn ich den HA Neustarten muss, z.B. wenn ich an der HA-configurations.yaml schraube, findet der HA leider den WR nicht mehr, da das Autodiscover nicht mehr via MQTT gesendet wird. Ich musste immer den Wemos Neustarten, damit er die Daten wieder hat. Parallel habe ich auch Zigbee2Mqtt, da sind die HA-Autodiscover immer da (prüfe das mit einem MQTT-Explorer Tool).

3. bzgl. der Intervalle: Ich glaub ich verstehe die einzelnen Punkte im Setup bzw. auch im Code noch nicht ganz:
a) im Setupbereich MQTT gibt es die letzte Zeile "Interval [s] (read-only)", welche derzeit den default-Wert 75 hat. ich vermute (Tuch Beaobachtungen), dass dies der Zyklus ist, indem MQTT Daten published werden.

b) in der "config.h" gibt es "#define MQTT_INTERVAL" => aktuell von mir auf 10 gestellt
c) in der "defines.h" gibt es "#define MIN_MQTT_INTERVAL"=> aktuell von mir auf 10 gestellt

=> nun habe ich in dem Feld aus a) einen Wert von 20....hm

Wie gesagt, ich bin fürs coden nicht die Idealbesetzung (eher T&E), schaue aktuell mehr aus testen und beobachten.
Ich denke, dass in diesem Issue der Punkt 2 aus diesem Kommentar aus meiner Sicht am ehesten geprüft werden sollte. Kann man je selbst einfach bei sich testen (der Wemos sollte aber eine Zeit lang schon laufen).

Ich danke euch schon mal. :)

fila612, 2022-07-18:
> Da diese Werte wie vorher beschrieben nicht über die Discovery Config übertragen werden, würde ich es jetzt erstmal noch nicht auf die Auto Discovery funktion schieben. Vlt. könntest du ja mal [...]
Kann ich eigentlich ausschließen, da anderen Werte (Zigbee2Mwtt) durchgehend übertragen wurden. Daher würde ich für mein Issue den Broker nicht als Ursache sehen. Ich konnte es ja lösen, in dem ich den Wemos neustartete. Dann waren die Autodiscover-Einträger wieder da und die Daten in HA auch.

KG3RK3N, 2022-07-18:
> 1. den Hersteller & Modell habe ich in der app.cpp (in 4.25 ab Zeile 900) im Bereich "deviceDoc" ergänzt:
>    ` deviceDoc["mf"] = "Hoymiles";
>    deviceDoc["mdl"] = iv->name;'

Sieht schon mal grundsätzlich gut aus, allerdings denke ich das iv->name hier nicht angebracht ist da es ja der selbst gewählte Name ist und nicht das tatsächliche Model. Bedeutet, wenn ich da "Inverter 1" eintrage würde das auch im Model stehen. Da wir bereits das Model erkennen können anhand der S/N würde ich das eher darüber machen. Zusätzlich was auch wichtig ist das die größe von doc und der maximalen MQTT Message Length angepasst wird, da es sonst dazu kommen kann das ArduinoJson Werte nicht mehr mit rein nimmt und dadurch welche fehlen. Aktuell liegt er bei 384. ArduinoJson hat auch einen Online Kalkulator wo du einfach mal eine Message angeben kannst, um den idealen Wert + etwas spielraum zu ermitteln.

> 2\. Was mir aufgefallen ist: Wenn ich den HA Neustarten muss, z.B. wenn ich an der HA-configurations.yaml schraube, findet der HA leider den WR nicht mehr, da das Autodiscover nicht mehr via MQTT [...]

Guter Punkt den ich so nicht getestet habe, werde ich mal ausprobieren wie es sich verhält. Allerdings müsstest du mal schauen ob im MQTT die Werte (nicht Discovery Config) überhaupt nach einem Neustart gesendet werden, den das sind eigentlich die Werte worauf HA nun schaut. Werden diese nicht mehr verschickt, wird in HA der Wert als unbekannt bzw. nicht verfügbar gekennzeichnet. Bei mir ist es scheinbar so, das Ahoy grundsätzlich Probleme hat mit dem reconnect. Wo wir dann wieder bei #106 landen würden. Genaure Analyse erfolgt aber noch meiner seits was da los ist.

KG3RK3N, 2022-07-18:
Um es vlt. ein weniger besser auf den Punkt zu bringen, ich versuche gerade heraus zu finden ob es tatsächlich das Auto Discovery ist, weil es nur einmalig versendet wird oder ob es an den tatsächlichen Werten liegt welche vlt. nicht mehr verschickt werden weil der reconnect nicht funktioniert.

fila612, 2022-07-18:
Verstehe, ich teste das jetzt nochmal genau, aber meine bisherigen Beobachtungen deuten auf den Autodiscover hin, weil:
a) die "normalen" Werten kommen durch AHOY weiter an
b) der Autodiscover ist nicht mehr vorhanden, weil die Message zu alt und "retained" ist (ich sehe sie auch nicht mehr im MQTT-Tool)

Wenn ich nun den HA Neustarte kann er die "normalen" Werte nicht mehr zuordnen bzw. zeigt an, dass das Gerät nicht verfügbar ist.

ich teste das aber nochmal genau.

KG3RK3N, 2022-07-18:
Gut, dann müssen wir es abändern das bei jedem connect/reconnect die Config erneut gesendet wird.

KG3RK3N, 2022-07-18:
Was du mal machen kannst, nur um das zu bestätigen und zum Testen einfach mal die Bedingung abändern und das `&& mMqttConfigSendState[id] != true` entfernen https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/app.cpp#L899 

Das würde dazu führen, das er bei jedem MQTT Interval die Config erneut sendet. Würde ich allerdings nur so zum Testen mal machen, final sollte es beim connect/reconnect ausgeführt werden damit wir die config nicht in jedem Interval verschicken.

wenzlaff, 2022-07-18:
Für diese Art von Nachrichten (Auto discovery) sollte das sogenannte "retain flag" gesetzt werden, das sorgt dafür dass die letzte jeweils die letzte Message in dem Topic im Mqtt Broker verbleibt.

KG3RK3N, 2022-07-18:
@wenzlaff Ah, danke für den Hinweis. Wird ausprobiert 👍

fila612, 2022-07-18:
> Für diese Art von Nachrichten (Auto discovery) sollte das sogenannte "retain flag" gesetzt werden, das sorgt dafür dass die letzte jeweils die letzte Message in dem Topic im Mqtt Broker verbleibt.
> Mehr Infos z.B. hier: https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/

ich glaube das ist die Lösung :)

KG3RK3N, 2022-07-18:
@fila612 Schau mal hier https://pubsubclient.knolleary.net/api#publish
Ist der 3. Parameter, muss dann in der mqtt.h als möglicher Parameter ergänzt werden damit man das beim senden dann übergeben kann.

KG3RK3N, 2022-07-18:
Mal fix aus dem Ärmel: 
mqtt.h
```
        void sendMsg(const char *topic, const char *msg) {
            //DPRINTLN(DBG_VERBOSE, F("mqtt.h:sendMsg"));
            char top[64];
            snprintf(top, 64, "%s/%s", mTopic, topic);
            sendMsg2(top, msg, false);
        }

        void sendMsg2(const char *topic, const char *msg, boolean retained) {
            if(mAddressSet) {
                if(!mClient->connected())
                    reconnect();
                if(mClient->connected())
                    mClient->publish(topic, msg, retained);
            }
        }
```

Und hier dann einfach `true` ergänzen
https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/app.cpp#L933

fila612, 2022-07-18:
habs grad mal eingefügt, kompiliert und geflashed. laut dem MQTT-Tool sind das noch keine "retained" Nachrichten.
Zum Vergleich eine Nachricht von Zigbee2Mqtt:
[image]

und dann die aktuelle von AHOY:
[image]

Bei letzteren fehlt das gelbe Retained Schild.

ich beobachte aber mal, evtl. bleiben die Nachrichten ja dennoch drin.

fila612, 2022-07-18:
Kommando zurück, scheint zu klappen (gerade nochmal neu verbunden):
[image]

Könnte man ja mal als PR einkippen

KG3RK3N, 2022-07-18:
> Kommando zurück, scheint zu klappen (gerade nochmal neu verbunden): [image]
> 
> Neustart das HA ist nun ohne Probleme - Daten sind sofort verfügbar :) Super - vielen Dank Könnte man ja mal als PR einkippen, oder?

Klar, rein damit 👍

fila612, 2022-07-18:
> 2. Mir ist auch schon aufgefallen, das es in der neusten Version nicht mehr möglich ist das Interval anzupassen übers Setup. Ich habe es so implementiert, das die Werte nach MQTT-Interval + 5s ablaufen. Dort habe ich allerdings auch schon festgestellt, das es sehr knapp bemessen ist. Hier würde ich das auf MQTT-Interval * 2 umstellen, das sollte deutlich unproblematischer sein.

=> evtl. als Hinweis: im Rahmen meiner Analyse & Tests habe ich folgenden Zeile auskommentiert:
https://github.com/grindylow/ahoy/blob/6dba1d757707426e1c8ff0c4d9d2ce5d22d6f26a/tools/esp8266/app.cpp#L924

fila612, 2022-07-25:
ich mach hier mal zu und schiebe die Themen um Hersteller und MF mal in einen separaten Issue.
das Intervall bzgl. exp_aft, wo @KG3RK3N noch Änderungen machen wollte pack ich da auch mit rein.

closed 2022-07-25 by fila612