lumapu/ahoy #106: MQTT Verbindung wird nach Neustart des Brokers nicht mehr hergestellt

JauntyJosef, 2022-07-16:
Ich habe die Version 0.4.25 installiert. Da bei mir ein Server, auf dem der MQTT Broker läuft, jede Nacht für Backups kurz runterfährt und wieder neu startet, bemerkte ich, dass die Daten danach nicht mehr per MQTT ausgeliefert werden. Bei der Weboberfläche steht: "MQTT: not connected". Nach einem reboot von Ahoy funktioniert es wieder.

timtimsson, 2022-07-16:
… kann ich bestätigen.

der2of6, 2022-07-17:
kann den Effekt ebenfalls bestätigen

thinklerde, 2022-07-21:
Auch wenn  die Internetverbindung weg war verbindet sich der MQTT ebenfalls nicht neu.

timtimsson, 2022-07-23:
> … kann ich bestätigen.

ein **lwt** wäre auch nicht schlecht, falls die Verbdg verloren geht
mClient->publish(topic, msg, retained);

stefan123t, 2022-07-25:
@timtimsson was meinst Du mit lwt ?

fila612, 2022-07-25:
ich denke, das hier das "Last Will and Testament" Feature von MQTT gemeint ist.
Von dem was ich gelesen habe, könnte das auch die Lösung für das Problem hier zu sein.
Code-technisch bin ich eine Niete :( kann hier nicht viel beisteuern.
in der API des PubSubClient gibt es die Will-Parameter bei Connect: https://pubsubclient.knolleary.net/api#connect

Allgemeine Infos zu LWT sind hier nochmal gut dargestellt: https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

timtimsson, 2022-07-25:
… ja genau das meine ich. Mit lwt kannst du schauen, ob z.B. öfters die Verbindung verloren wurde oder aktiv überwachen, ob der „ESP“ noch online ist…
Lwt wird ganz oft genutzt, Projekte wie tasmota oder easyEsP.
https://nodemcu.readthedocs.io/en/dev/modules/mqtt/

HorstG-57, 2022-07-25:
...Auch wenn ich hier neu bin , möchte ich mal einenen Vorschlag zum MQTT bringen.

a: 
das Lwt setzt einen Topic in dem Moment, wo der MQTT Server erkennt, das ein Publisher ( hier der Ahoy ) 
unerwartet die Verbindung zum MQTT-Server verlorren hat. 
Dieses Topic verfällt nicht, so das ein neuer Subscriber der dieses Topic abfragt den letzten Inhalt erhält.
Wenn man jetzt einen Topic "status" erstellt dieses nicht verfallen läst, und vom LWT dieses auf den Wert "unbekannt" setzen läst und vom Ahoy beim Anmelden am MQTT Server dieses Topic auf "Online" bzw. beim Abmelden auf "Offline" setzen läst, 
dann können die Subscriber sauber erkennen ob der Ahoy noch lebt.

b:
Ich habe ( wie einige andere ) auch das Problem, 
das im Fall, das der MQTT-Server zwischendurch nicht erreichbar war, 
der Ahoy sich auf "MQTT disconnected" setzt und sich dann nicht wieder verbindet. 
Der Ahoy müste eigentlich vorm senden der MQTT Pakete prüfen, ob er noch verbunden ist. 
Sollte das nicht der Fall sein, muss er sich neu mit dem MQTT-Server verbinden.

c: ( SendeInterval der MQTT Meldungen )

Wieso läst man den Ahoy nicht einfach immer mit einen Zeitversatz von einigen Sekunden ( z.B. 2 ), 
nachdem er die Daten vom Wandler erhalten hat, 
die MQTT Daten für dieses Wandler an den MQTT-Server aussenden, 
z.B. durch ein   "mMqttTicker = mMqttInterval -2" an Ende der Datenaufbereitung der Wandler Rohdaten ?
Werden keine Daten empfangen, sendet der Ahoy in seim normalem MQTT Intervall, 
kommen Daten an werden die sofort zum MQTT übertragen.

d:
Die Übertragung der "DiscoveryConfigDaten" vom Ahoy sollte in Setup abschalbar sein. 
Die sind doch nur von Intresse, wenn man die "homeassistant" Software einsetzt, oder ?

KG3RK3N, 2022-07-28:
@HorstG-57 

> b:
> Ich habe ( wie einige andere ) auch das Problem,
> das im Fall, das der MQTT-Server zwischendurch nicht erreichbar war,
> der Ahoy sich auf "MQTT disconnected" setzt und sich dann nicht wieder [...]

Genau das tut er eigentlich schon wie man hier sehen kann und dennoch scheint es nicht richtig zu klappen:
https://github.com/grindylow/ahoy/blob/ed4a99bf7feb9801af7cd02273a1da1a27fff732/tools/esp8266/mqtt.h#L47-L50

> d:
> Die Übertragung der "DiscoveryConfigDaten" vom Ahoy sollte in Setup abschalbar sein.
> Die sind doch nur von Intresse, wenn man die "homeassistant" Software einsetzt, oder ?

Absolut richtig und bin ich bei dir, das könnte man ruhig im Setup ergänzen.

HorstG-57, 2022-07-28:
Ich habe mal ein wenig mit Debugausgabe dem reconnect() nachgeforscht.
Für mich sieht das so aus, das der reconnect()  auch Richtig aufgerufen wird, 
aber in der PubSubClient::connenct nicht richtig ausgeführt wird. 
Auch ein zusätzlich eingebautes PubSubClient::disconnect bringt keine Besserrung.

HorstG-57, 2022-08-01:
Habe noch ein wenig weiter versucht dem Problem sein Geheinis zu entlocken, 
das Problem scheint aber tiefer in den Netzwerk Proceduren vom TCP-Stack zu liegen.

Die Reconnect Procedure in der mqtt.h wird sauber aufgerufen, nur 
bekomme der TCP_Client der unter der MQTT Verbindung liegt, die 
Verbindung nicht wieder aufgebaut ( Soweit konnte ich das bis jetzt 
nochvollziehen ).
Nach einer Verbindungsunterbrechung bringt
der MQTT Client sauber den Status -3 ( LOSS CONNECTION ) und
der TCP-Client bringt den Status 0.
Danach bekommt der TCP Client bzw. der MQTT Client die Verbindung nicht 
wieder aufgebaut. Ich habe es hier schon mit einem Disconnect probiert ( 
nachdem festgestellt wurde das die MQTT Verbindung abgebrochen ist ) 
bzw. mit einer Verzögerung vom bis zu 60 Sekunden bevor ein neuer 
connect Versuch unternommen wird. Auch ein "flush" und ein "stop" der 
TCP Verbindung haben keine Verbesserung der Situation gebracht.

stefan123t, 2022-08-02:
@Horst-G versuch doch mal die ESP_DEBUG_* Flags zu setzen um WiFi+SSL+TCP zu debuggen:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html

ESP_DEBUG_CORE
ESP_DEBUG_WIFI
ESP_DEBUG_MDNS
ESP_DEBUG_...

https://github.com/esp8266/Arduino/blob/04c2322721f6865efe0c518be57e795e8643c183/tools/boards.txt.py#L1308-L1309

stefan123t, 2022-08-03:
Ich fand den folgenden Beitrag und den Hinweis auf die [PlatformIO Espressif 8266 Dokumentation](https://docs.platformio.org/en/latest/platforms/espressif8266.html#serial-debug) hierzu ganz hilfreich 

**Debugging WiFiClient with an ESP8266 on PlatformIO**
https://community.platformio.org/t/debugging-wificlient-with-an-esp8266-on-platformio/9891/2

und habe meinen Debug Level in der platformio.ini unter meiner Zielplattform gesetzt:
```
[env:nodemcu_v2]
build_flags = -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_OOM -DDEBUG_ESP_PORT=Serial
```
https://github.com/grindylow/ahoy/blob/main/tools/esp8266/platformio.ini#L4-L31

HorstG-57, 2022-08-03:
Habe einen Fork erstellt ( horstg-57/ahoy ) , die Sourcen überarbeitet und das ganze hochgeladen, 
eine PULL Request habe ich aufgemacht.

( hat leider etwas gedauert, musste mich erst mal mit Guthub und Git  auseinandersetzen ).

stefan123t, 2022-08-03:
@HorstG-57 so wie @fila612 vorschlägt mache ich das auch. Fork erstellen und einen API Key registrieren. Dann einen git push auf die eigene Repo URL (`git push https://github.com/HorstG-57/ahoy/`) und dann einen PR in Deiner github Oberfläche erstellen.

HorstG-57, 2022-08-03:
ich habe eine Stelle eingebaut 
( app.cpp; sofortigen versenden der Daten per MQTT nachdem die Datem vom WR aufbereitet wurden) 
für die noch jemand einen Schalter im Setup einbauen müsste , so das man das je nach eigenen Wünschen aktivieren und deaktiviern kann.

Es währe ausserdem schön, wenn der jenige dann auch gleich ein Feld mit aufnehmen würde, 
wo man die Adresse des NTP-Servers eintragen kann.
 
Die Daten sollten natürlich auch im EPROM gespeichert werden.

stefan123t, 2022-08-03:
@baloo hat das Problem gerade mit mir gemeinsam untersucht.
Die Strings die in Zeile 92 und 94 an den PubSubClient übergeben werden sind nur temporär auf dem Speicher.
https://github.com/grindylow/ahoy/blob/c35c86210f8507be7352ead73c02e5855f3f9d63/tools/esp8266/mqtt.h#L88-L96 

Der PubSubClient merkt sich aber eine Referenz const char* domain; darauf.
Wenn er dann das nächste Mal was senden soll und will, dann ist an der Stelle im Speicher kein Domain Name / IP mehr.
Man sollte also den mBroker fest initialisieren:
```c
            memset(mBroker,  0, MQTT_ADDR_LEN);

            snprintf(mBroker, MQTT_ADDR_LEN, "%s", broker);
            
        char *getBroker(void) {
            //DPRINTLN(DBG_DEBUG, F("mqtt.h:getBroker"));
            return mBroker;
        }

        char mBroker[MQTT_ADDR_LEN];
```

stefan123t, 2022-08-03:
Das hast Du m.E. in Deinem Commit auch gemacht. Ich hatte bei mir die mqttAddr verwendet, eventuell klappt das nicht.

HorstG-57, 2022-08-03:
Das Problem ist nur, das die Refernz bei einem "LOS_Connect" leider auch kaputt geht 
und somit die Broker-Adresse neu gesetzt werden muss.
Darum merke ich mir in meiner Änderung die Broker-Adresse und setze im reconnect die Server-Adresse neu 
( siehe mqtt.h zeile 141 und 142 ).

stefan123t, 2022-08-03:
@HorstG-57 habe dem mClient->setServer Aufruf auch noch übernommen.

commit 1a679130 (2022-08-04): fix #106 PubSubClient reconnect, which maintains a reference to char* mAddr
commit f657ace3 (2022-08-04): fix #106 handpickled mDevName changes from #120
closed 2022-08-04 by lumapu