lumapu/ahoy #110: add handling for frame 0 response

stefan123t, 2022-07-25:
According to [OpenDTU #29 connection problems](https://github.com/tbnobody/OpenDTU/issues/29) the inverter sometimes answers with a Frame ID being zero instead of e.g. 84 being the last frame. This may cause problems in the payload decoder and may cause wrong MQTT messages.

Please see https://github.com/tbnobody/OpenDTU/commit/c6499e09bd18672e5283106feb72cd899e6ad3c4 for a plausible fix in OpenDTU.

Can this even prevent MQTT/WiFi stack from reconnecting ?

KG3RK3N, 2022-07-28:
Seems that we read the fragment number here:
https://github.com/grindylow/ahoy/blob/ed4a99bf7feb9801af7cd02273a1da1a27fff732/tools/esp8266/app.cpp#L233-L245

I think it would have to look like this to catch it:
```c++
uint8_t *pid = &p->packet[9];
if (*pid == 0x00) {
    DPRINT(DBG_DEBUG, "fragment number zero received and ignored");
} else {
    if((*pid & 0x7F) < 5) {
        memcpy(mPayload[iv->id].data[(*pid & 0x7F) - 1], &p->packet[10], len-11);
        mPayload[iv->id].len[(*pid & 0x7F) - 1] = len-11;
    }

    if((*pid & 0x80) == 0x80) {
        if((*pid & 0x7f) > mPayload[iv->id].maxPackId) {
            mPayload[iv->id].maxPackId = (*pid & 0x7f);
            if(*pid > 0x81)
                mLastPacketId = *pid;
        }
    }
}
```

On my setup I have the issue that the mqtt reconnect very often doesn't work, but I can open the web page from ahoy for the restart. I don't know if this changes can fix that, but I will try that and give feedback after a few days of testing.

KG3RK3N, 2022-07-29:
Seems that the changes makes no difference on my setup. Reconnection issue still exists.

stefan123t, 2022-07-29:
@KG3RK3N your change should be alright, would you commit it ?
But the FrameID being 0x00 is a very rare symptom, probably it has nothing to do with the reconnect issues.

stefan123t, 2022-07-29:
Thanks for your commit, I may close the issue for now. Please create a PR.

closed 2022-07-29 by stefan123t