lumapu/ahoy #114: ntp-Server konfiguration in the web frontend ?

Nemo135, 2022-07-30:
Hello,
it is possible to make the ntp-Server Adress or IP konfig in the web frontend ?
I have seen, that the DTU without internet access has a problem, .. no ntp-Server..
I have a lokal ntp, so for me would it nice to config the ntp .

Greetings , Marco

ZsZs73, 2022-08-02:
You can redirect NTP requests on your firewall to the local NTP server.
Or set up a static DNS entry (it is called override in my Undound config) in your DNS for pool.ntp.org pointing to your internal NTP server.
The later works for me flawlessly.

stefan123t, 2022-08-03:
@Nemo135 you would need to extend the UI, EEPROM and set a variable mNtpServer accordingly.
Please note that changing the EEPROM structure / layout (at least changing the SSID/PW part) can break the setup for existing users and may require a new setup saved from UI.

lumapu, 2022-08-05:
I see the point and agree that we should support that setting in near future.

@stefan123t wir könnten doch vor dem Update auf die Serielle Ausgabe wie in #117 verweisen bzw. ein downloadbares JSON anbieten, das alle Einstellungen enthält, dann kann man sie nach dem Update wieder einpflegen.
Für die Zukunft könnte man dann auch wieder ein file einlesbar machen - aber das sehe ich eher als "nice to have"

stefan123t, 2022-08-06:
1. Eingabefeld für NTP Habe ich irgendwo schon mal eingebaut
2. Debug/Info Ausgabe ist sinnvoll

Den dritten Punkt evtl separat:
3. JSON config endpoint ist sinnvoll, kann man den dann auch für die setup Seite verwenden damit wir nicht soviel. String copy&replace haben ?

stefan123t, 2022-08-06:
Das Problem ist noch, dass die Setup Seite bisher keine sicheren Default Werte vorgibt.
NTP Server
  NTP Server / IP: pool.ntp.org
  NTP Port: 8888
@lumapu wie können wir das Problem allgemein lösen, ich glaube auch für MQTT Server, Topic oder Interval gibt es aktuell keine sinnvollen Vorgaben. Auch die WiFi SSID & Password übernimmt er m.E. bisher noch nicht aus den Factory Defaults bzw. dem config.h ?

lumapu, 2022-08-07:
ja hier müssen wir noch bisschen recherchieren / probieren und es dann auf die genannten Stellen ausrollen

stefan123t, 2022-08-12:
@lumapu bitte als gelöst schließen. Danke
Laß uns lieber ein neues Issue für die ganzen Setup Werte bzw. deren Reset to Defaults aufmachen.
Momentan wird ja immer ein Erase Settings ausgeführt und alles mit 0xFF oder 0x00 überschrieben.
Es sollten aber einmal die Default Werte geschrieben werden. 
Ich weiß nicht ob man nicht einfach die aktuellen Werte im Memory überschreiben kann und dann die Config normal speichern ?

lumapu, 2022-08-12:
ja ich habe diesen Punkt im Kopf und wir sollten ihn sehr bald angehen.

commit 2a749f63 (2022-08-06): Merge pull request #126 from stefan123t/main

fix #114 ntp server and port in setup
closed 2022-08-12 by lumapu