lumapu/ahoy #117: Erweiterungsvorschlag Serielle Ausgabe

HFarryranz, 2022-08-02:
Ich hätte folgenden Erweiterungsvorschlag  in app.cpp (ab Zeile 201), um mehr Info an der seriellen Konsole zu bekommen:
(kann ja evtl. noch m weitere Variablen erweitert werden)

        DPRINTLN(DBG_INFO, F("to configure your device"));
        DPRINTLN(DBG_INFO, F("----------------------------------------\n"));

        DPRINTLN(DBG_INFO, F("Your config:\n"));

        DPRINT(DBG_INFO, F("mqttAddr: "));
        DPRINTLN(DBG_INFO,  F(">") + String(mqttAddr)  +F("<") );

        DPRINT(DBG_INFO, F("mqttPort: "));
        DPRINTLN(DBG_INFO,  F(">") + String(mqttPort)  +F("<") );

        DPRINT(DBG_INFO, F("mqttUser: "));
        DPRINTLN(DBG_INFO,  F(">") + String(mqttUser)  +F("<") );

        DPRINT(DBG_INFO, F("mqttPwd: "));
        DPRINTLN(DBG_INFO,  F(">") + String(mqttPwd)  +F("<") );

        DPRINT(DBG_INFO, F("mqttTopic: "));
        DPRINTLN(DBG_INFO,  F(">") + String(mqttTopic)  +F("<") );

        DPRINT(DBG_INFO, F("mDeviceName: "));
        DPRINTLN(DBG_INFO,  F(">") + String(mDeviceName)  +F("<") );

HFarryranz, 2022-08-05:
Das Ergebnis in der Console wäre etwa :

I: to configure your device
I: ----------------------------------------
I: Your config:

I: mVersion:   >0.4.25<
I: IP:         >192.168.100.224<
I: Name:       >name2<
I: mqttAddr:   >192.168.49.5<
I: mqttPort:   >1881<
I: mqttUser:   >UserMQTT<
I: mqttPwd:    >PWMQTT<
I: mqttTopic:  >INTERVERTER_TOPIC<
I: mMqttInter.:>105<
I: mDeviceName:>AHOY_DTU<
I: mSendInter.:>15<
I: name:     >name0<
I: invSerial:>123456780000<
I: name:     >name1<
I: invSerial:>ad0000e10000<
I: name:     >name2<
I: invSerial:>ad0000e20000<
I:
I: ----------------------------------------

Der Code dazu 👍 

 `         DPRINTLN(DBG_INFO, F("----------------------------------------"));
        DPRINTLN(DBG_INFO, F("Your config:\n"));

        DPRINTLN(DBG_INFO, F("mVersion:   >") + String(mVersion) +F("<") );

        DPRINTLN(DBG_INFO, F("IP:         >") + WiFi.localIP().toString()+ F("<") );
        DPRINTLN(DBG_INFO, F("Name:       >") + String(name) +F("<") );

        DPRINTLN(DBG_INFO, F("mqttAddr:   >") + String(mqttAddr) +F("<") );
        DPRINTLN(DBG_INFO, F("mqttPort:   >") + String(mqttPort) +F("<") );
        DPRINTLN(DBG_INFO, F("mqttUser:   >") + String(mqttUser) +F("<") );
        DPRINTLN(DBG_INFO, F("mqttPwd:    >") + String(mqttPwd)  +F("<") );
        DPRINTLN(DBG_INFO, F("mqttTopic:  >") + String(mqttTopic)+F("<") );
        DPRINTLN(DBG_INFO, F("mMqttInter.:>") + String(mMqttInterval)+F("<") );
        DPRINTLN(DBG_INFO, F("mDeviceName:>") + String(mDeviceName) +F("<") );
        
        DPRINTLN(DBG_INFO, F("mSendInter.:>") + String(mSendInterval) +F("<") );

        for(i = 0; i < MAX_NUM_INVERTERS; i ++) 
        {
            mEep->read(ADDR_INV_ADDR + (i * 8),               &invSerial);
            mEep->read(ADDR_INV_NAME + (i * MAX_NAME_LENGTH), name, MAX_NAME_LENGTH);
            mEep->read(ADDR_INV_CH_PWR + (i * 2 * 4),         modPwr, 4);
            
            DPRINTLN(DBG_INFO, F("name:     >") + String(name) +F("<") );
            DPRINTLN(DBG_INFO, F("invSerial:>") + String(invSerial, HEX)+F("<") );
           //DPRINTLN(DBG_INFO, F("modPwr:   >") + String(modPwr) +F("<") );
        }

        DPRINTLN(DBG_INFO, F(""));
        DPRINTLN(DBG_INFO, F("----------------------------------------\n"));`

lumapu, 2022-08-05:
das ganze sieht für mich sehr pflegeintensiv aus. Ich würde vorschlagen, das eher mit einem char Array ähnlich wie https://github.com/grindylow/ahoy/blob/375a1ca9553069b80a6928bb161b02041290e3e6/tools/esp8266/hmDefines.h#L32 zu lösen und die Ausgabe per loop zu machen, evtl. sind es dann nur wenige Zeilen. Für Erweiterungen muss dann nur das Array entsprechend geändert werden

DanielR92, 2022-08-17:
Ist der Punkt hier noch aktuell? 
Wurde geschlossen und wieder geöffnet... Stefan? :)

stefan123t, 2022-08-17:
Ist mW noch nicht umgesetzt und schon gar nicht wie es Lukas vorgeschlagen hatte.

closed 2022-09-04 by aschiffler