lumapu/ahoy #118: Keine Werte mehr vom Wechselrichter

DennisXK, 2022-08-03:


stefan123t, 2022-08-03:
Bitte Deine Aussage mit aussagekräftigen Serial Debug Logs belegen.
Vermutlich eines der/beide bekannten Problem in #106 MQTT Reconnect und/oder #112 RF Channel für TX/RX.

stefan123t, 2022-08-03:
vermutlich durch #112 bzw. PR #119 behoben.

stefan123t, 2022-08-04:
@DennisXK bitte probiere mal die neue Version 0.4.26 damit sollte das Problem behoben sein.
Bitte schließe das Issue wenn es für Dich damit funktioniert.

DennisXK, 2022-08-04:
ok, jetzt gibt es eine neue Version. Ich habe gestern Abend von der 0.4.22 auf die 0.4.25 upgedatet. Konnte aber dann nicht sehen ob das Problem behoben ist (war ja dunkel).  Versuch es nachher...

billy0xff, 2022-08-04:
Version 0.4.26, Kanalabfolge mit dem SA beobachtet. Da mit Max-Hold gescannt verliert man viele der Sequenzen, die einzige welche zweifelsfrei eine unmittelbare Folge ist ist mit  -> markiert:
[image]

Receive success natürlich wieder 0.

DennisXK, 2022-08-04:
OK geht wieder. auch mit der 0.4.22. Habe jetzt die 0.4.26 drauf, läüft auch. Mit "Max-Hold" meinst du die NRF24 sende/empfangsstärke? Den hab ich auch Max stehen, läuft aber auch auf Min.

Wo kann ich den SA beobachten? muss ich da mit der Arduinio IDE über den USB Port des Wemos dran und die Seriele Schnittstelle öffnen? Die Tabelle versteh ich nicht so richtig. TX 1 2 und RX 3 4 wäre perfekt und kommt oft vor?

stefan123t, 2022-08-05:
@billy0xff ich habe das mit den zwei Channel Abstand auch schon gesehen, aber eine wirkliche Sequenz konnte ich noch nicht ausmachen. Ich habe jetzt zwei Indizes mTxIdx und mRxIdx die immer weiterspringen. Mein RTL2832U kann den Frequenzbereich um die 2.4 GHz leider nicht und ein LNB um den Frequenzbereich herunter zu konvertieren ist nicht mehr verfügbar und kostet dementsprechend ein Vermögen.
Im Forum hatten Einige (martin Gast, Oliver F und B.G. golf2010) bereits auf den ersten Seiten einige interessante HackRF Scans eingestellt. Kannst Du mit Deinem Spectrum Analyzer beobachten, was passiert, wenn wir die Retries (aktuell 15, also 16 TX Pakete) anpassen und weniger anfragen, bekommen wir dann auch weniger RX Pakete / Antworten zurück. Ich habe nämlich das Gefühl dass wir da den WR eventuell etwas überlasten.

@DennisXK der Spectrum Analyzer ist ein Gerät (a bissle wie ein Oszilloskop für Frequenzen / Bänder) @billy0xff hat in #112 einige Screenshots von seinem SA eingestellt.

billy0xff, 2022-08-06:
@stefan123t
In diesem Fall ist ein SDR wie der HackRF einem Sweep-Type SA doch weit überlegen: Der 80MHz sweep beim SA dauert ca. 20ms und ist viel zu langsam für die gepulsten Signale. Hier muß man auf Max-Hold gehen, sieht dann zwar die Veränderungen, aber einerseits kann die Reihenfolgeinformation verlorengehen, und man muss dauernd neu initialisieren um nicht schwache Antworten unter den schon dargestellten Aussendungen zu "verlieren". 
Immerhin sieht man die belegten Kanäle und die Pegel (wenn man rechtzeitig die höheren Pegel wieder per clear write gelöscht hat.) und ich sehe dass der Inverter antwortet. Die Retries sieht man nicht,
Ich denke dass eine erweiterte Debuginformation viel zielführender wäre, man müsste herausfinden können wo denn die RX Daten geblieben sind, gibt es im NRF keinerlei weiterführenden Informationen aus den diversen Registern des Funkchips?

stefan123t, 2022-08-06:
@billy0xff GetLossRate | 0x15 als Kommando liefert uns die RX/TX Paketzähler auf Seite des Wechselrichters, 
das hilft uns bestimmt weiter dabei, das Channel Hopping besser zu verstehen! 
https://github.com/stefan123t/ahoy/wiki/FAQ-Frequently-Asked-Questions#getlossrate--0x15https://github.com/stefan123t/ahoy/wiki/FAQ-Frequently-Asked-Questions#getlossrate--0x15

billy0xff, 2022-08-07:
@stefan123t , hier wären wir aber schon weit oben in den Kommunikationsschichten, bei mir kommen ja gar keine Daten an, hier wäre interessant ob der NRF überhaupt etwas empfangen hat oder nicht. Das ist halt das Problem mit den ganzen fertigen Libraries wo man dann keinen Durchblick mehr hat; ich wüsste jetzt nicht, wo man da noch Debugcode einbauen könnte. Aktueller Status: NRF sendet, Inverter antwortet, NRF RX-Status im Prinzip unbekannt. (gibts hier keine Möglichkeit für PMs damit man nicht die Issues dafür verwenden muss?)

oischinger, 2022-08-07:
Ich habe ein ähnliches Problem, allerdings funktioniert bei mir noch die Senderichtung: ich kann den Wechselrichter an/ausschalten (gemessen über die Steckdose)
Ist das bei dir auch so @billy0xff ?

billy0xff, 2022-08-07:
@oischinger: soweit komme ich gar nicht, ahoy findet keinen Inverter; wo finde ich denn diese Funktionalität zum Ein/Ausschalten? Ich werde mir solche Schukozwischensteckdosen mit Powermonitoring besorgen müssen, die haben zwar grauslich schlechte Daten, aber besser als gar nichts; mit dem Ahoy komme ich nicht weiter...

stefan123t, 2022-08-07:
@billy0xff tut mir leid dass es nicht klappt bei Dir. Wie gesagt hilfreich ist das Serial Debug Log. Und wenn Du es nicht über ein Issue hochladen kannst/willst dann probiere doch mal den Discord Chat. Aktueller Link ist mW im Forum.

stefan123t, 2022-08-12:
@DennisXK bitte mit der v.0.5.9 überprüfen und ggf. als gelöst schließen. Danke

closed 2022-08-17 by aschiffler