lumapu/ahoy #125: Power Level Version 0.5.1

costa2, 2022-08-06:
Ich habe gerade auf die 0.5.1 upgedatet.
Bei Aktive Power Level hatte ich nichts eingetragen, also stand es bei 0.
Nun produziert mein HM600 nichts mehr, auch nachdem ich 600 eigetragen habe.
Wie behebe ich diesen Fehler nun?

aschiffler, 2022-08-06:
Danke für das Feedback. Das Verhalten mit dem initialen Fall wird/wurde geändert.
Wenn man mehr einträgt nimmt der Umrichter das nicht an. Das Limit sieht man in der Visulazation Seite.

lumapu, 2022-08-06:
am Code würde doch gar nichts mehr geändert oder?

aschiffler, 2022-08-06:
> am Code würde doch gar nichts mehr geändert oder?

Der Ansatz ist das Feedback, Issues aus den Rückmeldungen zu sammeln und Änderungen, Verbesserungen daraus abzuleiten. Dann wieder zusammen mit anderen Verbesserungen ein neues Release zu machen. 0.5.x+1
Ich habe bereits anfangen die Rückmeldung hier umzusetzen. Das war gemeint.
Siehe hier: [https://github.com/aschiffler/ahoy/tree/pwrlimit](https://github.com/aschiffler/ahoy/tree/pwrlimit)

lumapu, 2022-08-06:
danke für die Klarstellung, ich dachte ich habe was übersehen.

hubsi5, 2022-08-07:
Danke für die Top Version!
Ich habe es mit meinem HM1500 ausprobiert.
Leider bewölkt.
Hatte zb ohne Leistungsbegrenzung 250W Ertrag.
Dann stellte ich die Leistungsbegrenzung zum probieren auf 200W.
Danach hatte ich einen Ertrag von 100W, obwohl gleich bewölkt.
Stellte ich danach die Begrenzung wieder auf 1500W, hatte ich wieder 250W Ertrag.

Ist die Leistungsbegrenzung prozentuell oder sollte das ein Höchstwert sein?

Weiters komme ich mit den MQTT Datenpunkten im IObroker nicht klar.
habe mir einen datenpunkt ahoy/devcontrol/0/11 angelegt und dort versucht die Leistungsbegrenzung zu steuern.
Stimmt der Datenpunkt? Wenn ja was gehört da genau hinein? payload  "300" funktioniert nicht. Nur 300 auch nicht.

lumapu, 2022-08-07:
habe gerade auch Probleme, Power Limit in Setup last sich nicht speichern, mein Inverter läuft auf 0W

hier steht das MQTT topic:
 `// subcribed topics are mTopic + "/devcontrol/#" where # is <inverter_id>/<subcmd in dec>`

costa2, 2022-08-07:
> habe gerade auch Probleme, Power Limit in Setup lässt sich nicht speichern, mein Inverter läuft auf 0W
> 
> hier steht das MQTT topic: `// subcribed topics are mTopic + "/devcontrol/#" where # is <inverter_id>/<subcmd in dec>`

Trag einen Wert ein und starte den Wechselrichter neu, nur so ging es bei mir auch.

thinklerde, 2022-08-07:
Wie startet man den WR vernünftig neu? habe ihn jetzt für 20min von den Modulen und vom Netz getrennt. Produziert seit dem aber weiterhin 0W.

fila612, 2022-08-07:
Ich hatte heute früh ein ähnliches Phänomen.
Hab die 0.5.2 auf einem zusätzlichen Wemos installiert.
Limit hatte ich leer gelassen, nach dem Neustart hatte ich dann ein Limit von 12689 (oder ähnlich). 
die Testinstanz 0.5.2 hatte ich dann wieder ausgeschalten.
Problem war aber, der WR produzierte nichts mehr, obwohl davor schon 4Wh an dem Tag registriert wurden.

Neustart vom WR und vom Wemos half nichts.
Was etwas gebracht hatte, war eine Änderung des Amplifier Levels und anschließendem Neustart.
Evtl hilft es jemanden.

hubsi5, 2022-08-07:
Zitat von User Gerald R. im Mikroforum

costa2, 2022-08-07:
https://www.mikrocontroller.net/topic/525778?page=11#7151619

mslookup, 2022-08-07:
Wäre super, wenn das mal jemand gegenchecken könnte und nen commit macht, ich bin leider unterwegs und meine WRs zuhause machen gerade auch 0 Watt 😅

thinklerde, 2022-08-07:
> Wäre super, wenn das mal jemand gegenchecken könnte und nen commit macht, ich bin leider unterwegs und meine WRs zuhause machen gerade auch 0 Watt 😅 alternativ werd ich gegen 15:00 wieder zuhause sein, dann kann ich berichten/committen.

Funktioniert so wie hier beschrieben: https://www.mikrocontroller.net/topic/525778?page=11#7151619

lumapu, 2022-08-07:
so habe gemerged, war leider auch unterwegs.
Vorher habe ich durch mehrere Neustarts und auch AC Trennung von WR alles wieder ans Laufen gebracht

stefan123t, 2022-08-09:
@aschiffler & @lumapu m.E. gibt es hier zwei Probleme für neue Nutzer:
* AhoyDTU setzt ein Active PowerLimit von 0 was man eigentlich durch das Shutdown Command erreicht.
https://github.com/grindylow/ahoy/blob/155f9e3f41025fe7d0542a95014cc5f6762e8299/tools/esp8266/app.cpp#L392

Hier sollte m.E. eine if Condition außenrum und nur wenn iv->powerLimit>0 ist sollte hier auch das PowerLimit gesetzt werden.

* AhoyDTU setzt das Active PowerLimit immer persistent, obwohl man das eventuell gar nicht in das EEPROM des Wechselrichters schreiben will:
https://github.com/grindylow/ahoy/blob/f3e928244e3eb1a265acbe4158b2f91f7993ed5c/tools/esp8266/hmRadio.h#L180-L182

Hier sollte m.E. unser Default remanent / nicht-persistent also 0x00 00 sein.

Nur wenn man einen entsprechendes Check Mark in der Setup UI setzt (z.B. PowerLimit fest im WR EEPROM speichern), bzw. das entsprechende Flag beim MQTT Topic mit angibt, sollte hier das Active PowerLimit permanent gespeichert werden.
M.W. geht das mit PowerPFDev.Desc = 0x00 01 (nicht 0x01 00) und wird in der DTU-Pro nur für DRM_Limit_Switch oder SunSpec_Switch gesetzt, also wenn die DTU als "Remote Control" arbeitet und die Werte permanent setzen soll.
Im Normalfall arbeitet die DTU als "Master" im Zero_Export_Switch oder im Phase_Balance_Switch Modus und steuert die Wechselrichter selbst, d.h. sie stellt das Limit je nach Bedarf immer wieder neu nach.

aschiffler, 2022-08-10:
@stefan123t Also das mit PowerPFDev.Desc = 0x00 01 (nicht 0x01 00) ist immer noch ein Schwachpunkt. Ich habe mit meinem Gerät beides getestet und beides funktioniert. Es scheint als würde der WR einfach ">0" testen für die beiden Bytes.
--> Können wir anpassen wie wir wollen im Moment.

Zu dem CheckMark oder "Hacken" im Setup. Das habe ich erstmal nicht umgesetzt. Es wird nun mit dem [neuen pull request](https://github.com/grindylow/ahoy/pull/134) konsequent wie unter den Nutzern schon bekannt: MQTT -> nicht persistent und Setup -> Persistent das Power-Limit gesetzt.

lumapu, 2022-08-10:
@costa2 bitte die neueste Version 0.5.5 Git-SHA g6276957 verwenden

closed 2022-08-10 by lumapu