lumapu/ahoy #128: Active Power Limit (0)

XTeslaX, 2022-08-07:
Ich habe einen HM-600 im Einsatz und ein Update auf 0.5.2 gemacht. Jetzt produziert der HM-600 nicht mehr. Bei Active Power Level ist 0 eingetragen. Änderungen auf 600 werden nicht gespeichert. Ein Neustart hat nichts geholfen.

mslookup, 2022-08-07:
Schau mal in den Issue https://github.com/grindylow/ahoy/issues/125, Lösung ist schon im Pull request https://github.com/grindylow/ahoy/pull/127 enthalten.

stefan123t, 2022-08-07:
Vielleicht sollten wir noch eine Checkbox dazu machen oder bei PowerLimit 0 gar nichts senden ? Die Original Hoymiles unterstützt mW auch nur 2-100% der max Leistung.

lumapu, 2022-08-10:
@XTeslaX bitte die neueste Version 0.5.5 Git-SHA g6276957 verwenden

closed 2022-08-10 by lumapu