lumapu/ahoy #129: Verbesserungsvorschlag: Art des Setzens des PowerLimits

mslookup, 2022-08-07:
Einen Vorschlag zur Verbesserung der PowerLimit Funktion:
- Standardmäßig steht die PowerLimit Einstellung in der GUI auf -1 => Unbegrenzt
- 0 bedeutet 0
- x bedeutet Begrenzung auf x Watt

Ich bin heute früh, wie viele andere auch, erstmal schön mit https://github.com/grindylow/ahoy/issues/125 auf die Schnauze geflogen :D Nach einem Update von Ahoy steht der Wert ja standardmäßig auf 0, was 0 Watt bedeutet. Eventuell ist es möglich das ganze so anzupassen, dass es wie im Vorschlag beschrieben funktioniert. Bisher war Ahoy nur auslesen für mich "Normalo Ahoy User", aber wenn's schön langsam auch in die Richtung Settings des WR anpassen geht, hätte ich gerne volle Kontrolle ;)
Danke euch, ihr seid spitze!🥇

lumapu, 2022-08-07:
man kann doch auch einfach die 0 erkennen und dann nicht begrenzen, wenn man den WR ausschalten will gibt es doch noch andere CMDs . Dann wäre der Default nach dem Reset auch gleich mit behoben.

mslookup, 2022-08-07:
Auch möglich :)

stefan123t, 2022-08-07:
:+1: 
* "0" dürfte eigentlich kein valides PowerLimit sein. Laut Hoymiles ist ActivePowerLimit auch nur zwischen 2-100% definiert. 
  D.h. "0" dürfte keinen Wert von 0 als ActivePowerLimit setzen und sollte m.E. somit ignoriert werden.
  In der DTU-Pro wird ab 30W pro MPPT geregelt und dann m.W. auch nur in 30W Schritten. (Vielleicht ist das aber auch die AntiReflux Regelung für die MI-Wechselrichter, dort ist die Einschaltschwelle nochmal höher als bei den aktuellen HM-Invertern).
  + Zusätzlich sollte sowohl im GUI als auch per MQTT/JSON Kommando ein Flag mitgegeben werden wenn es permanent gespeichert werden soll. Der Default sollte remanent / temporär sein.
  + Alternativ könnte er bei "0" per MQTT ggf. auch das Shutdown Kommando senden.
* "-1" sehe ich wie @mslookup auch als unlimited an, vielleicht sollte dann das ActivePowerLimit auf das maximal Zulässige des WR gesetzt werden ?

lumapu, 2022-08-10:
@mslookup bitte die neueste Version 0.5.5 Git-SHA g6276957 verwenden

closed 2022-08-10 by lumapu