lumapu/ahoy #130: Bug: Setzen des PowerLimits auf -1 (unbegrenzt) funktioniert nicht

mslookup, 2022-08-07:
Hallo zusammen,
das manuelle Setzen des PowerLimits meiner 2 HM1500 auf 1500W funktioniert soweit, zumindest produzieren beide Einheiten wieder Strom. Leider funktioniert das Setzen des PowerLimits auf -1 (unbegrenzt) noch nicht fehlerfrei bei mir (ggf. Bug bei allen?).

Während der 1. Wechselrichter auf "Standgas" läuft, macht der 2. WR gar nichts mehr:
[image]

Ändere ich den Wert zurück auf 1500 (das Max. meines Wechselrichters), gehen beide wieder los:
[image]

VG mslookup

lumapu, 2022-08-07:
sieht so aus als gäbe es ein maximales Limit, evtl. muss man das auf z.B. 3000W nach oben begrenzen. -1 wird bei einem unsigned int16 zum Maximalwert 65535

Solateur, 2022-08-07:
Mein HM600 läßt sich ohne Probleme auf 800 W einstellen - mehr habe ich nicht getestet, weil sinnlos.

Noch eine Anmerkung: Ich betreibe den HM600 z . Z mit einem Modul. Dabei habe ich- mit einem 340 W peak Modul  - 320 W Leistung erreicht. Wenn ich unter 0.5.2 400 W Limit einstelle, ist das Maximum pro Modul 200 W.

mslookup, 2022-08-07:
@Solateur: Es geht mir mit der -1 weniger darum, die Drosselung auf den maximal möglichen Wert zu stellen, sondern die PowerLimit Funktion komplett zu deaktivieren. Meines Wissens nach macht das in der Config des Hoymiles einen Unterschied…

Solateur, 2022-08-07:
@mslookup: Meine Anmerkung bezog sich auf die Vermutung von lumapu: "sieht so aus als gäbe es ein maximales Limit, evtl. muss man das auf z.B. 3000W nach oben begrenzen."

mslookup, 2022-08-07:
@Solateur :D achso - ja jetzt wo du's sagst

lumapu, 2022-08-10:
@mslookup bitte die neueste Version 0.5.5 Git-SHA g6276957 verwenden

closed 2022-08-10 by lumapu