lumapu/ahoy #132: Wenn MQTT aktiviert kein Webserver mehr erreichbar

hubsi5, 2022-08-08:
Ich habe ein merkwürdiges Problem.
Es funtkioniert alles am ahoy stick.
Sobald ich aber eine MQTT Verbindung aufbaue, heißt ich den MQTT Server eintrage, wird ahoy noch neu gestartet.
Es steht dann auch MQTT Verbindung ok.
Dann erreiche ich aber die Web Oberfläche von ahoy nicht mehr und er startet auch laufend neu.
Lösche ich dann den MQTT Server in den ahoy Einstellungen wieder, funktioniert wieder alles einwandfrei.
Woran kann das liegen?
Der MQTT Server ist der gleiche wie immer, und bis vor kurzem hat das Ganze noch funktioniert.
Andere Geräte haben mit meinem MQTT Server (iobroker) keine Probleme

TobiDD79, 2022-08-08:
Welche Version hast du aktuell auf deinem esp laufen?

hubsi5, 2022-08-08:
Die neueste.
Ich versuche die Limitierung über MQTT zu realisieren.
Leider läuft bei mir ahoy generell etwas unstabil :-(

lumapu, 2022-08-08:
habe auch sowas ähnliches heute beobachtet, weiß aber noch nicht woran es liegt. Habe das Intervall auf 15s stehen, Mqtt Intervall ist demzufolge 75s. Nach 85s Uptime stürzte der ESP ständig ab und bootete neu, habe leider kein serielles Log.
Die Uptime hat er komischer Weise per MQTT übertragen. Nach ein paar Stunden hat es wieder funktioniert ohne was zu ändern.

Kann eine subscription zum Neustart führen, wenn der MQTT Broker das Topic nicht hat / kennt?

stefan123t, 2022-08-09:
@lumapu & @hubsi5 ihr solltet mal das Debugging in der platformio.ini aktivieren.
```c
build_flags = -DDEBUG_ESP_PORT=Serial
	-DDEBUG_ESP_CORE 
	-DDEBUG_ESP_WIFI 
	-DDEBUG_ESP_HTTP_CLIENT 
	-DDEBUG_ESP_HTTP_SERVER 
	-DDEBUG_ESP_OOM 
```
Eventuell will man auch HTTP_CLIENT UND _SERVER wider abschalten, da hier sehr viele Zeilen geloggt werden.

Ich speichere es bei mir mit `log2file` in eine Datei und lese mir diese nachträglich durch, da der Monitor im VSCode/PlatformIO nur einen eingeschränkten Scroll Buffer hat.
```
monitor_filters = 
	;default   ; Remove typical terminal control codes from input
	time       ; Add timestamp with milliseconds for each new line
	log2file  ; Log data to a file “platformio-device-monitor-*.log” located in the current working directory
```

stefan123t, 2022-08-12:
@hubsi5 bitte mit der v.0.5.9 überprüfen und ggf. als gelöst schließen. Danke

closed 2022-08-17 by hubsi5