lumapu/ahoy #135: Momentane Leistungsbegrenzung per MQTT übertragen

hubsi5, 2022-08-10:
Habe die neueste 0.5.5 Verion auf  meinem Wemos übertragen.
Muss sagen, ist die erste Version die bei mir bis jetzt stabil läuft.
Danke!

Ich steuere per MQTT die Leistungsbegrenzung.
Hätte da eine Frage. 
Wäre es möglich, per MQTT die Momentane Begrenzung mit zu übertragen?
Also einfach ein Topic mit der Zahl der Begrenzung.

aschiffler, 2022-08-10:
Einfach auf das Topic selbst noch mal ein subscribe machen. So habe ich das auch in OpenHAB. 
Dort (OpenHAB) sieht das so aus:
`commandTopic: home/devcontrol/0/11`
`stateTopic: home/devcontrol/0/11`

Also sowohl Lesen als auch Schreiben.

Das mit 800 --> 8000 kommt nun nicht mehr. War ein Bug.

hubsi5, 2022-08-10:
Ok danke.
Das mit den 8000 hatte ich gerade mit der 0.5.5 Version.

aschiffler, 2022-08-10:
> Ok danke. Das mit den 8000 hatte ich gerade mit der 0.5.5 Version.
Das ist nicht gut. Falls es nochmals / öfters auftritt versuche mal heraus zu finden ob das dann der Fall ist wenn du "häufig hintereinander" das commando sendest.

hubsi5, 2022-08-11:
Leider kommt bei mir noch immer eine falsche ;QTT Übertragung zustande.
Wollte auf 600W begrenzen, angekommen sind 6000W
Version 0.5.5

aschiffler, 2022-08-11:
Ich denke das liegt an der MQTT Implementierung. Wir sind hier dran. Grund ist hier wahrscheinlich das mehrfache Senden der Nachricht an die DTU und das fehlende Handling für diesen Fall.

HorstG-57, 2022-08-11:
Was ich gesehen habe ist, das beim Senden der Leistungsbegrenzung an den WR der Wert für die Leistung direkt vorm Senden mal 10 gerechnet wird und danach wieder durch 10 geteilt wird. 
Da die Rechnung auf der Variablen stattfindet die über einen Pointer erreicht wird, könnte da auch ein Problem vorliegen.

Was muss denn an der WR übertragen werden, die Leistung in W als Ganze Zahl oder muss die Leistung in 1/10 W als Ganzzahl übergegen werden. 

Oder ist das Ganze einfach nur ein Anzeige-Fehler innerhalb der Webseite, da der Original Leitungswert beim Senden an den WR kurz verändert wird ?

Ich habe ausserdem ein PR #138 erstellt, wegen des Subscription für die Steuerung über MQTT. 
Ev. spielt das da auch noch mit rein.

lumapu, 2022-08-11:
ist das jetzt mit Version 0.5.6 behoben?

hubsi5, 2022-08-11:
bis jetzt sieht es bei mir gut aus !

stefan123t, 2022-08-11:
@HorstG-57 & an den WR müssen wir den 10-fachen Wert in Watt senden, da der WR für das ActivePower Limit einen "fixed point integer" erwartet mit genau einer "Nachkommastelle".

@aschiffler hier noch die entsprechenden Kommentare aus dem DTU-Pro Source Code `AntiReflux.c`
```c
1101:    //Use power> When each unit is 30w,= Turn on the power limit of each unit and turn on the full amount when the full amount is exceeded
1212:    //When it is not enough for 30w per unit, because the power limit cannot be maintained, some units need to be turned off to make the remaining power limit
```

* `30.0 W` = `0d00000300` = `0x012C` das ist m.W. ein sinnvolles Minimum, darunter arbeitet der WR nicht richtig
* bzw. `600.0 W` = `0d00006000` = `0x1770`
* oder `1500.0 W` = `0d00015000` = `0x3A98` das sollte m.E. das Maximum sein oder haben wir auch einen HM-1800 ?

stefan123t, 2022-08-12:
@hubsi5 bitte mit der v.0.5.9 überprüfen und ggf. als gelöst schließen. Danke

aschiffler, 2022-08-15:
solved in new versions.
given information was taken into account

closed 2022-08-15 by aschiffler

MubiTec, 2022-08-18:
Ich hab im iobroker den state subscribed, aber ich bekomme keine Rückmeldung, ob mein Limit eingetragen ist oder nicht. Deshalb die Frage, gibt es wirklich eine Bestätigung, ob das Limit gesetzt wurde?

stefan123t, 2022-08-18:
Wir bekommen aktuell mW nur eine Kommando ActivePower Limit setzen erfolgreich Meldung. Wir können aber die Art und den Wert des Power Limits nicht abfragen. @aschiffler was bringt die MQTT Topic subscription an der Stelle ? Es ist ja nur eine synchrone Rückmeldung macht das als asynchroner „Zustand“ an der Stelle überhaupt Sinn ?

aschiffler, 2022-08-19:
@stefan123t Nein das ist in sofern nicht sinnvoll da keine andere/neue Information gegeben wird.