lumapu/ahoy #137: Power Limiting: Per MQTT gesendeter Wert wird falsch übernommen (v 0.5.3)

FSys77, 2022-08-10:
500 W per MQTT geschickt, es wird versucht 5000 W zu setzen.

[image]

roku133, 2022-08-10:
Das auf der Weboberfläche angezeigte Limit stammt aus der Setup-Konfiguration und wird permanent im Wechselrichter gespeichert.
Die Einstellung per MQTT bleibt dagegen nur bis zum Neustart des Wechselrichters bestehen, zur Kontrolle kann diese Einstellung auch wieder per MQTT ausgelesen werden.
Es gibt also zwei Einstelloptionen zur Leistungsbegrenzung.

HorstG-57, 2022-08-10:
Es sollte im Setup die Option geben, das man keine bzw. keine Permanente Leitungsbegrenzung senden möchte,
bzw. man sollte einstellen können das die Steuerung des WR nicht zugelassen ist.

Beispiel :

   Permanennte Leistungsbegrenzung                   : ( keine / Leistung in W )
   Leistungsbegrenzung per MQTT zugelassen     : ( ja / nein )

stefan123t, 2022-08-10:
@HorstG-57 
ist schon teilweise umgesetzt, siehe hier:
https://github.com/grindylow/ahoy/blob/6276957b22f4268e1daf38e8c4d5271d1dc5f819/tools/esp8266/app.cpp#L311-L318

und hier
https://github.com/grindylow/ahoy/blob/6276957b22f4268e1daf38e8c4d5271d1dc5f819/tools/esp8266/app.cpp#L439-L446

Vor allem hier müsste m.E. eine Checkbox rein um das ActivePower Limit `permanent` zu setzen.
https://github.com/grindylow/ahoy/blob/6276957b22f4268e1daf38e8c4d5271d1dc5f819/tools/esp8266/app.cpp#L608-L615

Und hier wird das ActivePower Limit aus dem MQTT Topic an den WR gesendet. 
https://github.com/grindylow/ahoy/blob/6276957b22f4268e1daf38e8c4d5271d1dc5f819/tools/esp8266/app.cpp#L735-L743

Letztendlich geht es um diese Stelle, hier wird es aktuell permanent gesetzt (>0).
Eigentlich sollte der Wert 0x00 01 sein nicht 0x01 00 aber dieser Abweichung vom DTU-Pro Code (Implementierungseigenart) hält sich hartnäckig :grimacing:
https://github.com/grindylow/ahoy/blob/6276957b22f4268e1daf38e8c4d5271d1dc5f819/tools/esp8266/app.cpp#L87-L93

FSys77, 2022-08-10:
> Das auf der Weboberfläche angezeigte Limit stammt aus der Setup-Konfiguration und wird permanent im Wechselrichter gespeichert. ...

Ja gut ... :-D

Mein Punkt ist aber: ich schicke 500 (fünfhundert) Watt per MQTT zum WR, der meint aber er müsse jetzt auf 5000 (fündtausend) Watt regeln. Ist mir auch mit 30 und 300 passiert. Usw.

Das ist das Issue was ich hier habe.

roku133, 2022-08-10:
Dieses Verhalten kann ich bei mir mit v 0.5.5 nicht beobachten.

Wird bei jedem MQTT-Wert jeweils eine weitere 0 hinzugefügt?

FSys77, 2022-08-10:
> Dieses Verhalten kann ich bei mir mit v 0.5.5 nicht beobachten.
> 
> Wird bei jedem MQTT-Wert jeweils eine weitere 0 hinzugefügt?

Nein, nur manchmal (ich hab aber noch kein Muster entdecken können).

Ich mach jetzt mal nen Update auf 0.5.5

stefan123t, 2022-08-10:
@FSys77 ist in v0.5.5 vorgeblich gefixt. Kann aber auch immer noch an der Übertragung vom MQTT Client zum Ahoy liegen...

FSys77, 2022-08-11:
> @FSys77 ist in v0.5.5 vorgeblich gefixt. Kann aber auch immer noch an der Übertragung vom MQTT Client zum Ahoy liegen...

Ist auch in v0.5.5 noch drin.

Limit via MQTT auf 650 gesetzt, im Web UI steht 6500

[image]

HFarryranz, 2022-08-11:
Dieses Problem mit 300 -> 3000 hatte ich auch schon. Passiert gelegentlich, Als wenn eine 0 hängenbleibt und der neue Wert links davon "dazugeklebt" wird. Eine Prüfung des MQTT-Wert (Leerzeichen, Character enthalten etc) findet meines Wissen ja nicht statt.

FSys77, 2022-08-11:
Mir wäre es sehr recht wenn man das Power Limit via API setzen bis diese MQTT-Geschichte mal stabil läuft.

Viele andere Kommandos gehen ja schon wie API.

Also so: "... tx_request = 81 (=0x51) und cmd = 11 setzen sowie payload dann z.B. auf 400..."
-> https://discord.com/channels/984173303147155506/992022163307638887/1006976536068493383

aschiffler, 2022-08-11:
Hi,

das Problem lag daran, dass WR der HM Serie nur ganzahlige Vielfache von 1/10 Watt annehmen. Deshalb wurde vor dem Senden das Limit des Umrichters mal 10 genommen und nach dem "Senden" wieder geteilt durch 10 gesetzt. Da es sich um einen Pointer auf eine Variable handelte wurde diese mehr oder weniger zufällig in der Webvisualisierung zum "falschen" Zeitpunkt angezeigt. Also es war nur die Anzeige falsch. An den WR wurde stets der richtige Wert gesendet.
Behoben mit:
https://github.com/grindylow/ahoy/pull/138/commits/e5be85cb2705fde41f837c83fa01eaf094bd3024

closed 2022-08-11 by lumapu