lumapu/ahoy #142: Backlog v0.6.0

stefan123t, 2022-08-11:
Hier die Liste der ToDo's für die Version 0.6.0:

* [ ] Settings exportieren / importieren (API/UI)
* [ ] Settings in settings.ini speichern (LittleFS statt EEPROM)
* [ ] Homepage aufräumen nur ein Status (aktuell drei AJAX Calls /uptime, /time, /cmdstat)
* [ ] app.cpp aufräumen und in hmRadio / hmProtokollGen3 auslagern
* [ ] MI Wechselrichter unterstützen (miSystem, miInverter, miDefines, miProtokollGen2 etc.)
* [ ] Device Control Kommandos aus dem Setup ermöglichen (TurnOn, TurnOff, Restart, ActivePower Limit, ReactivePower Limit, SetPowerFactor, etc.)
* [ ] AlarmData & AlarmUpdate Parsen und auf eigener Seite darstellen
* [ ] Device Info Kommandos (Firmware Version, etc.) über das Dashboard anzeigen 
* [ ] Wechsel zu AsyncWebServer und ElegantOTA für Stabilität
* [ ] klarer Scheduler / Task manager, der ggf. den Receive Task priorisieren kann
* [ ] nRF24 Interrupt Handling sinnvoll oder warum macht die nRF24 Bibliothek ständig `0x07` Statusabfragen
* [ ] Debug Level im Setup änderbar -> auch Livedata Visualisierung abschalten ?
* [ ] MQTT Discovery (HomeAssistant) im Setup optional machen
* [ ] MQTT Subscribe nur beim Reconnect

aschiffler, 2022-08-17:
Im development branch ist nun eine einfache Anzeige für den ersten (= leider noch hardcodiert in der index.html) Umrichter. Möglich.
Mittel REST API call kann man aber auch für andere Umrichter InfoAll anfragen und diese sollten dann im MQTT auftauchen.
https://github.com/grindylow/ahoy/commit/458639b6488311efb3c894c5f916c6e4d2df7241

DanielR92, 2022-08-17:
@stefan123t könnte man die Liste eventuell nach Dringlichkeit sortieren? =) 
Vielleicht kann ich hier und da Unterstützen.

lumapu, 2022-09-18:
habe die Liste mal aktualisiert (entspricht dem stand im 'developemnt' Branch)

DanielR92, 2022-10-12:
In der Swim-Lane am 3ten Punkt ist teilweise schon abgehackt oder? (fehlen würde ReactivePower Limit, SetPowerFactor; was ist mit etc. genau gemeint?)

DanielR92, 2022-10-26:
@stefan123t könntest du bitte folgende Punkte erweitern?
Alle Kommentare die noch offen sind in diesem PR zu bearbeiten und in das Styleguide(AUtoformatierung) einfließen zu lassen?
https://github.com/lumapu/ahoy/pull/361

stefan123t, 2022-11-07:
@DanielR92 was soll ich hier machen:
Soll ich die bereits geschlossenen Punkte aus dem ersten Post nehmen 
oder wollen wir ein neues Backlog Issue anlegen ?

DanielR92, 2022-11-18:
Hi Stefan, 
sorry hab erst jetzt gesehen das du darauf geantwortet hast.

Ein neues Issue müssen wir nicht aufmachen dafür. Sondern können dieses Issue als Backlog zweck entfremden. 

Die oberen Punkte die Abgeschlossen sind, kann man gern entfernen und dafür neue Punkte ergänzen.
Beispielsweise auch das Thema mit NRF aus diesem Issue https://github.com/lumapu/ahoy/issues/83#issuecomment-1306231933 das mit diesem Issue vom NRF24 zusammenhängt: https://github.com/nRF24/RF24/issues/877

Danke =)

stefan123t, 2022-11-18:
@DanielR92 Du somltest das mittels Edit mE auch ändern können aber ich habe das mal soweit nachgetragen und abgehakt. Ist glaube gar nicht mehr so viel offen ... oder ?

lumapu, 2022-11-20:
Habe bei Settings Import den Haken wieder entfernt, da dieser Punkt noch offen ist

closed 2023-05-25 by lumapu