lumapu/ahoy #148: v.0.5.9 - Anzeige der Leistugnsbegrenzung im Web

hubsi5, 2022-08-12:
habe nun die neueste Version.
Leistungsbegrenzung per MQTT auf 1000W.
Funktioniert. Wird auch begrenzt.
Nur auf der Webpage steht noch immer 1500W.
Also begrenzt wird die Leistung nur im Web steht noch immer Limit 1500W

aschiffler, 2022-08-12:
Hi,
kann ich nicht nach vollziehen. Ein kurzer Review des Codes und eigener Test zeigt dieses Verhalten nicht.
- Welches Build hast Du? g2df60b8 0.5.9
- Ggf. warten bis die Seite aktualisiert ist.

hubsi5, 2022-08-12:
Habe extra 5 Minuten gewartet und noch extra den Inet Browser refreshed.
Version: g87a5209  0.5.9

aschiffler, 2022-08-12:
Hi,
versuche mal diese binary.
https://github.com/aschiffler/ahoy/actions/runs/2846619040
Das ist die dich ich genutzt habe. Der Unterschied ist nur ein Define in der app.h. siehe:
https://github.com/grindylow/ahoy/compare/main...aschiffler:ahoy:main#diff-ccd999c4e199b49b13abbcf27ede0a2905c87d321b469d5f7ebe4d2686c15fd8

aschiffler, 2022-08-15:
closed due to inactivity

closed 2022-08-15 by aschiffler