lumapu/ahoy #154: power limits not working in 0.5.10?

duenengeist, 2022-08-14:
Hi, 
I have an HM600. Limit worked fine for me in 0.5.1. Now I am on the mist recent 0.5.10 and my inverter does not limit.

[image]

Any ideas what I can check?

Carsten

MubiTec, 2022-08-14:
I have the same Problem

lumapu, 2022-08-14:
which is the version which destroyed this feature? Came it up from 0.5.1 to 0.5.2?
It's easier to find the issue if we don't have to diff that many versions.

duenengeist, 2022-08-14:
> It's easier to find the issue if we don't have to diff that many versions.

I see, unfortunately I jumped directly from 0.5.1 to 0.5.10...

MubiTec, 2022-08-14:
Sorry, I cannot help, too. I come directly from 0.4.26. I am not @home,perhaps I can test <0.5.10 Version later

duenengeist, 2022-08-14:
I also can try to check later today ...

MubiTec, 2022-08-14:
AT 0.5.8 it works. Testing now 0.5.9

lumapu, 2022-08-14:
Ok I the issue just be related to the callback function of MQTT I changed recently For now I don't understand why the issue occurs.
We have to verify if the Mqtt callback is called once the subscribed topic is changed.

Other idea because of your screenshot. You set a persistent limit inside the setup to 100W? Did you reboot your device after changing the setting?
In past the were some voices which said that sometimes the Ahoy software send an additional zero with the limit, do a limit of 100W became 1000W.

MubiTec, 2022-08-14:
I changed it not via mqtt, I changed it via setup Page and did a reboot all Times after it

duenengeist, 2022-08-14:
> Other idea because of your screenshot. You set a persistent limit inside the setup to 100W? 
Yes in the setup via Webbrowser
> Did you reboot your device after changing the setting?
Yes

I can try to set it to 10W...

lumapu, 2022-08-14:
In the last versions was no change regarding persistent limit as far as I know.
Thank you for checking the versions.

duenengeist, 2022-08-14:
I guess the limit is off by a factor of 10. I set 1W on my HM600 and it limits now to 10W:
[image]

duenengeist, 2022-08-14:
Ich habe noch mal ein wenig experimentiert mit verschiedenen Werten für das Limit. Der WR limitiert in der 0.5.10, wenn ich über das Webinterface recht kleien Werte setze (10, 20, 50) aber die gemessene Ausgangsleistung passt nicht. Auch nicht mit dem Faktor 10. Irgendwas scheint mit der Berechnung nicht zu stimmen.

duenengeist, 2022-08-15:
Inzwischen bin ich mir recht sicher: die Werte sind in %. Ich kann bei mir Werte zwischen 2% und 100% setzen und es wird akzeptiert. Vielleicht in der Webpberfläche von "W" auf "%" ändern?

fila612, 2022-08-15:
Falls dem so ist, wäre doch eine Vorauswahl z.B. via Dropdown-Menü eine gute Möglichkeit, oder?

aschiffler, 2022-08-15:
Bitte beachten:
1. Die Einstellung des Limits gilt bzw. ist implementiert für WR der 3. Generation
2. Limits die kleiner sind als 2% der Spitzenleistung werden vom WR nicht angenommen
3. Debug-Option: Im seriellen Log via USB findet man "...has accepted pwr limit" oder "... has not accepted pwr limit" ggf. hilft dies für das Verständnis.
4. Aktuell >0.5.1 & < 0.5.12 ist es so, dass es beim Setzen des Limits via MQTT zu einem anderen Regelverhalten kommen kann wie beim Setzen via HTML Setup-Page. Dies liegt daran, dass es z.Zt. so ist, dass das eingestellte Limit via MQTT beim nächsten Power-Cycle des WR (vermutlich DC=0W) seitens des WR wieder verworfen wird.

closed 2022-08-15 by aschiffler