lumapu/ahoy #155: Inverter is available and is not producing

martin-at-schorsch, 2022-08-14:
I installed 0.5.5(?) on Tuesday, where the default power limit was set to 0. 
Now the HM is not producing anymore, I suspect that the power limit is still set to 0 even when I set it to 800...
0.5.10 did not work, I also tried 0.5.7 and 0.5.5, no change.

Any idea how to enable production again?

duenengeist, 2022-08-14:
I had the same issue when testing 0.5.1 for the first time. I set i back to 600 (for my HM600), but it took a while (5-10 Minutes) until the HM started to produce again.

martin-at-schorsch, 2022-08-14:
I tested it in 0.5.10, there the reset did not work. I guess 0.5.7 was also more than 10 minutes running.
With 0.5.5 indeed it started again after 10 minutes... Looks like I was not patient enough.

Maybe the issue on 0.5.10 is related to #154

hubsi5, 2022-08-15:
Ich habe das Problem seit der letzten Version 0.5.12 g9a14d53
Es wird nichts mehr produziert.
Habe schon versucht die Limitierung zu ändern. Wird im Web angezeigt nur produziert wird nichts.
Habe auch alte Versionen 0.5.9, 0.5.5 versucht.
Habe den Wechselrichter auch schon vom Strom genommen und gewartet.
Hilft leider alles nichts.
Kann ich den noch irgendwie resetten ?
Problem trat auf, als ich die neue ahoy Version raufgespielt habe und dieser startete.

duenengeist, 2022-08-15:
Hallo,

ich bin mir inzwischen recht sicher, dass die Werte in % sind. Zwischen 2% und 100% kann ich es bei meinem HM600 setzen und es wird akzeptiert. Versucht auf "100" zu setzen und beobachtet die serielle Ausgabe. Dort ist zu sehen, ob der Inverter das Limit akzeptiert hat.

fila612, 2022-08-15:
Ich hatte einmal Erfolg indem ich das amplifier Level geändert hatte

hubsi5, 2022-08-15:
Leider geht nichts mehr.
Natürlich nicht lustig.
Hoffe das erfangt sich wieder irgendwann, ansonsten ist der HM1500 nicht alt geworden

lumapu, 2022-08-15:
kann mir nicht erklären warum das passiert sein soll, hoffentlich haben nicht noch weitere dieses Problem. 
Danke für das Teilen der Lösung.

guilligan71, 2022-08-16:
Seit Version 0.5.12 ist als 3. Inverter der 00-4 voreingestellt (ich habe nur 2 Inverter)
Der lässt sich zwar löschen, aber ist nach dem reboot wieder da und produziert gefühlt Fehler in der Statistik.

aschiffler, 2022-08-16:
@guilligan71 Versuche mal ein "Erase settings" (oben rechts)

guilligan71, 2022-08-16:
> @guilligan71Versuche mal ein "Einstellungen löschen" (oben rechts)

Hat funktioniert.

closed 2022-08-17 by aschiffler