lumapu/ahoy #156: V5.12 interval Probleme

TomMeyer90, 2022-08-15:
bei der V5.12 steht der interval bei den Invertern immer auf 5s und lässt sicht nicht mehr ändern.
Der interval bei MQTT steht immer auf 0s somit werden keine Daten mehr übertragen.

fila612, 2022-08-15:
MQTT Intervall kann ich bestätigen => weiterhin nur readonly das Feld und zeigt "0" an.
Das Inverser-Intervall kann ich ändern, wird aber nicht gespeichert (nach einem reboot wird wieder der alte Wert anzeigt.)
Einzig das Intervall in der Seriellen Konsole lässt sich ändern und speichert auch korrekt, dafür wird der Haken bei "print RF24 debug" nicht gespeichert  (nach reboot wieder weg).

lumapu, 2022-08-15:
danke für die Hinweise, beim MQTT Intervall kann ich mir vorstellen, das hier was falsch implementiert ist. Bei mir überträgt er trotzdem, da das Intervall noch mit dem Sendintervall verrechnet wird.

HorstG-57, 2022-08-16:
Beim MQTT Intervall ist es inzwischen so eingebaut, 
das die MQTT Daten übertragen werden, sobald die Daten die vom Inverter gelesen wurden aufbereitet wurden.
Steht das Abfrage-Intervall der Inverter auf 30 Sekunden und die Daten die vom Inverter kommen können ohne Probleme intern aufbereitet werden, werden die Daten auch mit einem Intervall vom 30 Sekunden zum MQTT Broker übertragen.

Das Normale MQTT Intervall ist das Normale Inverter Abfrage-Intervall + 60 Sekunden und es wirkt nur, solange keine Daten vom Inverter kommen ( also z.B. Nachts ).  
Es wird dann nur die Uptime übertragen, weil ja keine Daten vom Inverter ankommen ( so erhält man auch Nachts ein Lebenszeichen vom Ahoy ).

aschiffler, 2022-08-16:
Ist es dann noch nötig diesen Wert im Setup anzuzeigen? Ich würde das dann lieber entfernen.

fila612, 2022-08-16:
> Ist es dann noch nötig diesen Wert im Setup anzuzeigen? Ich würde das dann lieber entfernen.

Sehe ich auch so, im Setup sollten m.M.n. nur Parameter sein, welche man ändern kann.

aschiffler, 2022-08-16:
Ich habe mir erlaubt es zu entfernen.
https://github.com/grindylow/ahoy/commit/ac79828eb511700a998e828892749bb63aba3f15

closed 2022-08-16 by aschiffler