lumapu/ahoy #163: Arduino Seriennummer Inverter

DeeJayMykel, 2022-08-17:
Hallo,
ich bekomme beim Kompilieren folgende Meldung:
ModWebserver.h:55:33: error: 'getSerialNoTxt' was not declared in this scope
   55 |     out += "<h3>S/N " + String (getSerialNoTxt(wr)) + "</h3>";
      |                                 ^~~~~~~~~~~~~~
exit status 1
'getSerialNoTxt' was not declared in this scope

Leider bin ich kein Programmierexperte. Kann mir jemand sagen in welcher Zeile bzw. Tab ich diese eintragen kann?
Vielen Dank.

stefan123t, 2022-08-17:
@DeeJayMykel das unter HoyDtuSim ist die Software von Hubi. MW stellt er die immer wieder auf Anfrage ins Forum. Die meisten hier im ahoy Repo arbeiten an der ESP8266 Variante die auch als Vorlage für die ESP32 Variante (OpenDTU) von Thomas dient.
Raspberry Pi wird auch mehr oder weniger aktiv hier maintained (Jan-Jonas S & Daniel R).
Kurz ich weiß es leider nicht welchen Stand die HoyDtuSim Software vom 28.4.2022 aktuell hat.

stefan123t, 2022-08-17:
Ich vermute Hubi pflegt seine Urversion HoyDtuSim unter https://github.com/hm-soft/Hoymiles-DTU-Simulation/

@aschiffler vielleicht sollten wir das Verzeichnis zugunsten eines Links auf Hubis Repo entfernen. Vielleicht machen wir das genauso mit einem Verzeichnis OpenDTU oder anderen Repos die hiervon abstammen aber sich bereits verselbständigt haben ?

stefan123t, 2022-08-17:
Andere Repos:
MI-1200..1500, MI-600, etc ?
https://github.com/Ziyatoe/DTUsimMI

ESPHome Variante von marcel-a
https://github.com/a-marcel/ahoy/tree/main/tools/esphome

ESP32 OpenDTU von tbnobody
https://github.com/tbnobody/OpenDTU/

https://github.com/tbnobody/OpenDTU/blob/master/docs/Wiring_ESP32.fzz

??? von O Freddy
https://github.com/OFreddy/hm_poll

DeeJayMykel, 2022-08-18:
Danke für die schnelle Antwort.

stefan123t, 2022-08-19:
@aschiffler 
Reopened to add the info to our Readme.md and delete the HoyDtuSim and eventually pther directories under tools/ which are unmaintained.

closed 2022-09-05 by aschiffler