lumapu/ahoy #164: Settings in settings.ini speichern (LittleFS statt EEPROM)

stefan123t, 2022-08-17:
Um die Settings zu speichern und beim Booten / Startup wieder zu lesen kann statt der festen EEPROM Struktur ggf auch ein SPIFFS/LittleFS verwendet werden.
* [SPIFFS](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html) ist eigentlich deprecated wird aber noch ca 1 Jahr in Arduino/ESP8266 weiter mitgeführt.
* [LittleFS](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html) ist ein copy-on-write FS (4k Blöcke) und hat daher eine schlechte(re) [random write performance](https://github.com/esp8266/Arduino/issues/7271)
* Anleitung für LittleFS + PlatformIO [hier](https://randomnerdtutorials.com/esp8266-nodemcu-vs-code-platformio-littlefs/)

Es gibt mE zwei Möglichkeiten die Daten in der settings.ini/.json Datei abzuspeichern:
* [SPIFFSIniFile](https://www.arduino.cc/reference/en/libraries/spiffsinifile/)
* [ArduinoJson](https://www.mischianti.org/2020/01/26/manage-json-file-with-arduino-and-esp8266/)

stefan123t, 2022-08-18:
Es gibt noch ein Caveat bzgl SPIFFS/LittleFS vs EEPROM:

1. Internal EEPROM  –  512 Bytes. You can write data 1 million times (but it has no file system).
2. Flash memory – connected to ESP through SPI. In this flash memory, ESP stores the program, files, WiFi, and configurations. Limitation of this memory is it has only 10000 write cycles.

stefan123t, 2022-09-08:
Wir müssen uns hier irgendwann entscheiden, ob wir eine ini Datei für die Config Properties der AhoyDTU pflegen wollen oder eine ArduinoJSON Datei.
1. Ini Datei:
+einfache Eintragungen
-Typsicherheit ?
2. Json Datei
+Bibliothek ist bereits im Projekt included
-kompliziertere Syntax
-kein Hex Datentyp (vielleicht kann man das ja mit einem String machen ?) 
   see [#439](https://github.com/bblanchon/ArduinoJson/issues/439) though JSON5 seems to suport hex in json docs.

DanielR92, 2022-09-23:
Ich bin für Json, somit lässt sich später auch easy eine Export/Import funktion einbauen. 

Wenn ich nächste Woche Zeit habe würde ich mich damit ausseinander setzen. 
Hatte schon Testweise LCD in Betrieb genommen (einfacher Text). 
RS485 war schwieriger (oder es liegt an mir). 

Aber diese Funktion würde ich mal als wichtig entnehmen...

lumapu, 2022-09-23:
aktuell gibt es doch keine nennenswerten Probleme mit dem Settings.
Ich bin nicht dafür es über das Filesystem zu machen, aktuell bin ich sehr glücklich wie es ist.

stefan123t, 2022-09-24:
@lumapu +1 für den bisherigen Weg über EEPROM/Preferences settings.
Beim LittleFS wird mit mklittlefs.bin ein Filesystem Image erzeugt das mW alle Daten/Dateien auf dem bisherigen ESP filesystem ersetzt. Dabei werden auch ggf. dort gespeicherte Config settings überschrieben.
Ich habe noch keinen Weg gefunden die Daten aus data/ nur als Overlay zu flashen dh das andere Dateien wie config.ini/config.json erhalten bleiben.

stefan123t, 2022-11-07:
@lumapu und @DanielR92 es mehreren sich die Anfragen und Wünsche ob wir die Settings auf JSON umstellen können.
Vielleicht ändern wir den Titel dieses Issues und machen erst die Umstellung auf JSON ?
Die JSON Config würde ich aber im EEPROM Bereich belassen.

Dann können wir den großen Rest des Flash Speichers jederzeit für LittleFS oder SPIFFS verwenden um dort z.B. HTML, CSS, SVG etc. Template Dateien für die Web Anwendung abzulegen ?
Mit den neuen ESP Web Tools auf https://ahoydtu.de/ können dann ja auch derartig komplexe Memory Layouts auf einen Schlag geflasht werden.

lumapu, 2022-11-08:
ja wir sollten sehr bald was ändern, es nett zu zunehmend, dass ständig alles neu eingegeben werden muss.
Gibt es irgendein Beispiel wie wir es einfach integrieren könnten?

lumapu, 2022-11-08:
@stefan123t bei den Settings sind wir bzgl der Größe inzwischen bei 0x2f2 von 0x3ff angekommen.
Ich bin mir nicht sicher, ob der Overhead vom JSON den Rahmen nicht sprengt.

DanielR92, 2022-11-16:
Danke an @lumapu, ist nun auf LittleFuse geändert worden. Whoho! :D
Kann dann geclosed werden?

lumapu, 2022-11-16:
nicht fuse sondern fs (Filesystem)
ich weiß du meintest fs ;-)

closed 2022-11-16 by lumapu