lumapu/ahoy #169: [DEV-Branch] 0.5.14 ga06341a läuft instabil aufgrund ständiger Reboots

fila612, 2022-08-18:
ich hab o.g. Version auf meinem Testsystem. Nachdem ich die Probleme wie in #162 zu Beginn ebenfalls hatte, aber vermutlich lösen konnte, stelle ich nun fest, dass das System im Vergleich zur 5.13 um einiges instabiler läuft und häufig abstürzt.
in den letzten 10min 3 mal.
lt. Konsole sieht das immer so aus:

07:34:42.992 > 
07:34:42.993 >  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
07:34:42.995 > 
07:34:42.995 > wdt reset
07:34:42.995 > load 0x4010f000, len 3460, room 16 
07:34:43.003 > tail 4
07:34:43.003 > chksum 0xcc
07:34:43.003 > load 0x3fff20b8, len 40, room 4 
07:34:43.010 > tail 4
07:34:43.010 > chksum 0xc9
07:34:43.010 > csum 0xc9
07:34:43.010 > v0005f2d0
07:34:43.013 > ~ld

evtl. hilft es ja bei Analysen. ich gebe Rückmeldung wenn ich weitere Infos bemerke.

stefan123t, 2022-08-18:
Hier der Guide von Espressif https://www.espressif.com/sites/default/files/documentation/esp8266_reset_causes_and_common_fatal_exception_causes_en.pdf

fila612, 2022-08-18:
Guter Hinweis - mach ich mal

DanielR92, 2022-08-18:
Wäre es Sinnvoll aus der PDF eine Abfrage beim nächsten Start abzufragen bezüglich eines Crashes?

struct rst_info *rtc_info = system_get_rst_info();
os_printf("reset reason: %x\n", rtc_info->reason);

if (rtc_info->reason == REASON_WDT_RST || 
rtc_info->reason == REASON_EXCEPTION_RST ||
rtc_info->reason == REASON_SOFT_WDT_RST) 
{
  if (rtc_info->reason == REASON_EXCEPTION_RST)  {
  os_printf("Fatal exception (%d):\n", rtc_info->exccause);
  }
os_printf("epc1=0x%08x, epc2=0x%08x, epc3=0x%08x, excvaddr=0x%08x,
depc=0x%08x\n",
rtc_info->epc1, rtc_info->epc2, rtc_info->epc3, rtc_info->excvaddr, rtc_info->depc); 
//The address of the last crash is printed, which is used to debug garbled output.
}

stefan123t, 2022-08-18:
@DanielR92 könnte lustig sein wobei eine Einschränkung genannt war wann das eigentlich sinnvoll ist:
Use this method when garbled output is printed where crash occurs and can not be debugged.

fila612, 2022-08-19:
ich habe es mit dem Debug-Release nicht mehr reproduzieren können. Sollen wir dann hier schließen oder braucht jemand das Issue aufgrund des Vorschlags von @DanielR92?

stefan123t, 2022-08-19:
@fila612 danke für die positive Rückmeldung! Ich schließe das Issue als „cannot reproduce“

closed 2022-08-19 by aschiffler